revyos/revyos #11: terminal文字选中存在偏移

FIFCC, 2023-07-10:
在terminal中选中文字时，鼠标指针与实际选中情况存在偏差

FIFCC, 2023-07-10:
偏移量大约是横向7像素，纵向7像素（右下鼠标指针）
[image]

aiminickwong, 2023-08-16:
> 偏移量约是横向7像素，纵向7像素（观测右下鼠标指针） [image]

**I also encountered this BUG, ​​has anyone confirmed and responded?**

nekorouter, 2023-10-09:
This problem is related with hardware cursor render, enable software cursor in X11 can temporary fix this issue.

Rabenda, 2023-10-14:
https://github.com/revyos/mkimg-th1520/commit/f9867c522485046f1965a3101bd4545520803623

closed 2023-10-14 by Rabenda