revyos/revyos #16: [gcc-10] Bad code generation for Vector Mask-Register Logical Operations.

gandalfn, 2023-08-21:
I'm trying to use the vectorization feature of GCC-10 on RevyOS. 

Here's a small example that attempts to remove duplicate entries in a vector of integers. 
It's built with GCC-10 from the revyoS-c910v repository.

**unique.cpp**
```c++
#include <cstdio>
#include <cstdint>
#include <cstdlib>
#include <ctime>
#include <string>
#include <sstream>
#include <iostream>
#include <chrono>
#include <riscv_vector.h>

inline static size_t vlmax()
{
    static size_t max = vsetvlmax_e32m4();
    return max;
}

inline vint32m4_t scalarToVec(int32_t inVal)
{
    return vmv_v_x_i32m4 ( inVal, vlmax() );
}

inline vint32m4_t load ( const int32_t* inpArray )
{
    return vle32_v_i32m4 ( inpArray, vlmax() );
}

inline void store ( int32_t* outpArray, vint32m4_t inVal )
{
    vse32_v_i32m4 ( outpArray, inVal, vlmax() );
}

inline std::string toString ( vint32m4_t inVal )
{
    std::stringstream content;
    int32_t s[vlmax()] = {0};

    store ( s, inVal );

    for ( size_t iter = 0; iter < vlmax(); ++iter )
    {
        content << s[iter] << ",\t";
    }

    return content.str();
}

inline std::string toString ( vbool8_t inVal )
{
    vint32m4_t sequence =  scalarToVec ( 0 );
    
    vint32m4_t val = vadd_vx_i32m4_m ( inVal, sequence, sequence, 1, vlmax() );

    return toString(val);
}

inline vint32m4_t shiftRight ( vint32m4_t inVal, size_t inOffset )
{
    vint32m4_t dstDown = vslidedown_vx_i32m4 ( vundefined_i32m4(), inVal, vlmax() - inOffset, vlmax() );
    return vslideup_vx_i32m4 ( dstDown, inVal, inOffset, vlmax() );
}

inline vint32m4_t filter ( vint32m4_t inVal, vbool8_t inFilter )
{
[... 280 lines cut ...]
rcmp:  1,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	
cpt: 15
val:   6,	5,	7,	8,	3,	7,	4,	9,	2,	3,	8,	3,	6,	8,	4,	6,	
shift: 5,	7,	8,	3,	7,	4,	9,	2,	3,	8,	3,	6,	8,	4,	6,	6,	
test:  0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	
cmp:   1,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	
rcmp:  1,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	0,	
vals: 6, 5, 7, 8, 3, 7, 4, 9, 2, 3, 8, 3, 6, 8, 4, 6, 
ret:  0, 5, 7, 8, 3, 7, 4, 9, 2, 3, 8, 3, 6, 8, 4, 6, 
```
We notice that the result of vmwor_mm_b8 between the "test" and "cmp" returns an incorrect value. After investigating, I discovered that in the generated code, vmor_mm_b8 produces the following code:
```asm
...
vsetvli zero,zero,e8,m1,tu,mu
vle.v   v1,(a5)
mul     a5,a4,a3
add     a5,a5,-544
add     a5,a5,s0
vle.v   v2,(a5)
vsetvl  zero,t5,t6
mul     a5,a4,a3
vmor.mm v1,v1,v2
...
```
You can observe that the generated code enforces SEW 8 and LMUL 1, which corresponds to the vbool8_t format, but the mask is intended for SEW 32 LMUL 4. If we enforce the correct format using inline assembly (enabled with the -DWORKAROUND option during the build), the comparison works fine.

pz9115, 2023-08-23:
I think the result is it not support auto-vectorization yet, since it still under developing in gcc 13, it will be a new feature in gcc14. In this time, rvv only support completely intrinsic calls.

gandalfn, 2023-08-23:
> I think the result is it not support auto-vectorization yet, since it still under developing in gcc 13, it will be a new feature in gcc14. In this time, rvv only support completely intrinsic calls.

I am using only intrinsic calls, and GCC 10.4 which does not support auto-vectorization. I suspect that the automatic vsetvli for the vbool8_t type to e8 m1 is the source of the problem, as I have observed in the GCC source.

Cooper-Qu, 2023-08-24:
The vtype needs to be set for the Vector Mask-Register Logical Operations.
The compiler cannot infer the vtype of these interfaces from the interface name. For example, both i32m4 and i16m2 are x xxx_b8.
Change the line in the code
`    return vmor_mm_b8(inLeft, inRight, vlmax());`
to
`    return vmor_mm_b8(inLeft, inRight, vsetvlmax_e32m4());`
and then it can get the correct result.

gandalfn, 2023-08-25:
Effectively forcing "vsetvlmax_e32m4" resolves the issue. 
I still tried to understand why this is necessary. Upon inspecting the "vmxor_mm_b8" function in the "riscv_vector" header by running "gcc -E riscv_vector.h,"  I noticed that unlike the other functions, the functions related to vector masks do not include the "vsetvl_..." instructions and ignore vl paramerter. 
```c++
...
 __extension__ extern __inline vbool8_t __attribute__ ( ( __always_inline__, __gnu_inline__, __artificial__ ) ) vmor_mm_b8 ( vbool8_t a, vbool8_t b, word_type vl )
{
    return __builtin_riscv_vorbool8 ( a, b );
} 
...
```
This makes sense as it's not feasible to determine it universally; each function should be specialized for the specific SEW and LMUL involved. 

By incorporating the "vsetvlmax_e32m4" call, the problem gets resolved as it introduces the necessary "vsetvl..." call.

Currently, I prefer to address the problem using this approach.
```c++
...
inline vbool8_t bitwise_or ( vbool8_t inLeft, vbool8_t inRight )
{
        vsetvl_e32m4 ( vlmax() );
        return vmor_mm_b8(inLeft, inRight, 0);
}
```
Given that the 'vl' parameter is ignored, I believe it might be necessary to consider adding specific functions to "riscv_vector.h" to handle this particular scenario.

closed 2026-01-07 by RevySR