revyos/revyos #24: gcc -march=rv64gcv0p7 还在使用v1p0

hleft, 2023-09-19:
环境: lichee4a + gcc10/13

```
sudo sed -i '1ideb https://mirror.iscas.ac.cn/revyos/revyos-c910v/ revyos-c910v main' /etc/apt/sources.list
sudo apt update
sudo apt install -y build-essential # gcc-10
sudo apt install -y gcc-13 g++-13 # gcc-13
sudo reboot
```

从riscv-v-spec-0.7.1 的16.8复制下面的内容到vlw.s

```
compact_non_zero: 
    li a6, 0                      # Clear count of non-zero elements 
loop: 
    vsetvli a5, a0, e32,m8  # 32-bit integers 
    vlw.v v8, (a1)                # Load input vector 
      sub a0, a0, a5              # Decrement number done 
      slli a5, a5, 2              # Multiply by four bytes 
    vmsne.vi v0, v8, 0            # Locate non-zero values 
      add a1, a1, a5              # Bump input pointer 
    vmpopc.m a5, v0               # Count number of elements set in v0 
    viota.m v16, v0              # Get destination offsets of active elements 
      add a6, a6, a5              # Accumulate number of elements 
    vsll.vi v16, v16, 2, v0.t     # Multiply offsets by four bytes 
      slli a5, a5, 2              # Multiply number of non-zero elements by four bytes 
    vsuxw.v v8, (a2), v16, v0.t   # Scatter using scaled viota results under mask 
      add a2, a2, a5              # Bump output pointer 
      bnez a0, loop               # Any more? 
 
      mv a0, a6                   # Return count 
      ret
```

gcc-10 -c vlw.s -march=rv64gcv0p7

```
vlw.s: Assembler messages:
vlw.s:5: Error: unrecognized opcode `vlw.v v8,(a1)'
vlw.s:10: Error: unrecognized opcode `vmpopc.m a5,v0'
vlw.s:15: Error: unrecognized opcode `vsuxw.v v8,(a2),v16,v0.t'
```

gcc-10 -march=rv64gcv0p7 main.c
readelf -A a.out

```
Attribute Section: riscv
File Attributes
  Tag_RISCV_stack_align: 16-bytes
  Tag_RISCV_arch: "rv64i2p1_m2p0_a2p1_f2p2_d2p2_c2p0_v0p7_zicsr2p0_zifencei2p0_zmmul1p0_zfh1p0_zfhmin1p0_zve32f1p0_zve32x1p0_zve64d1p0_zve64f1p0_zve64x1p0_zvl128b1p0_zvl32b1p0_zvl64b1p0_xtheadba1p0_xtheadbb1p0_xtheadbs1p0_xtheadc1p0_xtheadcmo1p0_xtheadcondmov1p0_xtheadfmemidx1p0_xtheadfmv1p0_xtheadint1p0_xtheadmac1p0_xtheadmemidx1p0_xtheadmempair1p0_xtheadsync1p0"
```

attribute看来在用1.0

Rabenda, 2023-09-19:
```
debian@lpi4a:~$ gcc -march=rv64gc -mabi=lp64d a.c
debian@lpi4a:~$ readelf -A a.out
Attribute Section: riscv
File Attributes
  Tag_RISCV_arch: "rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v0p7_zmmul1p0_zfh1p0_zfhmin1p0_xtheadba1p0_xtheadbb1p0_xtheadbs1p0_xtheadc1p0_xtheadcmo1p0_xtheadcondmov1p0_xtheadfmemidx1p0_xtheadfmv1p0_xtheadint1p0_xtheadmac1p0_xtheadmemidx1p0_xtheadmempair1p0_xtheadsync1p0"
debian@lpi4a:~$ gcc -march=rv64gcv0p7_zfh_xtheadc -mabi=lp64d a.c
debian@lpi4a:~$ readelf -A a.out
Attribute Section: riscv
File Attributes
  Tag_RISCV_arch: "rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v0p7_zmmul1p0_zfh1p0_zfhmin1p0_xtheadba1p0_xtheadbb1p0_xtheadbs1p0_xtheadc1p0_xtheadcmo1p0_xtheadcondmov1p0_xtheadfmemidx1p0_xtheadfmv1p0_xtheadint1p0_xtheadmac1p0_xtheadmemidx1p0_xtheadmempair1p0_xtheadsync1p0"
debian@lpi4a:~$ nano vlw.s
debian@lpi4a:~$ gcc-10 -c vlw.s -march=rv64gcv0p7
```

无法复线

Rabenda, 2023-09-19:
[image]

请严格按照操作步骤操作

hleft, 2023-09-19:
[image]

原因： 前面几天lichee4a的文档都有`sudo apt upgrade -y`，今天复制时`sudo apt upgrade -y` 却没有，然后我自己复制过来执行，也没更新，然后binutils版本不对，导致了问题

closed 2023-09-19 by hleft

Rabenda, 2023-09-19:
https://revyos.github.io/docs/build/debian/enable_optimization_gcc/
补充文档