revyos/revyos #25: 5 GHz Wi-Fi is not detected on lichee pi 4a with 20230916

bgklc, 2023-09-19:
[    3.756049] [WLAN_RFKILL]: Enter rfkill_wlan_init
[    3.761862] [WLAN_RFKILL]: Enter rfkill_wlan_probe
[    3.766816] [WLAN_RFKILL]: can't find rockchip,grf property
[    3.772487] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = rtl8723ds
[    3.779749] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi power will enabled while kernel starting and keep on.
[    3.789742] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi power controled by gpio.
[    3.797346] [WLAN_RFKILL]: wlan_platdata_parse_dt: The ref_wifi_clk not found !
[    3.804732] [WLAN_RFKILL]: rfkill_wlan_probe: init gpio
[    3.810034] [WLAN_RFKILL]: rockchip_wifi_power: 1
[    3.814804] [BT_RFKILL]: rfkill_get_bt_power_state: rfkill-bt driver has not Successful initialized
[    3.823951] [WLAN_RFKILL]: wifi turn on power. -1
[    3.828726] [WLAN_RFKILL]: Exit rfkill_wlan_probe
[    3.833762] [BT_RFKILL]: Enter rfkill_rk_init
[    3.839006] [BT_RFKILL]: bluetooth_platdata_parse_dt: uart_rts_gpios is no-in-use.
[    3.846746] [BT_RFKILL]: bluetooth_platdata_parse_dt: clk_get failed!!!.
[    3.853967] [BT_RFKILL]: bt_default device registered.

opsiff, 2023-09-19:
Maybe the reson is the contorller ,as only support 2.4Ghz .
In realtek website they say "RTL8723DS-CG is an 802.11bgn 2.4G single-chip"
Link:https://www.realtek.com/en/products/communications-network-ics/item/rtl8723ds

nekorouter, 2023-09-24:
This board use RTL8723DS Wifi/BT combo chip, which is 2.4Ghz only.

closed 2023-09-24 by nekorouter