revyos/revyos #26: bluetooth controller is not detected on lichee pi 4a with 20230916

bgklc, 2023-09-19:
debian@lpi4a:~$ sudo systemctl restart bluetooth
debian@lpi4a:~$ sudo bluetoothctl
Agent registered
[bluetooth]# scan on
No default controller available
[bluetooth]#

nekorouter, 2023-09-19:
Make sure rtk-hciattach.service is running when using Bluetooth.
Also more information is welcomed.

bgklc, 2023-09-20:
[dmesg.txt](https://github.com/revyos/revyos/files/12669200/dmesg.txt)

nekorouter, 2023-09-24:
Do you have this problem only on hot-reset or both hot/cold start?
Also which image do you using? (source of image file and cat /etc/revyos-release)

bgklc, 2023-09-24:
Image from https://mirror.iscas.ac.cn/revyos/extra/images/lpi4a/20230916/

debian@lpi4a:~$ cat /etc/revyos-release
20230916-231502

Right after installation 
sudo bluetoothctl
Agent registered
[bluetooth]# scan on

did succeed, it found the controller and several discovered devices, I did not save its output though. After reboot it fails. 
debian@lpi4a:~$ systemctl status rtk-hciattach
× rtk-hciattach.service - Realtek H5 bluetooth support
     Loaded: loaded (/etc/systemd/system/rtk-hciattach.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sun 2023-09-24 16:28:44 UTC; 2min 16s ago
   Duration: 5.519s
    Process: 782 ExecStart=/usr/local/bin/rtk_hciattach -n -s 115200 /dev/ttyS4 rtk_h5 (code=exited, stat>
   Main PID: 782 (code=exited, status=1/FAILURE)
        CPU: 11ms

вер 24 16:28:44 lpi4a rtk_hciattach[782]: Realtek Bluetooth WARN: OP_H5_SYNC Transmission timeout
вер 24 16:28:44 lpi4a rtk_hciattach[782]: Realtek Bluetooth WARN: OP_H5_SYNC Transmission timeout
вер 24 16:28:44 lpi4a rtk_hciattach[782]: Realtek Bluetooth WARN: OP_H5_SYNC Transmission timeout
вер 24 16:28:44 lpi4a rtk_hciattach[782]: Realtek Bluetooth WARN: OP_H5_SYNC Transmission timeout
вер 24 16:28:44 lpi4a rtk_hciattach[782]: Realtek Bluetooth WARN: OP_H5_SYNC Transmission timeout
вер 24 16:28:44 lpi4a rtk_hciattach[782]: Realtek Bluetooth WARN: OP_H5_SYNC Transmission timeout
вер 24 16:28:44 lpi4a rtk_hciattach[782]: Realtek Bluetooth WARN: OP_H5_SYNC Transmission timeout
вер 24 16:28:44 lpi4a rtk_hciattach[782]: Realtek Bluetooth ERROR: Retransmission exhausts
вер 24 16:28:44 lpi4a systemd[1]: rtk-hciattach.service: Main process exited, code=exited, status=1/FAILU>
вер 24 16:28:44 lpi4a systemd[1]: rtk-hciattach.service: Failed with result 'exit-code'.

After poweroff + cold start it works.

nekorouter, 2024-06-06:
This problem has been resolved after 20240601, closed this issue.

closed 2024-06-06 by nekorouter