revyos/revyos #27: Sporadic Nodejs segfault or strange errors on sg2042 and lpi4a

kxxt, 2023-10-12:
Hi,

I got sporadic nodejs segfault or strange errors on sg2042 and lpi4a when building electron and chromium.

On sg2042, segfaults are encountered. The stack traces only contain one line and look cryptic. examples:
- `#0  0x0000003f839f1d64 n/a (libnode.so.115 + 0xdf1d64)`
- `#0  0x0000003f7c1a7460 n/a (n/a + 0x0)`
- `#0  0x0000003f8f847eb0 n/a (n/a + 0x0)`
- `#0  0x0000003f6e9b31f8 n/a (n/a + 0x0)`

On lpi4a, strange errors are encountered.  examples:
<details>

<summary> Sample 1</summary>

```
FAILED: gen/front_end/ui/components/diff_view/diff_view-tsconfig.json gen/front_end/ui/components/diff_view/DiffView.js gen/front_end/ui/components/diff_view/DiffView.js.map gen/front_end/ui/components/diff_view/DiffView.d.ts  
python3 ../../third_party/typescript/ts_library.py --tsconfig_output_location gen/front_end/ui/components/diff_view/diff_view-tsconfig.json --deps ./css_files-tsconfig.json ../../../core/common/bundle-tsconfig.json ../../../core/sdk/bund  
le-tsconfig.json ../../../third_party/diff/bundle-tsconfig.json ../../lit-html/bundle-tsconfig.json ../code_highlighter/bundle-tsconfig.json ../helpers/bundle-tsconfig.json --front_end_directory ../../front_end/ui/components/diff_view --  
reset_timestamps --sources ../../front_end/ui/components/diff_view/DiffView.ts  
  
TypeScript compilation failed. Used tsconfig gen/front_end/ui/components/diff_view/diff_view-tsconfig.json  
  
/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:117138  
const prevSignature = info.signature;  
^  
  
TypeError: Cannot read properties of undefined (reading 'signature')  
at updateShapeSignature (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:117138:32)  
at Object.getFilesAffectedByWithOldState (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:117099:10)  
at getNextAffectedFile (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:117582:40)  
at getSemanticDiagnosticsOfNextAffectedFile (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:118363:24)  
at Object.getSemanticDiagnostics (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:118399:12)  
at emitFilesAndReportErrors (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:119946:42)  
at emitFilesAndReportErrorsAndGetExitStatus (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:119983:39)  
at performIncrementalCompilation (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:120206:22)  
at performIncrementalCompilation2 (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:123702:22)  
at executeCommandLineWorker (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:123479:7)  
  
Node.js v20.8.0
```

</details>
<details>

<summary> Sample 2</summary>

```
FAILED: gen/front_end/entrypoints/lighthouse_worker/lighthouse_worker-tsconfig.json gen/front_end/entrypoints/lighthouse_worker/LighthouseWorkerService.js gen/front_end/entrypoints/lighthouse_worker/LighthouseWorkerService.js.map gen/fro  
nt_end/entrypoints/lighthouse_worker/LighthouseWorkerService.d.ts  
python3 ../../third_party/typescript/ts_library.py --tsconfig_output_location gen/front_end/entrypoints/lighthouse_worker/lighthouse_worker-tsconfig.json --deps ../../core/root/bundle-tsconfig.json ../../core/sdk/bundle-tsconfig.json ../  
../services/puppeteer/bundle-tsconfig.json ../../third_party/lighthouse/lighthouse-tsconfig.json --front_end_directory ../../front_end/entrypoints/lighthouse_worker --reset_timestamps --sources ../../front_end/entrypoints/lighthouse_work  
er/LighthouseWorkerService.ts  
  
TypeScript compilation failed. Used tsconfig gen/front_end/entrypoints/lighthouse_worker/lighthouse_worker-tsconfig.json  
  
/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:117988  
if (!state.program.getFileIncludeReasons().get(file.path).some((r) => r.kind === 0 /* RootFile */))  
^  
  
TypeError: Cannot read properties of undefined (reading 'path')  
at tryAddRoot (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:117988:57)  
at /home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:117850:5  
at arrayFrom (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:715:24)  
at getBuildInfo2 (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:117847:21)  
at newProgram.getBuildInfo (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:118148:41)  
at Object.getBuildInfo (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:114560:68)  
at emitBuildInfo (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:107247:28)  
at emitSourceFileOrBundle (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:107213:5)  
at forEachEmittedFile (/home/kxxt/devtools/devtools-frontend/node_modules/typescript/lib/tsc.js:106981:16)  
[... 80 lines cut ...]
```bash
build/landmines.py
build/util/lastchange.py -o build/util/LASTCHANGE
```
16. Generate Release mode ninja files: `gn gen out/Release --args='is_debug = false'`
17. Start the build: `autoninja -C out/Release`

To retry the build when it fails, run `autoninja -C out/Release` again.

# Environment Information 

common:
- OS: Arch Linux riscv
- nodejs version: v20.8.0

sg2042:
- Kernel: `Linux 6.1.42-1-sophgo-05620-gf3e0e4527b17 #1 SMP Wed, 16 Aug 2023 11:11:48 +0000 riscv64 GNU/Linux`

lpi4a:
- Kernel: `Linux  5.10.113-g387b6863253c-dirty #21 SMP PREEMPT Fri Jul 7 03:29:22 SAST 2023 riscv64 GNU/Linux`

# Results on qemu and unmatched

- I cannot reproduce this in qemu-user(8.1.1).
- On unmatched board, I did two whole builds, one succeeded, the other segfault only once and a single retry solves it. This is unlikely the same issue and is probably caused by the memory issue of the board, which has caused lots of random segfaults in other programs on that board.

kxxt, 2023-10-13:
I also can't reproduce this bug on VisionFive boards.

kxxt, 2023-11-16:
This bug still reproduces after sg2042's firmware and kernel is updated to latest version.

kernel version: 6.1.61-1-sophgo-08348-gdb74e759247f (on commit ecf94af96338bc6d8b47547aac14d8783768ca45 (kernel: Adjust the log level of the tick_switch_to_oneshot function))

kxxt, 2023-12-18:
For lpi4a, the bug disappeared after the kernel is upgraded to 6.6.4-1-th1520-g267da9a02021, which makes me believe it's fixed :tada:.

kxxt, 2024-11-01:
This bug also cannot be reproduced on P550 (4 x `eswin,eic770x`).

Note that for reproduction, we need to use an older version of emscription now: https://archriscv-archive.felixc.at/archive/emscripten-3.1.51-1-riscv64.pkg.tar.zst

kxxt, 2025-07-01:
GDB could be attached into the faulty node.js process by running the following script in a tmux session(after a previous run that built `out/Release/gen/front_end/models/breakpoints/breakpoints.prebundle.js`):

```bash
[kxxt@articuno devtools-frontend]$ cat ./reproducer 
#!/bin/bash

tmux neww -d "env gdb -q -ex='set confirm on' -ex=run -ex=quit --args node \
  'node_modules/rollup/dist/bin/rollup' '--config' 'scripts/build/rollup.config.js' '--input' \
  'out/Release/gen/front_end/models/breakpoints/breakpoints.prebundle.js' '--file' '/dev/null' \
  '--configDCHECK'; tmux wait -S node-$1"
tmux wait "node-$1"
```

```bash
 seq 1 100 | parallel -j5 ./reproducer
```

However, the stack is already corrupted when GDB caught the segfault.

RevySR, 2026-02-05:
https://chromium-review.googlesource.com/c/v8/v8/+/7540554

Fixed.