revyos/revyos #28: Rare glibc mutex related crashes on TH1520

felixonmars, 2023-10-12:
```
Cloning into bare repository '/home/felix/packages/gcc12/gcc'...
remote: Enumerating objects: 3967120, done.
remote: Counting objects: 100% (13022/13022), done.
remote: Compressing objects: 100% (7616/7616), done.
remote: Total 3967120 (delta 8594), reused 6934 (delta 5232), pack-reused 3954098
Receiving objects: 100% (3967120/3967120), 1.77 GiB | 7.31 MiB/s, done.
Fatal glibc error: pthread_mutex_lock.c:94 (___pthread_mutex_lock): assertion failed: mutex->__data.__owner == 0
fatal: fetch-pack: invalid index-pack output
```

I am encountering these weird crashes on TH1520, but very rarely. There is no known way to reliably reproduce it but this may indicate something wrong is happening.

Environment: Arch Linux riscv64
gcc 13.2.1
glibc 2.38

Rabenda, 2023-10-15:
firefox: pthread_mutex_lock.c:167: ___pthread_mutex_lock: Assertion `mutex->__data.__owner == 0' failed.

felixonmars, 2023-10-27:
```
npm ERR! git dep preparation failed
npm ERR! signal SIGABRT           
npm ERR! command /usr/bin/node /usr/lib/node_modules/npm/bin/npm-cli.js install --force --cache=/build/.npm --prefer-offline=false --prefer-online=false --offline=false --no-progress --no-save --no-audit --include=dev --include=peer --include=optional --no-package-lock-only --no-dry-run
npm ERR! npm WARN using --force Recommended protections disabled.
npm ERR! Fatal glibc error: pthread_mutex_lock.c:94 (___pthread_mutex_lock): assertion failed: mutex->__data.__owner == 0
```

RevySR, 2023-11-30:
revyos 已修复

closed 2023-11-30 by RevySR