revyos/revyos #32: 尝试在 debian sid 打包 cppgir 失败，因没有 debian gitlab 授权所以只有把编译失败日志转发于此

HougeLangley, 2023-10-19:
```
root@Archlinux-Milk-V:~/cppgir# git submodule update --init
root@Archlinux-Milk-V:~/cppgir# debuild -b -uc -us
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_MONETARY = "zh_CN.UTF-8",
	LC_PAPER = "zh_CN.UTF-8",
	LC_MEASUREMENT = "zh_CN.UTF-8",
	LC_TIME = "zh_CN.UTF-8",
	LC_NUMERIC = "zh_CN.UTF-8",
	LANG = "zh_CN.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_MONETARY = "zh_CN.UTF-8",
	LC_PAPER = "zh_CN.UTF-8",
	LC_MEASUREMENT = "zh_CN.UTF-8",
	LC_TIME = "zh_CN.UTF-8",
	LC_NUMERIC = "zh_CN.UTF-8",
	LANG = "zh_CN.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
 dpkg-buildpackage -us -uc -ui -b
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_MONETARY = "zh_CN.UTF-8",
	LC_PAPER = "zh_CN.UTF-8",
	LC_MEASUREMENT = "zh_CN.UTF-8",
	LC_TIME = "zh_CN.UTF-8",
	LC_NUMERIC = "zh_CN.UTF-8",
	LANG = "zh_CN.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
dpkg-buildpackage: info: source package cppgir
dpkg-buildpackage: info: source version 0~git20231003.e856d76-1
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Nicholas Guriev <guriev-ns@ya.ru>
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_MONETARY = "zh_CN.UTF-8",
	LC_PAPER = "zh_CN.UTF-8",
	LC_MEASUREMENT = "zh_CN.UTF-8",
	LC_TIME = "zh_CN.UTF-8",
	LC_NUMERIC = "zh_CN.UTF-8",
	LANG = "zh_CN.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
 dpkg-source --before-build .
dpkg-buildpackage: info: host architecture riscv64
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_MONETARY = "zh_CN.UTF-8",
	LC_PAPER = "zh_CN.UTF-8",
	LC_MEASUREMENT = "zh_CN.UTF-8",
	LC_TIME = "zh_CN.UTF-8",
	LC_NUMERIC = "zh_CN.UTF-8",
	LANG = "zh_CN.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
[... 418 lines cut ...]
make[3]: *** Waiting for unfinished jobs....
[ 84%] Linking CXX executable example-gio-dbus-client
/usr/bin/cmake -E cmake_link_script CMakeFiles/example-gio-dbus-client.dir/link.txt --verbose=1
/usr/bin/c++ -g -O2 -ffile-prefix-map=/root/cppgir=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -Wl,-z,relro -Wl,-z,now "CMakeFiles/example-gio-dbus-client.dir/examples/gio-dbus-client.cpp.o" -o example-gio-dbus-client  -L/usr/lib/riscv64-linux-gnu -lgio-2.0 -lgobject-2.0 -lglib-2.0 
make[3]: Leaving directory '/root/cppgir/obj-riscv64-linux-gnu'
[ 84%] Built target example-gio-dbus-client
[ 87%] Linking CXX executable example-gio-qt-async
/usr/bin/cmake -E cmake_link_script CMakeFiles/example-gio-qt-async.dir/link.txt --verbose=1
/usr/bin/c++ -g -O2 -ffile-prefix-map=/root/cppgir=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -Wl,-z,relro -Wl,-z,now "CMakeFiles/example-gio-qt-async.dir/example-gio-qt-async_autogen/mocs_compilation.cpp.o" "CMakeFiles/example-gio-qt-async.dir/examples/gio-qt-async.cpp.o" -o example-gio-qt-async  -L/usr/lib/riscv64-linux-gnu -lgio-2.0 -lgobject-2.0 -lglib-2.0 /usr/lib/riscv64-linux-gnu/libQt5Core.so.5.15.8 
make[3]: Leaving directory '/root/cppgir/obj-riscv64-linux-gnu'
[ 87%] Built target example-gio-qt-async
[ 90%] Linking CXX executable example-gio-async
/usr/bin/cmake -E cmake_link_script CMakeFiles/example-gio-async.dir/link.txt --verbose=1
[ 93%] Linking CXX executable example-gio
/usr/bin/cmake -E cmake_link_script CMakeFiles/example-gio.dir/link.txt --verbose=1
/usr/bin/c++ -g -O2 -ffile-prefix-map=/root/cppgir=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -Wl,-z,relro -Wl,-z,now "CMakeFiles/example-gio-async.dir/examples/gio-async.cpp.o" -o example-gio-async  -L/usr/lib/riscv64-linux-gnu -lgio-2.0 -lgobject-2.0 -lglib-2.0 /usr/lib/riscv64-linux-gnu/libboost_fiber.so.1.74.0 /usr/lib/riscv64-linux-gnu/libboost_context.so.1.74.0 /usr/lib/riscv64-linux-gnu/libboost_filesystem.so.1.74.0 
/usr/bin/c++ -g -O2 -ffile-prefix-map=/root/cppgir=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -Wl,-z,relro -Wl,-z,now "CMakeFiles/example-gio.dir/examples/gio.cpp.o" -o example-gio  -L/usr/lib/riscv64-linux-gnu -lgio-2.0 -lgobject-2.0 -lglib-2.0 
make[3]: Leaving directory '/root/cppgir/obj-riscv64-linux-gnu'
[ 93%] Built target example-gio
make[3]: Leaving directory '/root/cppgir/obj-riscv64-linux-gnu'
[ 93%] Built target example-gio-async
make[3]: Leaving directory '/root/cppgir/obj-riscv64-linux-gnu'
make[2]: *** [CMakeFiles/Makefile2:1014: CMakeFiles/example-gst.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
```

HougeLangley, 2023-10-19:
base 来自 revyos

Rabenda, 2023-10-19:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1054195
上游重复bug 

暂时打包了上一个版本0~git20230822.89cf56d-1
已推送

HougeLangley, 2023-10-20:
Done