revyos/revyos #34: Kernel crash when running 6.6

yuzibo, 2023-10-31:
Hi,
This is a long term issue to track upstream kernel supporting licheepi4a.

# 1 get source code from [th1520-linux-kernel](https://github.com/revyos/th1520-linux-kernel)
```
git clone git@github.com:revyos/th1520-linux-kernel.git
```
# 2 Checkout the latest tags(you can always do this)
```
git checkout th1520-v6.6-rc7-v3
```
# 3 Cross compiling on x86
```
make CROSS_COMPILE=riscv64-unknown-linux-gnu- ARCH=riscv revyos_defconfig
make CROSS_COMPILE=riscv64-unknown-linux-gnu- ARCH=riscv INSTALL_MOD_STRIP=1 -j`nproc` bindeb-pkg
```

# 4 Upgrade kernel on licheepi4a
Copying these generated *.deb into licheepi4a
```
dpkg -i *.deb
```
Then update dtb file from `/usr/lib/linux-image-6.6-*/` to `/boot` and edit `/boot/extlinux/extlinux.conf` to ensure FDT entry.

# 5 Reboot 
You need to enable network manually(ifup) if necessary

# 6 Reproducing the crash
I'm still looking for a code snippet on how to reproduce the bug, but before that running debci service will constantly trigger the crash.

## 6.1 Setup debci
```
sudo apt install debci autopkgtest
sudo adduser ${YOUR_USERNAME} debci
sudo env debci_mirror=https://mirror.iscas.ac.cn/debian debci setup
```

## 6.2 Running debci
Let's take `tnseq-transit`  as an example:
```
apt source tnseq-transit
# cd  tnseq-transit dir
autopkgtest --apt-upgrade  --user debci --output-dir /tmp/output-dir .  -- lxc --sudo autopkgtest-unstable-riscv64
```

# 7 The crash log
The specific log is as follows：
```
[ 1367.310761] ---[ end trace 0000000000000000 ]---
[ 1367.322789] Unable to handle kernel access to user memory without uaccess routines at virtual address 000000202020201c
[ 1367.333601] Oops [#5]
[ 1367.335892] Modules linked in: veth nft_masq nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 bridge stp llc nf_tables libcrc32c nfnetlink binfmt_misc rtw88_8723ds rtw88_8723d rtw88_sdio rtw88_core mac80211 libarc4 cfg80211 onboard_usb_hub rfkill mr75203 pwm_thead pwm_fan fuse loop drm dm_mod drm_panel_orientation_quirks dax configfs ip_tables x_tables autofs4 xhci_plat_hcd xhci_hcd usbcore dwc3 usb_common dwmac_generic realtek dwc3_thead i2c_designware_platform i2c_designware_core dwmac_thead fixed
[ 1367.381757] CPU: 2 PID: 1 Comm: systemd Tainted: G      D            6.6.0-rc7+ #5
[ 1367.389341] Hardware name: Sipeed Lichee Pi 4A (DT)
[ 1367.394225] epc : __d_lookup_rcu+0x4a/0xb8
[ 1367.398347]  ra : lookup_fast+0x36/0xf0
[ 1367.402197] epc : ffffffff802c63cc ra : ffffffff802b52de sp : ffffffc800023b10
[ 1367.409429]  gp : ffffffff8199bd10 tp : ffffffd9001ed280 t0 : ffffffd904668d40
[ 1367.416658]  t1 : 0000000400000000 t2 : ffffffff80e00540 s0 : ffffffc800023b40
[ 1367.423888]  s1 : ffffffc800023d10 a0 : ffffffd919545300 a1 : ffffffc800023d20
[ 1367.431117]  a2 : ffffffc800023d54 a3 : 0000000000000001 a4 : 2020202020202020
[ 1367.438347]  a5 : 2020202020202020 a6 : 000b1d3500000000 a7 : 00000000000428a5
[ 1367.445576]  s2 : 0000000000000002 s3 : ffffffd919545300 s4 : 339dc50787ccab19
[ 1367.452804]  s5 : 0000000000200000 s6 : 0000000000700000 s7 : fffffffffffffff6
[ 1367.460033]  s8 : 0000000000000002 s9 : ffffffd9009cf025 s10: ffffffc800023d10
[ 1367.467261]  s11: 000000000000002f t3 : ffffffd919545300 t4 : 0000000358e9ae79
[ 1367.474490]  t5 : ffffffc800023d54 t6 : ffffffd9009cf021
[ 1367.479809] status: 0000000200000120 badaddr: 000000202020201c cause: 000000000000000d
[ 1367.487734] [<ffffffff802c63cc>] __d_lookup_rcu+0x4a/0xb8
[ 1367.493148] [<ffffffff802b8a54>] walk_component+0x2c/0x164
[ 1367.498649] [<ffffffff802b8d30>] link_path_walk.part.0.constprop.0+0x1a4/0x280
[ 1367.505888] [<ffffffff802b96a0>] path_openat+0xaa/0xe34
[ 1367.511127] [<ffffffff802bac02>] do_filp_open+0x88/0x10e
[ 1367.516453] [<ffffffff802a4bd6>] do_sys_openat2+0xa2/0xd4
[ 1367.521867] [<ffffffff802a4f3e>] __riscv_sys_openat+0x66/0x92
[ 1367.527625] [<ffffffff8099f7d2>] do_trap_ecall_u+0xe4/0xf6
[ 1367.533127] [<ffffffff80003cf4>] ret_from_exception+0x0/0x64
[ 1367.538815] Code: 4685 8e2a 8f32 7793 ffe7 e563 00e6 a8a1 639c cbb1 (a703) ffc7
[ 1367.546269] ---[ end trace 0000000000000000 ]---
[ 1367.550964] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[ 1367.558631] SMP: stopping secondary CPUs
[ 1367.562612] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---
```

yuzibo, 2023-11-17:
Now this issue can be attributed to different toolchains. 

if use riscv64-glibc-ubuntu-22.04-gcc-nightly-2023.10.18-nightly.tar.gz from https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2023.10.18. the kernel will not crash for the case(for 8GB version). 

The issue can be reproduced by the toolchain:
```
gcc version 12.2.0 (g2ee5e430018)
Configured with: /src/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv-toolchains --with-sysroot=/opt/riscv-toolchains/sysroot --with-pkgversion=g2ee5e430018 --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=/src/riscv-gnu-toolchain/gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
```

RevySR, 2023-11-30:
https://github.com/revyos/th1520-linux-kernel/tree/th1520-v6.6.3
复测gcc-13/thead-gcc v2.8.0没有问题

closed 2023-11-30 by RevySR