revyos/revyos #37: xfce4-about Segmentation fault

ksco, 2023-11-02:
```
debian@lpi4a:~$ DISPLAY=:0.0 xfce4-about

(xfce4-about:12714): dbind-WARNING **: 13:46:07.913: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
Segmentation fault
debian@lpi4a:~$ cat /etc/revyos-release
BUILD_ID=20231026_181638
BUILD_DATE=20231026
RELEASE_ID=20231026
COMMIT_ID=610d74cb6d5c6a6aaaeeeb6a2dfd3f3e73045aee
RUNNER_ID=6656950486
debian@lpi4a:~$
```

Looks like it's dereferencing a null pointer:

```
Thread 1 "xfce4-about" received signal SIGSEGV, Segmentation fault.
0x0000003ff4077f04 in ?? () from /lib/riscv64-linux-gnu/libGLX_indirect.so.0
(gdb) bt
#0  0x0000003ff4077f04 in ?? () from /lib/riscv64-linux-gnu/libGLX_indirect.so.0
#1  0x0000003ff4074c7c in ?? () from /lib/riscv64-linux-gnu/libGLX_indirect.so.0
#2  0x0000002aaaaaf78a in ?? ()
(gdb) x/10i 0x0000003ff4077f04
=> 0x3ff4077f04:	lw	a5,12(a0)
   0x3ff4077f06:	lw	a3,4(a0)
   0x3ff4077f08:	lw	a4,8(a0)
   0x3ff4077f0a:	lw	a2,0(a0)
   0x3ff4077f0c:	sw	a5,20(s2)
   0x3ff4077f10:	sw	a3,12(s2)
   0x3ff4077f14:	sw	a4,16(s2)
   0x3ff4077f18:	sw	a2,8(s2)
   0x3ff4077f1c:	mv	a0,s1
   0x3ff4077f1e:	sd	s1,24(s2)
(gdb) i r a0
a0             0x0	0
(gdb)
```

ksco, 2023-11-02:
In addition, after running `sudo switch-gl gl4es`, xfce4-about runs normally.

Rabenda, 2023-11-02:
Fixed. repo published.

closed 2023-11-02 by Rabenda