revyos/revyos #39: lxc-net: iptables: can't initialize iptables table `mangle': Table does not exist

weilinfox, 2023-11-27:
## RevyOS 版本

```bash
$ cat /etc/revyos-release 
BUILD_ID=20231009_134626
BUILD_DATE=20231009
COMMIT_ID=f9867c522485046f1965a3101bd4545520803623
RUNNER_ID=6457521686
```

## 问题描述

试图在 RevyOS 中使用 lxc 容器，在启用 lxc-net 时遇到错误， ``lxc-net.service`` 日志如下

```
× lxc-net.service - LXC network bridge setup
     Loaded: loaded (/lib/systemd/system/lxc-net.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Mon 2023-11-27 15:18:49 UTC; 9min ago
       Docs: man:lxc
   Main PID: 385 (code=exited, status=1/FAILURE)
        CPU: 166ms

Nov 27 15:18:48 lpi4a lxc-net[426]: Warning: Extension CHECKSUM revision 0 not supported, missing kernel module?
Nov 27 15:18:48 lpi4a lxc-net[426]: iptables v1.8.9 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Nov 27 15:18:48 lpi4a lxc-net[426]: Perhaps iptables or your kernel needs to be upgraded.
Nov 27 15:18:48 lpi4a lxc-net[385]: Failed to setup lxc-net.
Nov 27 15:18:49 lpi4a lxc-net[446]: Warning: Extension CHECKSUM revision 0 not supported, missing kernel module?
Nov 27 15:18:49 lpi4a lxc-net[446]: iptables v1.8.9 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Nov 27 15:18:49 lpi4a lxc-net[446]: Perhaps iptables or your kernel needs to be upgraded.
Nov 27 15:18:49 lpi4a systemd[1]: lxc-net.service: Main process exited, code=exited, status=1/FAILURE
Nov 27 15:18:49 lpi4a systemd[1]: lxc-net.service: Failed with result 'exit-code'.
Nov 27 15:18:49 lpi4a systemd[1]: Failed to start lxc-net.service - LXC network bridge setup.
```

这看起来是内核没有启用 ``CONFIG_IP_NF_MANGLE`` 导致的

```bash
$ zcat /proc/config.gz |grep -i mangle
# CONFIG_IP_NF_MANGLE is not set
# CONFIG_IP6_NF_MANGLE is not set
```

RevySR, 2023-11-30:
https://github.com/revyos/thead-kernel/commit/16d835b66eb0d61271a990b9159d802c328957bb
enable CONFIG_IP_NF_MANGLE & CONFIG_IP6_NF_MANGLE

内核产物:
https://github.com/revyos/thead-kernel/actions/runs/7050879548
下载 thead-kernel-thead-gcc 相关二进制 安装lpi4a相关内核 重启 复测

weilinfox, 2023-12-01:
坏了，复测，报了一个新的错误

```bash
Dec 01 06:11:47 lpi4a systemd[1]: Starting lxc-net.service - LXC network bridge setup...
░░ Subject: A start job for unit lxc-net.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit lxc-net.service has begun execution.
░░ 
░░ The job identifier is 925.
Dec 01 06:11:47 lpi4a lxc-net[1191]: Warning: Extension CHECKSUM revision 0 not supported, missing kernel module?
Dec 01 06:11:47 lpi4a lxc-net[1191]: iptables: No chain/target/match by that name.
Dec 01 06:11:47 lpi4a lxc-net[1163]: Failed to setup lxc-net.
Dec 01 06:11:47 lpi4a lxc-net[1206]: Warning: Extension CHECKSUM revision 0 not supported, missing kernel module?
Dec 01 06:11:47 lpi4a lxc-net[1206]: iptables: No chain/target/match by that name.
Dec 01 06:11:47 lpi4a systemd[1]: lxc-net.service: Main process exited, code=exited, status=1/FAILURE
```

似乎是这个，但是我不太确定

```bash
$ zcat /proc/config.gz | grep -i CHECKSUM
# CONFIG_NETFILTER_XT_TARGET_CHECKSUM is not set
```

RevySR, 2023-12-02:
https://github.com/revyos/thead-kernel/commit/0325872bee9251df20724b82b0cec45fcaba88e1

已修复 等待合并

closed 2023-12-02 by RevySR