revyos/revyos #43: 256G SD 卡，在进行大 IO 写入时，dmesg 报 IO 错误

atzlinux, 2023-12-07:
fdisk -l
Disk /dev/mmcblk1: 230.47 GiB, 247463936000 bytes, 483328000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x2d0952a9

Device         Boot Start       End   Sectors   Size Id Type
/dev/mmcblk1p1       2048 483327999 483325952 230.5G 83 Linux

[四 12月  7 16:38:23 2023] blk_update_request: I/O error, dev mmcblk1, sector 615457 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[四 12月  7 16:38:24 2023] blk_update_request: I/O error, dev mmcblk1, sector 617981 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[四 12月  7 16:38:24 2023] blk_update_request: I/O error, dev mmcblk1, sector 628021 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[四 12月  7 16:38:27 2023] print_req_error: 8 callbacks suppressed
[四 12月  7 16:38:27 2023] blk_update_request: I/O error, dev mmcblk1, sector 652669 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[四 12月  7 16:38:27 2023] blk_update_request: I/O error, dev mmcblk1, sector 264522047 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[四 12月  7 16:38:27 2023] blk_update_request: I/O error, dev mmcblk1, sector 264523680 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 0

uname -a
Linux lpi4a 5.10.113-lpi4a #2023.10.24.02.18+3e042d29a SMP PREEMPT Tue Oct 24 02:19:10 UTC  riscv64 GNU/Linux

触发场景：
从其它机器用 scp 拷贝 Debian  DVD  iso 文件到荔枝派

atzlinux, 2023-12-07:
hdparm -t /dev/mmcblk1

运行这个命令测试 SD 卡速度时， dmesg 也有这个报错信息

wu-yue-yu, 2023-12-12:
请问TF卡的具体型号有吗？

atzlinux, 2023-12-13:
荔枝派的硬件信息：

https://linux-hardware.org/?probe=50c635dba3

MMC 	usd00-247gb [D2C] 	
	USD00 247GB 	disk 	sdhci_dwcmshc, mmcblk 	detected

从这里看，这个 SD 卡是 MMC 类型，具体型号应该是 USD00 247GB。

dmesg|grep mmc1

[    3.428465] mmc1: SDHCI controller on ffe7090000.sd [ffe7090000.sd] 
using ADMA 64-bit
[    3.788109] mmc1: new ultra high speed SDR104 SDXC card at address 1388
[    3.796813] mmcblk1: mmc1:1388 USD00 230 GiB

不知道是否还有其它在操作系统里面查看 SD 卡型号的命令或者工具？

目前这类报错信息是出现在对 SD 卡进行大 IO 读取时：

[三 12月 13 09:28:24 2023] blk_update_request: I/O error, dev mmcblk1, 
sector 272644 op 0x0:(READ) flags 0x80700 phys_seg 6 prio class 0
[三 12月 13 09:28:24 2023] blk_update_request: I/O error, dev mmcblk1, 
sector 272645 op 0x0:(READ) flags 0x80700 phys_seg 6 prio class 0
[三 12月 13 09:28:24 2023] blk_update_request: I/O error, dev mmcblk1, 
sector 274432 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 0
[三 12月 13 09:28:24 2023] blk_update_request: I/O error, dev mmcblk1, 
sector 275462 op 0x0:(READ) flags 0x80700 phys_seg 10 prio class 0
[三 12月 13 09:28:25 2023] blk_update_request: I/O error, dev mmcblk1, 
sector 276792 op 0x0:(READ) flags 0x80700 phys_seg 6 prio class 0
[三 12月 13 09:28:25 2023] blk_update_request: I/O error, dev mmcblk1, 
sector 276793 op 0x0:(READ) flags 0x80700 phys_seg 6 prio class 0

导致 SD 
卡读取速率非常慢。目前倒是没有发现文件无法读取，或者文件读取错误的现象。

mmc extcsd read /dev/mmcblk1 ，运行这个命令，dmesg 有如下报错：
[三 12月 13 09:45:41 2023] sdhci-dwcmshc ffe7090000.sd: 
__mmc_blk_ioctl_cmd: cmd error -110

我是把这个 SD 卡只分了一个区，使用的  ext4 文件系统。

这个卡 是 MMC 类型的？需要特别分区才能够对齐 文件系统的 block 和 卡的 物理 
block sector 吗？

wu-yue-yu, 2023-12-13:
可以尝试将
light-lpi4a-ref.dts中的
```c
&sdhci0 {
	max-frequency = <198000000>;
	bus-width = <4>;
	pull_up;
	wprtn_ignore;
	status = "okay";
};
```
改为
```c
&sdhci0 {
	max-frequency = <100000000>;
	bus-width = <4>;
	pull_up;
	wprtn_ignore;
	status = "okay";
};
```
降低clock频率看看是否还会出现错误

atzlinux, 2023-12-13:
> 可以尝试将
> light-lpi4a-ref.dts中的
>
> &sdhci0  {
> 	max-frequency  =  <198000000>;
> 	bus-width  =  <4>;
> 	pull_up;
> 	wprtn_ignore;
> 	status  =  "okay";
> };
>
> 改为
>
> &sdhci0  {
> 	max-frequency  =  <100000000>;
> 	bus-width  =  <4>;
> 	pull_up;
> 	wprtn_ignore;
> 	status  =  "okay";
> };
>
> 降低clock频率看看是否还会出现错误

在安装好的系统里面，找不到 light-lpi4a-ref.dts 这个文件。

只有这个目录下，有些 dtb 的二级制文件。
***@***.***:/boot/dtbs/linux-image-5.10.113-lpi4a/thead# ls -l light-lpi4a
light-lpi4a-16gb.dtb          light-lpi4a-dsi0-hdmi.dtb
light-lpi4a-cluster-16gb.dtb  light-lpi4a.dtb
light-lpi4a-cluster.dtb       light-lpi4a-laptop.dtb
light-lpi4a-ddr2G.dtb

请问源代码 light-lpi4a-ref.dts 在哪里？如何才能够编译为 二进制文件？
直接替换

/boot/dtbs/linux-image-5.10.113-lpi4a/thead

  目录下的文件，重启就可以生效了吗？

wu-yue-yu, 2023-12-14:
> […](#)
> -- 肖盛文 xiao sheng wen 微信(wechat)：atzlinux 《铜豌豆 Linux》https://www.atzlinux.com 基于 Debian 的 Linux 中文 桌面 操作系统 GnuPG Public Key: 0x2F338C7DC7909957

是的，在源代码中编译直接替换。板子上也可以用dtc改。dts在**内核**源代码目录中的arch/riscv/boot/dts/thead目录下

atzlinux, 2023-12-14:
遇到几个问题：

1. revyos 系统上 
/boot/dtbs/linux-image-5.10.113-lpi4a/thead/light-lpi4a.dtb 的文件， 使用 
dtc 解开后的 light-lpi4a.dts 文件里面，找不到你说的文字章节。

2. 
https://raw.githubusercontent.com/revyos/thead-kernel/lpi4a/arch/riscv/boot/dts/thead/light-lpi4a.dts 
文件，无法直接用 dtc 生成 light-lpi4a.dtb 文件。

这个 256G 卡是我在华强北 赛格 
市场买的，我有他微信联系方式，不知道厂家是否可以购买一块直接测试？

我本人不是嵌入式开发者，只是想作为一个普通使用者把这个 卡 用起来。

谢谢！

Icenowy, 2023-12-14:
> EHPILYJJKXPAVCNFSM6AAAAABAKWDHGWVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK

> 遇到几个问题：
> 
> 1. revyos 系统上 
> /boot/dtbs/linux-image-5.10.113-lpi4a/thead/light-lpi4a.dtb 的文件， 使用 
> dtc 解开后的 light-lpi4a.dts 文件里面，找不到你说的文字章节。

此处章节对应原始源码，反编译后请寻找 ***@***.*** 而不是 &sdhci0 （因为后者在编译过程中已经丢失了），并修改其中的
max-frequency 。

atzlinux, 2023-12-15:
谢谢指导！

在 2023/12/14 20:11, Icenowy Zheng 写道:
> 在 2023-12-14星期四的 01:40 -0800，atzlinux写道：
> > 遇到几个问题：
> >
> > 1. revyos 系统上
> > /boot/dtbs/linux-image-5.10.113-lpi4a/thead/light-lpi4a.dtb 的文件， 
> 使用
> > dtc 解开后的 light-lpi4a.dts 文件里面，找不到你说的文字章节。
>
> 此处章节对应原始源码，反编译后请寻找 ***@***.*** 而不是 &sdhci0 
> （因为后者在编译过程中已经丢失了），并修改其中的
> max-frequency 。
light-lpi4a.dts 文件里面是如下这个章节，

                 ***@***.*** {
                         compatible = "snps,dwcmshc-sdhci";
                         reg = <0xff 0xe7090000 0x00 0x10000>;
                         interrupt-parent = <0x16>;
                         interrupts = <0x40>;
                         interrupt-names = "sdhci0irq";
                         clocks = <0x3f>;
                         clock-names = "core";
                         max-frequency = <0x5f5e100>;
                         bus-width = <0x04>;
                         pull_up;
                         wprtn_ignore;
                         status = "okay";
                 };

将 max-frequency = <0xbcd3d80>; 修改为 max-frequency = <0x5f5e100>;

这里进行了一个 16 进制到 10 进制的转换哈：

***@***.***:~/f3# echo $((0x5f5e100))
100000000

重新编译
dtc -I dts light-lpi4a.dts -O dtb -o light-lpi4a.dtb

替换 /boot/dtbs/linux-image-5.10.113-lpi4a/thead/light-lpi4a.dtb 
文件后，重启荔枝派.

测试：

hdparm -t /dev/mmcblk1

/dev/mmcblk1:
  Timing buffered disk reads: 134 MB in  3.00 seconds =  44.62 MB/sec
> Message ID: ***@***.***> 

F3 工具测试：
https://gitee.com/mirrors_AltraMayor/f3.git

***@***.***:~/f3# ./f3write /mnt/1
F3 write 8.0
Copyright (C) 2010 Digirati Internet LTDA.
This is free software; see the source for copying conditions.

Free space: 200.67 GB
Creating file 1.h2w ... OK!
..................
Creating file 15.h2w ... 7.45% -- 11.33 MB/s -- 3:24:01
OK!
Creating file 16.h2w ... 7.48% -- 11.03 MB/s -- 3:24:13
OK!
Creating file 17.h2w ... OK!
Creating file 18.h2w ... 8.57% -- 10.78 MB/s -- 3:21:31^C

./f3read /mnt/1
F3 read 8.0
Copyright (C) 2010 Digirati Internet LTDA.
This is free software; see the source for copying conditions.

                   SECTORS      ok/corrupted/changed/overwritten
Validating file 1.h2w ... 5.14% -- 37.83 MB/s -- 7:18
2097152/        0/      0/      0
Validating file 2.h2w ... 6.67% -- 38.06 MB/s -- 7:12

dmesg 均没有报错。

问题基本解决。

加上其它测试，发现 SD 卡 IO 写速率约 16MB/s, IO 读速率约 44 MB/s .

不知道这个 IO 值，是否已经达到了 256GB SD 卡的最大值呢？
有人反馈测试 SD 卡，读速率可以到 80MB/s.

-- 
肖盛文 xiao sheng wen
微信(wechat)：atzlinux
《铜豌豆 Linux》https://www.atzlinux.com
基于 Debian 的 Linux 中文 桌面 操作系统
GnuPG Public Key: 0x2F338C7DC7909957

Icenowy, 2023-12-15:
这个修改的本质是拿速度换稳定性，不过稳定性达不到的话速度本来也无从谈起……

atzlinux, 2023-12-15:
SD 卡，SD 卡内核驱动，设备树，这些是不是有一个规范标准来确定最大频率？

大家如果都按规范标准来设置，就应该不会出问题。

原有的设备树文件，

连接 mmc 的 ***@***.*** 的设置 198000000 的 max-frequency 
就没有问题，mmc 正常使用。
而连接 sd 卡的 ***@***.*** 的设置 198000000 的 max-frequency 
就有问题，修改降低到 100000000 才正常，降低一半。

198000000 有没有超过 sd 卡的最大理论值呢？SD 卡最大理论值是多少呢？

在 2023/12/15 10:17, Icenowy Zheng 写道:
>
> 这个修改的本质是拿速度换稳定性，不过稳定性达不到的话速度本来也无从谈起……
>

Icenowy, 2023-12-15:
> SD 卡，SD 卡内核驱动，设备树，这些是不是有一个规范标准来确定最大频率？
> 
> 大家如果都按规范标准来设置，就应该不会出问题。
> 
> 原有的设备树文件，
> 
> 连接 mmc 的 ***@***.*** 的设置 198000000 的 max-frequency 
> 就没有问题，mmc 正常使用。

eMMC 目前最高的标准是 HS400 ， 等价 400MHz

> 而连接 sd 卡的 ***@***.*** 的设置 198000000 的 max-frequency 
> 就有问题，修改降低到 100000000 才正常，降低一半。
> 
> 198000000 有没有超过 sd 卡的最大理论值呢？SD 卡最大理论值是多少呢？

SD 卡本身的最高理论值是 208MHz ，对应 SDR194 标准；但特定硬件不一定能够跑到 208MHz
，这个频率对于各个硬件的信号完整性有很高的要求，信号完整性达不到就只能降速。

RevySR, 2023-12-15:
请不要把bug以及相关解决抄送到无关lug 会给revyos和相关lug用户带来额外困扰

RevySR, 2023-12-15:
这并不是一个泛用的方案 问题也并没有完全解决

atzlinux, 2023-12-15:
szlug 和 gzlug 是 深圳和广州的 Linux User Group， 我经常在里面和大家交流 
Linux 相关内容。

这次的问题，也是和 Linux 
相关的，是和大家共享分析下问题分析解决过程，也是技术交流。

目前没有在 lug 邮件列表里面收到 lug 的用户反馈有困惑。 如果只是 lug 
交流范围的讨论，在 lug 里面交流更好。

revyos 是基于 Linux 的，在 lug 里面讨论 revyos，也没有超出 lug 
的技术讨论范围。

revyos 作为中科院做的一个开源操作系统，难道不欢迎大家讨论吗？

这个问题，到底是硬件厂家那边的问题，还是 revyos 
这边的问题，目前我也确实是不清楚的，
我也没有说这个问题是哪边的责任，其实可以出来澄清下。

weilinfox, 2023-12-15:
相同的 I/O error 在初代 8+8 的荔枝派 4A 就见过，不光光是 SD 卡， U 盘也见过。感觉更像个硬件问题，而不应当由 RevyOS 来解决。

RevySR, 2024-03-27:
内核已合并相关修改 问题已解决

closed 2024-03-27 by RevySR