revyos/revyos #50: Issues running K8S/K3S on Licheepi4a / Revyos -> missing cgroups cpu.weight directory

pascal71, 2023-12-30:
Good evening all,

I am experimenting with running K3S on RISCV64. So far my Vision Five 2 boards work very well with ArchLinux (kernel 5.15) with it. On my 3 Licheepi4a I have less luck.

In the k3s logs I can see the following, it looks like there's some issue with cgroup configuration:

: rpc error: code = Unknown desc = failed to create containerd task: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error setting cgroup config for procHooks process: openat2 /sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod9f91904f_1f07_43e4_b4ee_46531661561a.slice/cri-containerd-fbdb1604a5977d9427590d85feb539358f3fc6721bd9cbd3c6176c31bfb89414.scope/cpu.weight: no such file or directory: unknown\"" pod="kube-system/metrics-server-7c55d89d5d-brcrd" podUID=9f91904f-1f07-43e4-b4ee-46531661561a
Dec 30 18:21:58 lpi4a-n03 k3s[11534]: E1230 18:21:58.798979   11534 qos_container_manager_linux.go:377] "Failed to update QoS cgroup configuration" err="openat2 /sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/cpu.weight: no such file or directory"
Dec 30 18:21:59 lpi4a-n03 k3s[11534]: E1230 18:21:59.415286   11534 cgroup_manager_linux.go:475] cgroup manager.Set failed: openat2 /sys/fs/cgroup/kubepods.slice/kubepods-besteffort.slice/kubepods-besteffort-podd0dd4fd6_f850_4af2_8f0e_c2c4d8bc3030.slice/cpu.weight: no such file or directory
Dec 30 18:22:00 lpi4a-n03 k3s[11534]: E1230 18:22:00.103644   11534 qos_container_manager_linux.go:377] "Failed to update QoS cgroup configuration" err="openat2 /sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/cpu.weight: no such file or directory"
Dec 30 18:22:00 lpi4a-n03 k3s[11534]: E1230 18:22:00.391385   11534 qos_container_manager_linux.go:377] "Failed to update QoS cgroup configuration" err="openat2 /sys/fs/cgroup/kubepods.slice/kubepods-besteffort.slice/cpu.weight: no such file or directory"
Dec 30 18:22:00 lpi4a-n03 k3s[11534]: I1230 18:22:00.392663   11534 qos_container_manager_linux.go:140] "Failed to reserve QoS requests" err="openat2 /sys/fs/cgroup/kubepods.slice/kubepods-besteffort.slice/cpu.weight: no such file or directory"
Dec 30 18:22:00 lpi4a-n03 k3s[11534]: E1230 18:22:00.696887   11534 cgroup_manager_linux.go:475] cgroup manager.Set failed: openat2 /sys/fs/cgroup/kubepods.slice/kubepods-besteffort.slice/kubepods-besteffort-pod797f79ba_feff_4015_bd3b_09908f226223.slice/cpu.weight: no such file or directory

Is this fixable and how? And.. I did see there are alt 6.x kernels in the tree, but I can't get the configured. How do I configure/set these up?

If more info is needed, kindly let me know. My next step is to get K8S/KIND running on the Licheepi4a and Vison Five 2s

Thanks in advance.

Kind regards,

   Pascal van Dam

RevySR, 2023-12-30:
Need an operation step to reproduce it.

Used to verify the modified kernel config.

pascal71, 2023-12-30:
Thank you for your quick reply:

Installing 6.x kernel just made my Licheepi4a go nuts. :(

But..

as root:

# Download
#
rm -f k3s-riscv64.gz*
sudo apt install wget -y
wget https://github.com/CARV-ICS-FORTH/k3s/releases/download/20230721/k3s-riscv64.gz.aa
wget https://github.com/CARV-ICS-FORTH/k3s/releases/download/20230721/k3s-riscv64.gz.ab
wget https://github.com/CARV-ICS-FORTH/k3s/releases/download/20230721/k3s-riscv64.gz.ac
cat k3s-riscv64.gz.* | gunzip > /usr/local/bin/k3s
chmod +x /usr/local/bin/k3s

# Install
curl -sfL https://get.k3s.io > k3s-install.sh
INSTALL_K3S_SKIP_DOWNLOAD="true" bash -x k3s-install.sh

then check:

journalctl -u k3s -l 

and filter for cgroup messages

If more info is needed, kindly let me know. 

Kind regards,

   Pascal

pascal71, 2023-12-30:
Corrected the licheepi4a; can boot again; it was missing a device tree for kernel 6.x.

opsiff, 2023-12-31:
Hello All:

I found this commit [1],maybe is helpful,and use [2] to check.
[1]:https://github.com/revyos/thead-kernel/commit/fba465199e44e2ff995ba72f245d97d1d61cbc92
[2]:https://github.com/moby/moby/blob/master/contrib/check-config.sh

Regards.

pascal71, 2023-12-31:
Currently comparing the kernel config of the vision52-n06  
5.15.2-cwt-5.10.3-1 with the one on the licheepi4a.
It might be an issue in the kernel version itself.

Small question; how can I setup revyos for using one of the (more)  
stable linux 6.x kernels? It appears I am missing the correct device  
tree files?

Thanks in advance.

Kind regards,

    Pascal

opsiff, 2023-12-31:
> Currently comparing the kernel config of the vision52-n06 5.15.2-cwt-5.10.3-1 with the one on the licheepi4a. It might be an issue in the kernel version itself. Small question; how can I setup [...]

Sorry... I am not have it now,maybe the link [1] helpful.

[1]: https://wiki.sipeed.com/hardware/en/lichee/th1520/lpi4a/7_develop_mainline.html

Regrads.

pascal71, 2024-02-18:
Good morning all,

Good news;

pascal@lpi4a-n02:~$ sudo k3s kubectl get nodes
NAME        STATUS   ROLES                  AGE   VERSION
lpi4a-n02   Ready    control-plane,master   17h   v1.27.3+k3s-9d376dfb-dirty

It appears I can get K3s now to work with the Licheepi4a's.

Used kernel:

ii  linux-config-6.1:riscv64               6.1.27-1                         riscv64      Debian kernel configurations for Linux 6.1
ii  linux-headers-5.10.113-lpi4a           2024.01.31.14.11+384c5e9e2       riscv64      Linux kernel headers for 5.10.113-lpi4a on riscv64
ii  linux-image-5.10.113-lpi4a             2024.01.31.14.11+384c5e9e2       riscv64      Linux kernel, version 5.10.113-lpi4a

pascal@lpi4a-n02:~$ uname -a
Linux lpi4a-n02 5.10.113-lpi4a #2024.01.31.14.11+384c5e9e2 SMP PREEMPT Wed Jan 31 14:11:31 UTC  riscv64 GNU/Linux

Keep up the good work.

I will try to report back when I have my 3 lpi4a's integrated as workernodes in the Vision5-2 cluster.

Thansk!

RevySR, 2024-03-01:
> Good morning all,
>
> Good news;
>
> pascal@lpi4a-n02:~$ sudo k3s kubectl get nodes NAME STATUS ROLES AGE VERSION lpi4a-n02 Ready control-plane,master 17h v1.27.3+k3s-9d376dfb-dirty
>
> It appears I can get [...]

If you only use k3s, you can use lpi4amain image.

https://mirror.iscas.ac.cn/revyos/extra/images/lpi4amain/20240127/

RevySR, 2024-03-18:
resovled.