revyos/revyos #58: 修改启动顺序，spl跳转opensbi卡住

CoderWangJL, 2024-02-29:
```
void (*opensbi)(ulong hart, void *dtb, struct fw_dynamic_info *p);
opensbi = (void (*)(ulong, void *, struct fw_dynamic_info*))LIGHT_FW_ADDR;
//仿照uboot跳转opensbi  完善opensbi_info结构体
opensbi_info.magic = FW_DYNAMIC_INFO_MAGIC_VALUE;
opensbi_info.version = 0x1;
opensbi_info.next_addr = CONFIG_SPL_TEXT_BASE;
opensbi_info.next_mode = FW_DYNAMIC_INFO_NEXT_MODE_S;
opensbi_info.options = 0;
opensbi_info.boot_hart = 0;
//iopmp设置，不设置emmc状态获取出错
light_iopmp_config();
//初始化emmc，并从emmc中加载opensbi到ddr 0x0位置
load_opensbi_from_emmc();
//跳转opensbi执行
opensbi(0, LIGHT_DTB_ADDR, &opensbi_info);
```
上述跳转opensbi之后，开发板卡死，没有输出，想问一下为什么会这样，这样跳转之前需要改什么pc状态寄存器之类的嘛？还有iopmp为什么会影响emmc的初始化；还有一个问题，emmc 在spl阶段的初始化除了和iopmp相关之外，还和MMC的SDMA开关相关，我将该开关关闭之后才能初始化成功，这是为什么呢

CoderWangJL, 2024-02-29:
补充一下：上述代码是在board/light/spl.c 的light_board_init_r中加的，现象就是跳转之后没有任何报错和输出，然后卡在那

RevySR, 2024-03-01:
补充原始来源：https://maixhub.com/discussion/100172