revyos/revyos #59: What to do when cpu doesn't have v-extension

predator2k, 2024-02-29:
I want to try this on C906fd. Is this possible or not supported at all? What part should I modify (kernel or just the compiling stuff)? 

Looking forward to your reply.

nekorouter, 2024-03-01:
What hardware/chip do you mean "C906fd"?

predator2k, 2024-03-01:
It's https://github.com/T-head-Semi/openc906 running on FPGA actually

nekorouter, 2024-03-01:
The implementation of this soft core has no v-extension support, unfortunately you need to realize this part on your own.

predator2k, 2024-03-01:
So this OS heavily depends on V-extension?

nekorouter, 2024-03-01:
Actually no, we have both kernel and software that enabled vector support or not, but revyos's kernel and software are mainly work on real hardware (TH1520), we are not sure if those can run on openc906...

predator2k, 2024-03-01:
So the issues are more of soc stuff(IO, device tree, etc.), and OS's V-extension dependence can probably be turned-off without too much impact?

nekorouter, 2024-03-01:
That should be correct, though we didn't tested on openc906 so no guarantee to work.

predator2k, 2024-03-01:
ok

closed 2024-03-01 by predator2k