revyos/revyos #62: gcc10.4.0的向量指令扩展报错

liu200010023555, 2024-04-02:
在荔枝派4a上，gcc10.4.0执行一个普通的数组相加测试，已经通过官网的wiki使用了优化的工具链
·  输出如下：
sipeed@lpi4a:~/Desktop/test_rvv$ gcc -march=rv64gcv0p7_zfh_xtheadc -mabi=lp64d test1.cpp -o test1.out
Assembler messages:
Error: x ISA extension `xtheadc' must be set with the versions
/tmp/ccGEi3Nz.s:135: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/ccGEi3Nz.s:155: Error: unrecognized opcode `vse.v v2,0(a5)'
/tmp/ccGEi3Nz.s:215: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/ccGEi3Nz.s:235: Error: unrecognized opcode `vse.v v2,0(a5)'
/tmp/ccGEi3Nz.s:252: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/ccGEi3Nz.s:257: Error: unrecognized opcode `vse.v v2,0(a4)'
/tmp/ccGEi3Nz.s:274: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/ccGEi3Nz.s:279: Error: unrecognized opcode `vse.v v2,0(a4)'
/tmp/ccGEi3Nz.s:333: Error: unrecognized opcode `vle.v v4,0(a4)'
/tmp/ccGEi3Nz.s:338: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/ccGEi3Nz.s:360: Error: unrecognized opcode `vse.v v2,0(a5)'
/tmp/ccGEi3Nz.s:391: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/ccGEi3Nz.s:396: Error: unrecognized opcode `vse.v v2,0(a4)'
/tmp/ccGEi3Nz.s:450: Error: unrecognized opcode `vle.v v2,0(a4)'
/tmp/ccGEi3Nz.s:453: Error: unrecognized opcode `vse.v v2,0(a5)'
/tmp/ccGEi3Nz.s:639: Error: unrecognized opcode `th.lrw a5,a5,a4,2', extension `xtheadmemidx' required
·  然后：
sipeed@lpi4a:~/Desktop/test_rvv$ gcc -march=rv64gcv1p0 -mabi=lp64d test1.cpp -o test1.out
sipeed@lpi4a:~/Desktop/test_rvv$ ./test1.out 
111
Illegal instruction
只打印了向量化测试前的111，后面直接Illegal instruction，
源码段：
 for (; vl = 2; n -= vl) {
        printf("%d :", vl);
        vb = vle32_v_i32m2(b, vl);
        vc = vle32_v_i32m2(c, vl);
        va = vadd_vv_i32m2(vb, vc, vl);
        vse32_v_i32m2(a, va, vl);

        for (int i = 0; i < 10; i++) printf("%d ", p[i]);
        printf("\n");

        a += vl;
        b += vl;
        c += vl;

    }
在其他的risc_v平台上执行没问题，在荔枝派4a上有问题，希望尽快解答

RevySR, 2024-04-02:
不支持v1p0

4a平台支持的是v0p7/xtheadvector 使用v1p0必非法

liu200010023555, 2024-04-02:
这是v0p7的输出：
***@***.***:~/Desktop/test_rvv$ gcc -march=rv64gcv0p7 -mabi=lp64d test1.cpp -o test1.out
/tmp/cc8ykPcv.s: Assembler messages:
/tmp/cc8ykPcv.s:177: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/cc8ykPcv.s:200: Error: unrecognized opcode `vse.v v2,0(a5)'
/tmp/cc8ykPcv.s:284: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/cc8ykPcv.s:307: Error: unrecognized opcode `vse.v v2,0(a5)'
/tmp/cc8ykPcv.s:330: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/cc8ykPcv.s:335: Error: unrecognized opcode `vse.v v2,0(a4)'
/tmp/cc8ykPcv.s:358: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/cc8ykPcv.s:363: Error: unrecognized opcode `vse.v v2,0(a4)'
/tmp/cc8ykPcv.s:438: Error: unrecognized opcode `vle.v v4,0(a4)'
/tmp/cc8ykPcv.s:443: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/cc8ykPcv.s:468: Error: unrecognized opcode `vse.v v2,0(a5)'
/tmp/cc8ykPcv.s:511: Error: unrecognized opcode `vle.v v2,0(a5)'
/tmp/cc8ykPcv.s:516: Error: unrecognized opcode `vse.v v2,0(a4)'
/tmp/cc8ykPcv.s:591: Error: unrecognized opcode `vle.v v2,0(a4)'
/tmp/cc8ykPcv.s:594: Error: unrecognized opcode `vse.v v2,0(a5)'

请问我应该使用什么编译指令呢？

≒小梦想°?? 
***@***.***

抄送: "≒小梦想°♡♥ ***@***.******@***.***&gt;;

RevySR, 2024-04-02:
https://revyos.github.io/docs/build/debian/enable_optimization_gcc/

参考这个

你这个是没有使用优化仓库的binutils导致的

liu200010023555, 2024-04-03:
rvv没报错了，但是我尝试使用内联汇编时，报
test0.cpp: In function ‘int main()’:
test0.cpp:21:5: error: impossible constraint in ‘asm’
   21 |     __asm__ __volatile__ (
      |     ^~~~~~~

这个是不支持内联汇编吗？还是说要进行处理？

≒小梦想°?? 
***@***.***

抄送: "≒小梦想°♡♥ ***@***.******@***.***&gt;;

 
https://revyos.github.io/docs/build/debian/enable_optimization_gcc/

RevySR, 2024-04-03:
请发完整测试用例

liu200010023555, 2024-04-04:
#include <stdio.h&gt;

int main() {
    int x[10] = {1, 2, 3, 4, 5, 6, 7, 8, 9, 0};
    int y[10] = {0, 9, 8, 7, 6, 5, 4, 3, 2, 1};
    int z[10];

    for (int i = 0; i < 10; i++) {
        asm (
                "add %0, %1, %2\n"
                : "=r"(z[i])
                : "r"(x[i]), "r"(y[i])
                );
    }

    for (int i = 0; i < 10; i++) {
        printf("z[%d] = %d\n", i, z[i]);
    }

    return 0;
}这个没问题(lzm) ***@***.***:/home/sipeed/Desktop/test_rvv# gcc -march=rv64gcv0p7_zfh_xtheadc -mabi=lp64d -mtune=c910 test0.cpp -o test0.out (lzm) ***@***.***:/home/sipeed/Desktop/test_rvv# ./test0.out  z[0] = 1 z[1] = 11 z[2] = 11 z[3] = 11 z[4] = 11 z[5] = 11 z[6] = 11 z[7] = 11 z[8] = 11 z[9] = 1但是想向量化时
#include <stdio.h&gt;

int main() {
    // 定义并初始化输入数组
    int x[10] = {1, 2, 3, 4, 5, 6, 7, 8, 9, 0};
    int y[10] = {0, 9, 8, 7, 6, 5, 4, 3, 2, 1};
    int z[10]; // 结果数组

    // 使用RISC-V向量指令进行向量加法
    asm  (
            "vsetvli t0, %2, e32;m8\n\t"       // 设置向量长度和数据类型
            "vlw.v v1, (%0)\n\t"               // 加载x到向量寄存器v1
            "vlw.v v2, (%1)\n\t"               // 加载y到向量寄存器v2
            "vadd.vv v3, v1, v2\n\t"           // 将v1和v2相加，结果存储在v3
            "vsw.v v3, (%3)\n\t"               // 将结果从v3存储到z
            : // 无输出
            : "r"(x), "r"(y), "r"(10), "r"(z)  // 输入
            : "t0", "v0", "v1", "v2", "v3"     // 被破坏的寄存器列表
            );

    // 打印结果
    printf("Result:\n");
    for (int i = 0; i < 10; i++) {
        printf("%d ", z[i]);
    }
    printf("\n");

    return 0;
}
报(lzm) ***@***.***:/home/sipeed/Desktop/test_rvv# gcc -march=rv64gcv -mabi=lp64d -o test0_vec.out test0_vec.cpp test0_vec.cpp: Assembler messages: test0_vec.cpp:10: Error: unrecognized opcode `vlw.v v1,(a5)', extension `v' or `zvlsseg' or `zvamo' required test0_vec.cpp:11: Error: unrecognized opcode `vlw.v v2,(a4)', extension `v' or `zvlsseg' or `zvamo' required test0_vec.cpp:13: Error: unrecognized opcode `vsw.v v3,(a2)', extension `v' or `zvlsseg' or `zvamo' required  

≒小梦想°?? 
***@***.***

抄送: "≒小梦想°♡♥ ***@***.******@***.***&gt;;

liu200010023555, 2024-04-04:
我按照您给的参考进行gcc工具链配置，在荔枝派4a的开发板上配置好了，但是在刷了openkylin的板子上

***@***.***:/home/openkylin# sudo apt update &amp;&amp; sudo apt upgrade -y
命中:1 http://ppa.build.openkylin.top/kylinsoft/anything/openkylin nile InRelease
命中:2 http://archive.build.openkylin.top/openkylin nile InRelease        
命中:3 http://archive.build.openkylin.top/openkylin nile-security InRelease
命中:4 http://archive.build.openkylin.top/openkylin nile-updates InRelease
获取:5 https://mirror.iscas.ac.cn/revyos/revyos-c910v revyos-c910v InRelease [5,798 B]
命中:6 http://archive.build.openkylin.top/openkylin nile-proposed InRelease
错误:5 https://mirror.iscas.ac.cn/revyos/revyos-c910v revyos-c910v InRelease
  由于没有公钥，无法验证下列签名： NO_PUBKEY 2FB3A9E77911527E
正在读取软件包列表... 完成
W: GPG 错误：https://mirror.iscas.ac.cn/revyos/revyos-c910v revyos-c910v InRelease: 由于没有公钥，无法验证下列签名： NO_PUBKEY 2FB3A9E77911527E
E: 仓库 “https://mirror.iscas.ac.cn/revyos/revyos-c910v revyos-c910v InRelease” 没有数字签名。
N: 无法安全地用该源进行更新，所以默认禁用该源。
N: 参见 apt-secure(8) 手册以了解仓库创建和用户配置方面的细节。
W: 目标 Packages (main/binary-riscv64/Packages) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:2 中被配置了多次
W: 目标 Packages (main/binary-all/Packages) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:2 中被配置了多次
W: 目标 Translations (main/i18n/Translation-zh_CN) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:2 中被配置了多次
W: 目标 Translations (main/i18n/Translation-zh) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:2 中被配置了多次
W: 目标 Translations (main/i18n/Translation-en) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:2 中被配置了多次
W: 目标 Packages (main/binary-riscv64/Packages) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:3 中被配置了多次
W: 目标 Packages (main/binary-all/Packages) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:3 中被配置了多次
W: 目标 Translations (main/i18n/Translation-zh_CN) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:3 中被配置了多次
W: 目标 Translations (main/i18n/Translation-zh) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:3 中被配置了多次
W: 目标 Translations (main/i18n/Translation-en) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:3 中被配置了多次
W: 目标 Packages (main/binary-riscv64/Packages) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:4 中被配置了多次
W: 目标 Packages (main/binary-all/Packages) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:4 中被配置了多次
W: 目标 Translations (main/i18n/Translation-zh_CN) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:4 中被配置了多次
W: 目标 Translations (main/i18n/Translation-zh) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:4 中被配置了多次
W: 目标 Translations (main/i18n/Translation-en) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:4 中被配置了多次
W: 目标 Packages (main/binary-riscv64/Packages) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:5 中被配置了多次
W: 目标 Packages (main/binary-all/Packages) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:5 中被配置了多次
W: 目标 Translations (main/i18n/Translation-zh_CN) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:5 中被配置了多次
W: 目标 Translations (main/i18n/Translation-zh) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:5 中被配置了多次
W: 目标 Translations (main/i18n/Translation-en) 在 /etc/apt/sources.list:1 和 /etc/apt/sources.list:5 中被配置了多次

然后继续

***@***.***:/home/openkylin# sudo apt install -y build-essential gcc-13 g++-13

得到的gcc不是10版本的

***@***.***:/home/openkylin# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-linux-gnu/12/lto-wrapper
Target: riscv64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Openkylin 12.2.0-ok7' --with-bugurl=file:///usr/share/doc/gcc-12/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-12 --program-prefix=riscv64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --disable-multilib --with-arch=rv64gc --with-abi=lp64d --enable-checking=release --build=riscv64-linux-gnu --host=riscv64-linux-gnu --target=riscv64-linux-gnu
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.2.0 (Openkylin 12.2.0-ok7) 

请问问题出在哪？

≒小梦想°?? 
***@***.***

抄送: "≒小梦想°♡♥ ***@***.******@***.***&gt;;

RevySR, 2024-04-04:
> #include <stdio.h&gt; int main() { int x[10] = {1, 2, 3, 4, 5, 6, 7, 8, 9, 0}; int y[10] = {0, 9, 8, 7, 6, 5, 4, 3, 2, 1}; int z[10]; for (int i = 0; i < 10; i++) { asm ( "add %0, %1, %2\n" : [...]

"gcc -march=rv64gcv0p7_zfh_xtheadc" 这就是v0p7向量化
"-march=rv64gcv" 不支持 这是v1p0

willliu, 2024-04-11:
您好想咨询一个困扰半年的问题（刚看到这里，终于抱到大腿了～），Lichee PI 4A RVV编译报错，已经完成上面说明。

【操作系统】uname -a

Linux lpi4a 5.10.113+ #1 SMP PREEMPT Thu Dec 7 10:37:05 CST 2023 riscv64 GNU/Linux

(因为apt update会造成Console 4A重启死机，已经多次和Sipeed沟通，结论是不能执行apt update)

【编译器】gcc-10 -v （已经切换源：https://revyos.github.io/docs/build/debian/enable_optimization_gcc/）

Using built-in specs.
COLLECT_GCC=gcc-10
COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-linux-gnu/10/lto-wrapper
Target: riscv64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 10.4.0-8revyos2.4' --with-bugurl=file:///usr/share/doc/gcc-10/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-10 --program-prefix=riscv64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libsanitizer --disable-libquadmath --disable-libquadmath-support --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --disable-multilib --with-arch=rv64gcv0p7_zfh_xtheadc --with-abi=lp64d --enable-checking=release --build=riscv64-linux-gnu --host=riscv64-linux-gnu --target=riscv64-linux-gnu
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.4.0 (Debian 10.4.0-8revyos2.4) 

【编译指令】g++-10 -march=rv64gcv0p7 add.cpp -o add

【错误结果】

/tmp/cc1uxnpe.s: Assembler messages:
/tmp/cc1uxnpe.s:336: Error: unrecognized opcode `vle.v v1,0(a5)'
/tmp/cc1uxnpe.s:349: Error: unrecognized opcode `vse.v v1,0(a5)'
/tmp/cc1uxnpe.s:433: Error: unrecognized opcode `vle.v v1,0(a5)'
/tmp/cc1uxnpe.s:446: Error: unrecognized opcode `vse.v v1,0(a5)'
/tmp/cc1uxnpe.s:469: Error: unrecognized opcode `vle.v v1,0(a5)'
/tmp/cc1uxnpe.s:474: Error: unrecognized opcode `vse.v v1,0(a4)'
/tmp/cc1uxnpe.s:497: Error: unrecognized opcode `vle.v v1,0(a5)'
/tmp/cc1uxnpe.s:502: Error: unrecognized opcode `vse.v v1,0(a4)'
/tmp/cc1uxnpe.s:577: Error: unrecognized opcode `vle.v v2,0(a4)'
/tmp/cc1uxnpe.s:582: Error: unrecognized opcode `vle.v v1,0(a5)'
/tmp/cc1uxnpe.s:597: Error: unrecognized opcode `vse.v v1,0(a5)'
/tmp/cc1uxnpe.s:640: Error: unrecognized opcode `vle.v v1,0(a5)'
/tmp/cc1uxnpe.s:645: Error: unrecognized opcode `vse.v v1,0(a4)'
/tmp/cc1uxnpe.s:720: Error: unrecognized opcode `vle.v v1,0(a4)'
/tmp/cc1uxnpe.s:723: Error: unrecognized opcode `vse.v v1,0(a5)'

【程序】向量加法 add.cpp，这个玄铁C处理器论文：https://blog.csdn.net/weixin_47100480/article/details/137565230

#include <riscv_vector.h>
#include <stdio.h>
#include <math.h>

#define ARRAY_SIZE 11

float a_array[ARRAY_SIZE] = {
1.18869953, 1.55298864, -0.17365574, -1.86193886,
-1.52391526, -0.36566814, 0.70753702, 0.73992422,
-0.13493693, 1.09563677,1.03797902
};

float b_array[ARRAY_SIZE] = {
1.19655525, 0.23393777, -0.11629651, -0.54508896,
-1.2424749, -1.54835913, 0.86935212, 0.12946646,
0.81831905, -0.42723697, -0.89793257
};

float c_array_ref[ARRAY_SIZE];
float c_array_vec[ARRAY_SIZE];

void add(const float *a, const float *b, float *c, size_t length)
{
 for (int i = 0; i < length; i++) {
 c[i] = a[i] + b[i];
 }
}

void add_vec(const float *a, const float *b, float *c, size_t length)
{
 while (length > 0) {
 size_t vl = vsetvl_e32m1(length); 

 printf("rvv version is: %d\n", __riscv_v);
 printf("vector length is %d, length is %lu\n", vl, length);
 vfloat32m1_t va = vle32_v_f32m1(a, vl);  
 vfloat32m1_t vb = vle32_v_f32m1(b, vl); 
 vfloat32m1_t vc = vfadd_vv_f32m1(va, vb, vl); 
  vse32_v_f32m1(c, vc, vl); 

  a += vl;
  b += vl;
  c += vl;
  length -= vl;
  }
}

int main()
{
 printf("Test add function by rvv0.7.1 intrinsic \n");
 add(a_array, b_array, c_array_ref, ARRAY_SIZE);
 add_vec(a_array, b_array, c_array_vec, ARRAY_SIZE);

 for (int i = 0; i < ARRAY_SIZE; i++) {
 if (fabsf(c_array_ref[i] - c_array_vec[i]) > 1e-6) {
 printf("index[%d] failed, %f=!%f\n", i, c_array_ref[i], c_array_vec[i]);
 } else {
  printf("index[%d] successed, %f=%f\n", i, c_array_ref[i], c_array_vec[i]);
 }
 }
 return 0;
}

已经困扰半年，花费大量时间，跪谢🙇🙇

RevySR, 2024-04-11:
> https://blog.csdn.net/weixin_47100480/article/details/137565230

https://github.com/revyos/revyos/issues/62#issuecomment-2031374275

注意binutils的版本

最近我们在做迁移从v0p7到xtheadvector的工作 完事之后可能会使用简单一些
之前不重视这块 因为需要开扩展仓库 现在在做无需扩展仓库也可以使用的方法

willliu, 2024-04-12:
非常感谢，能不能提供给我一个测试版本，我非常愿意帮你们测试～

willliu, 2024-04-12:
好的，十分感谢，之后上线了非常愿意测试～

RevySR, 2024-06-01:
> 好的，十分感谢，之后上线了非常愿意测试～

现在最新版本已经提供主线gcc14 可以使用march=rv64gc_zfh_xtheadvector来编译了 不需要扩展仓库
扩展仓库会在之后的系统中重新提供

closed 2024-06-05 by RevySR