revyos/revyos #63: LicheePi 4A Console 在使用 lxd init 的时候，提示缺少内核模块

HougeLangley, 2024-04-25:


nekorouter, 2024-04-26:
https://github.com/k3s-io/k3s/issues/7118
https://github.com/NixOS/nixpkgs/issues/242853

@RevySR  Looks like kernel need include those modules? 
```
CONFIG_NETFILTER_XT_MARK
CONFIG_NETFILTER_XT_MATCH_COMMENT
CONFIG_NETFILTER_XT_MATCH_MULTIPORT
```

KevinMX, 2026-01-08:
Those kmods are already added on latest releases and LXD should be able to init (passed on my side).

```
mx@revyos-lpi4a:~$ zgrep -E 'XT_MARK|XT_MATCH_COMMENT|XT_MATCH_MULTIPORT' /proc/config.gz 
CONFIG_NETFILTER_XT_MARK=m
CONFIG_NETFILTER_XT_MATCH_COMMENT=m
CONFIG_NETFILTER_XT_MATCH_MULTIPORT=m
mx@revyos-lpi4a:~$ uname -a
Linux revyos-lpi4a 6.6.119-th1520 #2025.12.24.18.43+2a44b04d6 SMP Wed Dec 24 19:01:45 UTC 2025 riscv64 GNU/Linux
```

closed 2026-01-08 by KevinMX