revyos/revyos #69: revyos-meles chromium

armisis, 2024-06-20:
Chromium crashes at launch unable to use. Running on a Milk-V Meles board. Revyos-Meles.

FIFCC, 2024-06-21:
Did you start it through the terminal or by clicking the chromium icon on the desktop?

If you start it through the command line, please enter
```
chromium --no-sandbox --use-gl=egl 
```

armisis, 2024-06-21:
I clicked the Icon on the desktop. I did install firefox and got it working
fine. I can see this OS is in development and respect that, i see right how
it is quite sluggish for some reason, I will keep gong though things and
make reports as needed. I will try the command line next and let you know
the results.

armisis, 2024-06-22:
Ok, treid it on the command line, sending this email via firefox which is
working, but it took 15 min to load gmail and ther eis a huge lag typing.
The icon launch still fails, the command line launch asdirected failed and
here is teh result error messages: chromium --nosand-box --use-gl=egl
471.774123 (10808) PVR:(Error): PVRDRICreateScreen: Failed to open render
device node [ pvrscreen_impl.c:218 ]
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to authenticate
[10808:10808:0622/173545.626012:ERROR:gpu_init.cc(523)] Passthrough is not
supported, GL is egl, ANGLE is
[0622/173546.985387:WARNING:process_reader_linux.cc(147)] no stack mapping
[0622/173546.988496:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq: No such file or
directory (2)
[0622/173546.988737:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq: No such file or
directory (2)
[0622/173547.176331:WARNING:process_reader_linux.cc(147)] no stack mapping
[0622/173547.177737:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq: No such file or
directory (2)
[0622/173547.177938:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq: No such file or
directory (2)
[0622/173547.607208:WARNING:process_reader_linux.cc(147)] no stack mapping
[0622/173547.609629:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq: No such file or
directory (2)
[0622/173547.609874:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq: No such file or
directory (2)
[10762:10762:0622/173548.414373:ERROR:gpu_process_host.cc(991)] GPU process
exited unexpectedly: exit_code=11
476.997547 (10865) PVR:(Error): PVRDRICreateScreen: Failed to open render
device node [ pvrscreen_impl.c:218 ]
libEGL warning: DRI2: failed to create dri screen
[0622/173549.483533:WARNING:process_reader_linux.cc(147)] no stack mapping
[0622/173549.490434:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq: No such file or
directory (2)
[0622/173549.490700:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq: No such file or
directory (2)
libEGL warning: DRI2: failed to authenticate
[0622/173549.735801:WARNING:process_reader_linux.cc(147)] no stack mapping
[0622/173549.737472:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq: No such file or
directory (2)
[0622/173549.737743:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq: No such file or
directory (2)
[0622/173551.054150:ERROR:elf_dynamic_array_reader.h(64)] tag not found
[0622/173551.117979:WARNING:process_reader_linux.cc(147)] no stack mapping
[0622/173551.120346:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq: No such file or
directory (2)
[0622/173551.120638:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq: No such file or
directory (2)
[10865:10865:0622/173551.093728:ERROR:gpu_init.cc(523)] Passthrough is not
supported, GL is egl, ANGLE is
[10809:10815:0622/173551.426445:ERROR:broker_posix.cc(40)] Recvmsg error:
Connection reset by peer (104)
[10809:10815:0622/173551.428956:ERROR:broker_posix.cc(110)] Error sending
sync broker message: Broken pipe (32)
Segmentation fault
***@***.***:~$ [0622/173553.308397:WARNING:process_reader_linux.cc(147)]
no stack mapping
[0622/173553.310219:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq: No such file or
directory (2)
[0622/173553.310406:ERROR:file_io_posix.cc(144)] open
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq: No such file or
directory (2)

FIFCC, 2024-06-25:
> Ok, treid it on the command line, sending this email via firefox which is working, but it took 15 min to load gmail and ther eis a huge lag typing. The icon launch still fails, the command line [...]

It is chromium --no-sandbox --use-gl=egl,instead of chromium --nosand-box --use-gl=egl,Please try again.

armisis, 2024-06-25:
Ok will try that. Thank you!
Dave

FIFCC, 2024-07-11:
The latest version 0601 image failed to reproduce this problem. The new version of the kernel has been pushed. It can be updated after apt update&&apt upgrade

> Chromium 在启动时崩溃无法使用。在 Milk-V Meles 主板上运行。Revyos-Meles。

FIFCC, 2024-08-01:
Running Chromium on the 0720 image, both clicking the desktop icon and starting it from the command line are successful. This issue is closed

closed 2024-08-01 by FIFCC

armisis, 2024-08-01:
Awesome, thanks for the update!
David D Stanton