revyos/revyos #70: LPi4a: video decoder broken (aon: tx_done timeout)

cgutman, 2024-07-01:
Using RevyOS image 20240601 on a LicheePi 4a 16GB:
```
Linux revyos-lpi4a 5.10.113-th1520 #2024.05.31.16.27+2dec14431 SMP PREEMPT Fri May 31 16:28:40 UTC  riscv64
```

The video decoder fails to load at boot:
```
[   29.085764] hantrodec: module is from the staging directory, the quality is unknown, you have been warned.
[   29.176642] enter hantrodec_init
[   29.687714] light-aon aon: tx_done timeout
[   29.691946] light-aon aon: RPC send msg failed: -110
[   29.697047]  vdec: failed to power up resource 1 ret -5
```

It sems like there's some major issue with `aon` . `dmesg` output is full of errors related to it.
```
# dmesg | grep aon
[    5.066415] light_aon_probe:virtual_log_mem=0x(____ptrval____), phy base=0x33600000,size:2097152
[    8.272887] light-aon aon: RPC send msg timeout
[    8.277488] light-aon-reg aon:appcpu_dvdd: failed to get Voltages!
[    8.290024] light-aon-reg aon:appcpu_dvdd: Failed to initialize regulatorappcpu_dvdd
[    8.297822] light-aon-reg: probe of aon:appcpu_dvdd failed with error -22
[    8.816908] light-aon aon: tx_done timeout
[    8.821071] light-aon aon: RPC send msg failed: -110
[    8.826082] light-aon-reg aon:appcpu_dvddm: failed to get Voltages!
[    8.838782] light-aon-reg aon:appcpu_dvddm: Failed to initialize regulatorappcpu_dvddm
[    8.846752] light-aon-reg: probe of aon:appcpu_dvddm failed with error -22
[    9.360880] light-aon aon: tx_done timeout
[    9.365044] light-aon aon: RPC send msg failed: -110
[    9.370049] soc_dvdd18_aon: failed to enable: -EIO
[    9.374885] light-aon-reg aon:soc_dvdd18_aon: Failed to initialize regulatorsoc_dvdd18_aon
[    9.383198] light-aon-reg: probe of aon:soc_dvdd18_aon failed with error -5
[    9.904881] light-aon aon: tx_done timeout
[    9.909045] light-aon aon: RPC send msg failed: -110
[    9.918968] light-aon-reg aon:soc_avdd33_usb3: Failed to initialize regulatorsoc_avdd33_usb3
[    9.927448] light-aon-reg: probe of aon:soc_avdd33_usb3 failed with error -5
[   10.448906] light-aon aon: tx_done timeout
[   10.453066] light-aon aon: RPC send msg failed: -110
[   10.458070] soc_dvdd08_aon: failed to enable: -EIO
[   10.462901] light-aon-reg aon:soc_dvdd08_aon: Failed to initialize regulatorsoc_dvdd08_aon
[   10.471216] light-aon-reg: probe of aon:soc_dvdd08_aon failed with error -5
[   10.992885] light-aon aon: tx_done timeout
[   10.997045] light-aon aon: RPC send msg failed: -110
[   11.006884] light-aon-reg aon:soc_dvdd08_ddr: Failed to initialize regulatorsoc_dvdd08_ddr
[   11.015196] light-aon-reg: probe of aon:soc_dvdd08_ddr failed with error -5
[   11.536880] light-aon aon: tx_done timeout
[   11.541041] light-aon aon: RPC send msg failed: -110
[   11.550962] light-aon-reg aon:soc_vdd_ddr_1v8: Failed to initialize regulatorsoc_vdd_ddr_1v8
[   11.559455] light-aon-reg: probe of aon:soc_vdd_ddr_1v8 failed with error -5
[   12.080905] light-aon aon: tx_done timeout
[   12.085064] light-aon aon: RPC send msg failed: -110
[   12.094988] light-aon-reg aon:soc_vdd_ddr_1v1: Failed to initialize regulatorsoc_vdd_ddr_1v1
[   12.103472] light-aon-reg: probe of aon:soc_vdd_ddr_1v1 failed with error -5
[   12.624893] light-aon aon: tx_done timeout
[   12.629034] light-aon aon: RPC send msg failed: -110
[   12.638956] light-aon-reg aon:soc_vdd_ddr_0v6: Failed to initialize regulatorsoc_vdd_ddr_0v6
[   12.647441] light-aon-reg: probe of aon:soc_vdd_ddr_0v6 failed with error -5
[   13.168880] light-aon aon: tx_done timeout
[   13.173040] light-aon aon: RPC send msg failed: -110
[   13.182789] light-aon-reg aon:soc_dvdd18_ap: Failed to initialize regulatorsoc_dvdd18_ap
[   13.190927] light-aon-reg: probe of aon:soc_dvdd18_ap failed with error -5
[   13.712925] light-aon aon: tx_done timeout
[   13.717084] light-aon aon: RPC send msg failed: -110
[   13.726834] light-aon-reg aon:soc_dvdd08_ap: Failed to initialize regulatorsoc_dvdd08_ap
[   13.734974] light-aon-reg: probe of aon:soc_dvdd08_ap failed with error -5
[   14.256876] light-aon aon: tx_done timeout
[   14.261034] light-aon aon: RPC send msg failed: -110
[   14.271392] light-aon-reg aon:soc_avdd08_mipi_hdmi: Failed to initialize regulatorsoc_avdd08_mipi_hdmi
[   14.280744] light-aon-reg: probe of aon:soc_avdd08_mipi_hdmi failed with error -5
[   14.800878] light-aon aon: tx_done timeout
[   14.805037] light-aon aon: RPC send msg failed: -110
[   14.815396] light-aon-reg aon:soc_avdd18_mipi_hdmi: Failed to initialize regulatorsoc_avdd18_mipi_hdmi
[   14.824750] light-aon-reg: probe of aon:soc_avdd18_mipi_hdmi failed with error -5
[   15.344909] light-aon aon: tx_done timeout
[   15.349072] light-aon aon: RPC send msg failed: -110
[   15.888874] light-aon aon: tx_done timeout
[   15.893034] light-aon aon: RPC send msg failed: -110
[   16.400880] light-aon aon: tx_done timeout
[   16.405033] light-aon aon: RPC send msg failed: -110
[   16.944876] light-aon aon: tx_done timeout
[   16.949035] light-aon aon: RPC send msg failed: -110
[   17.456878] light-aon aon: tx_done timeout
[   17.461042] light-aon aon: RPC send msg failed: -110
[   17.476138] light-aon-reg aon:soc_dovdd18_scan: Failed to initialize regulatorsoc_dovdd18_scan
[   17.484802] light-aon-reg: probe of aon:soc_dovdd18_scan failed with error -22
[   18.000864] light-aon aon: tx_done timeout
[   18.005008] light-aon aon: RPC send msg failed: -110
[   18.020003] light-aon-reg aon:soc_dvdd12_scan: Failed to initialize regulatorsoc_dvdd12_scan
[   18.028492] light-aon-reg: probe of aon:soc_dvdd12_scan failed with error -22
[   18.544877] light-aon aon: tx_done timeout
[   18.549039] light-aon aon: RPC send msg failed: -110
[   18.564298] light-aon-reg aon:soc_avdd28_scan_en: Failed to initialize regulatorsoc_avdd28_scan_en
[   18.573305] light-aon-reg: probe of aon:soc_avdd28_scan_en failed with error -22
[   18.582617] thead,light-aon-test aon:light-aon-test: Successfully registered
[   18.788879] light-pm soc:aon_suspend_ctrl: Light power management control sys successfully registered
[   19.312884] light-aon aon: tx_done timeout
[   19.317051] light-aon aon: RPC send msg failed: -110
[   19.322057] light-event soc:light-event: failed to set aon reservemem
[   19.328532] light-event soc:light-event: set aon reservemem failed!
[   19.856910] light-aon aon: tx_done timeout
[   19.861053] light-aon aon: RPC send msg failed: -110
[   29.687714] light-aon aon: tx_done timeout
[   29.691946] light-aon aon: RPC send msg failed: -110
[   30.352946] light-aon aon: tx_done timeout
[   30.357124] light-aon aon: RPC send msg failed: -110
[   34.001030] light-aon aon: tx_done timeout
[   34.005528] light-aon aon: RPC send msg failed: -110
[   34.544987] light-aon aon: tx_done timeout
[   34.549240] light-aon aon: RPC send msg failed: -110
[   35.088988] light-aon aon: tx_done timeout
[   35.093402] light-aon aon: RPC send msg failed: -110
[   35.633134] light-aon aon: tx_done timeout
[   35.637512] light-aon aon: RPC send msg failed: -110
[   36.176960] light-aon aon: tx_done timeout
[   36.181215] light-aon aon: RPC send msg failed: -110
[   36.720974] light-aon aon: tx_done timeout
[   36.725364] light-aon aon: RPC send msg failed: -110
[   37.264992] light-aon aon: tx_done timeout
[   37.269194] light-aon aon: RPC send msg failed: -110
[   41.361045] light-aon aon: tx_done timeout
[   41.366660] light-aon aon: RPC send msg failed: -110
```

RevySR, 2024-07-01:
Need update uboot

cgutman, 2024-07-03:
Thanks! Flashing the correct U-boot version from your mirror solved the problem.

I used the Sipeed Mega drive mirror, which had the wrong U-Boot version. I filed https://github.com/sipeed/LicheePi4A/issues/38 to put the proper U-boot version in Mega.

closed 2024-07-03 by cgutman