revyos/revyos #75: Latest revyos version for hdmi+dsi?

ihtamnetu, 2024-09-05:
Could you advice what revyos version to choose, if i'm going to use both hdmi and dsi displays. 
Is there a way to know what kernel commit id was used to build it?

FIFCC, 2024-09-05:
Currently, the devices supported by RevyOS only support HDMI display. What device would you like to use?
You can check the relevant information by using the command `cat /etc/revyos-release`.

ihtamnetu, 2024-09-06:
I downloaded LPI4A_FULL_2024011 image version from MEGA cloud. Both HDMI and DSI (jadard) displays seemed to work fine with that distro. I want to add a kernel driver for another DSI panel. Can i somehow obtain the kernel commit id(thead-kernel), that was used for that version?

FIFCC, 2024-09-09:
I am not sure about the version of the image downloaded from mega cloud. You need to provide me with the information of `cat/etc/revyos-release`.
I recommend downloading the latest image from [ISCASmirror station](https://mirror.iscas.ac.cn/revyos/extra/images/lpi4a/) .
The latest image version is 0720, and the kernel commit ID of thead-kernel is d8f77de535567ec01408ba45f9b82c40b5b9ef5d

ihtamnetu, 2024-09-11:
BUILD_ID=20231009_134626
BUILD_DATE=20231009
COMMIT_ID=f9867c522485046f1965a3101bd4545520803623
RUNNER_ID=6457521686

FIFCC, 2024-09-24:
This image file is not directly released by revyos. It has been modified. 
Please refer to the reply in this issue for the commit of the latest version of the image. 
Please try to use the latest version of the image and download it from [this directory](https://mirror.iscas.ac.cn/revyos/extra/images/lpi4a/) . 
This issue is closed.

closed 2024-09-24 by FIFCC