revyos/revyos #77: `sd` instruction stalls when write to a `PROT_WRITE | PROT_EXEC` page

Kingfish404, 2024-10-12:
Minimal proof of concept `sd_stall.c`:
```c
#include <stdio.h>
#include <sys/mman.h>
#include <unistd.h>

int main() {
  volatile long page_size = sysconf(_SC_PAGESIZE);
  void *mem = mmap(NULL, page_size, PROT_READ | PROT_WRITE | PROT_EXEC,
                   MAP_PRIVATE | MAP_ANONYMOUS, -1, 0);
  if (mem == MAP_FAILED) {
    perror("mmap");
    return 1;
  }
  ((char *)mem)[0] = 42;
  volatile char data = ((char *)mem)[0];
  if (mprotect(mem, page_size, PROT_WRITE | PROT_EXEC) == -1) {
    perror("mprotect");
    munmap(mem, page_size);
    return 1;
  }
  ((char *)mem)[0] = 42;
  munmap(mem, page_size);
  return 0;
}
```

```shell
$ uname -a
Linux lpi4a 5.10.113-lpi4a #2024.02.29.14.50+3eee2bf4d SMP PREEMPT Thu Feb 29 14:51:04 UTC  riscv64 GNU/Linux
$ gcc sd_stall.c -o sd_stall && ./sd_stall 
# no return nor segment fault
```
The `gdb` debug result:

```
┌───────────────────────────────────────────────────────────────────────────────────────────────────┐
│    0x2aaaaaa818 <main+102>         mv      a1,a5                                                  │
│    0x2aaaaaa81a <main+104>         ld      a0,-24(s0)                                             │
│    0x2aaaaaa81e <main+108>         jal     0x2aaaaaa6c0 <mprotect@plt>                            │
│    0x2aaaaaa822 <main+112>         mv      a5,a0                                                  │
│    0x2aaaaaa824 <main+114>         mv      a4,a5                                                  │
│    0x2aaaaaa826 <main+116>         li      a5,-1                                                  │
│    0x2aaaaaa828 <main+118>         bne     a4,a5,0x2aaaaaa84a <main+152>                          │
│    0x2aaaaaa82c <main+122>         auipc   a0,0x0                                                 │
│    0x2aaaaaa830 <main+126>         add     a0,a0,84                                               │
│    0x2aaaaaa834 <main+130>         jal     0x2aaaaaa6b0 <perror@plt>                              │
│    0x2aaaaaa838 <main+134>         ld      a5,-32(s0)                                             │
│    0x2aaaaaa83c <main+138>         mv      a1,a5                                                  │
│    0x2aaaaaa83e <main+140>         ld      a0,-24(s0)                                             │
│    0x2aaaaaa842 <main+144>         jal     0x2aaaaaa690 <munmap@plt>                              │
│    0x2aaaaaa846 <main+148>         li      a5,1                                                   │
│    0x2aaaaaa848 <main+150>         j       0x2aaaaaa866 <main+180>                                │
│    0x2aaaaaa84a <main+152>         ld      a5,-24(s0)                                             │
│    0x2aaaaaa84e <main+156>         li      a4,42                                                  │
│  > 0x2aaaaaa852 <main+160>         sb      a4,0(a5)                                               │
│    0x2aaaaaa856 <main+164>         ld      a5,-32(s0)                                             │
│    0x2aaaaaa85a <main+168>         mv      a1,a5                                                  │
│    0x2aaaaaa85c <main+170>         ld      a0,-24(s0)                                             │
│    0x2aaaaaa860 <main+174>         jal     0x2aaaaaa690 <munmap@plt>                              │
│    0x2aaaaaa864 <main+178>         li      a5,0                                                   │
│    0x2aaaaaa866 <main+180>         mv      a0,a5                                                  │
│    0x2aaaaaa868 <main+182>         ld      ra,40(sp)                                              │
│    0x2aaaaaa86a <main+184>         ld      s0,32(sp)                                              │
│    0x2aaaaaa86c <main+186>         add     sp,sp,48                                               │
│    0x2aaaaaa86e <main+188>         ret                                                            │
│    0x2aaaaaa870 <_IO_stdin_used>   nop                                                            │
│    0x2aaaaaa872 <_IO_stdin_used+2> c.slli64        zero                                           │
│    0x2aaaaaa874                    unimp                                                          │
│    0x2aaaaaa876                    unimp                                                          │
│    0x2aaaaaa878                    lui     s10,0x1b                                               │
│    0x2aaaaaa87a                    c.lui   zero,0xffff8                                           │
│    0x2aaaaaa87c                    unimp                                                          │
│    0x2aaaaaa87e                    unimp                                                          │
└───────────────────────────────────────────────────────────────────────────────────────────────────┘
multi-thre Thread 0x3ff7fc5c60 In: main                                       L??   PC: 0x2aaaaaa852 
(gdb) r
Starting program: /home/jinyu/Developer/code/sd_stall
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/riscv64-linux-gnu/libthread_db.so.1".
^C
  Program received signal SIGINT, Interrupt.
0x0000002aaaaaa852 in main ()
(gdb) s 1000
Single stepping until exit from function main,
which has no line number information.
^C
  Program received signal SIGINT, Interrupt.
0x0000002aaaaaa852 in main ()
(gdb) 
```

No stall at `qemu-riscv64` and other architecture e.g. `x86-64`.

RevySR, 2024-10-15:
```
debian@revyos-lpi4a:~$ cat a.c
#include <stdio.h>
#include <sys/mman.h>
#include <unistd.h>

int main() {
  volatile long page_size = sysconf(_SC_PAGESIZE);
  void *mem = mmap(NULL, page_size, PROT_READ | PROT_WRITE | PROT_EXEC,
                   MAP_PRIVATE | MAP_ANONYMOUS, -1, 0);
  if (mem == MAP_FAILED) {
    perror("mmap");
    return 1;
  }
  ((char *)mem)[0] = 42;
  volatile char data = ((char *)mem)[0];
  if (mprotect(mem, page_size, PROT_WRITE | PROT_EXEC) == -1) {
    perror("mprotect");
    munmap(mem, page_size);
    return 1;
  }
  ((char *)mem)[0] = 42;
  munmap(mem, page_size);
  return 0;
}
debian@revyos-lpi4a:~$ gcc a.c
debian@revyos-lpi4a:~$ ./a.out
debian@revyos-lpi4a:~$ uname -a
Linux revyos-lpi4a 6.6.52-th1520 #2024.09.21.18.08+d1fe4fea5 SMP Sat Sep 21 19:54:44 UTC 2024 riscv64 GNU/Linux
```

It could be a problem with the 5.10.113 kernel

Kingfish404, 2024-10-15:
How could I get and replace with `6.6.52-th1520` kernel ?

Below document is still using `linux-5.10.113`.
[revyos/Build/手动替换内核(Manually replace the kernel).md at dcef25095501d00fdf85eeb629676f9b74030d67 · revyos/revyos](https://github.com/revyos/revyos/blob/dcef25095501d00fdf85eeb629676f9b74030d67/Build/%E6%89%8B%E5%8A%A8%E6%9B%BF%E6%8D%A2%E5%86%85%E6%A0%B8%28Manually%20replace%20the%20kernel%29.md)

RevySR, 2024-10-16:
> How could I get and replace with `6.6.52-th1520` kernel ?
>
> Below document is still using `linux-5.10.113`. [revyos/Build/手动替换内核(Manually replace the kernel).md at [...]

https://github.com/revyos/thead-u-boot/actions/runs/10965384532
new uboot sdk2.0.2 - lpi4amain

https://github.com/revyos/th1520-linux-kernel/actions/runs/10970502815
only lpi4a works.

https://github.com/revyos/mkimg-th1520/actions/runs/10973771394
new lpi4a image.

It's in testing. There's a bug in the codec kernel module.

RevySR, 2024-10-16:
> How could I get and replace with `6.6.52-th1520` kernel ?
>
> Below document is still using `linux-5.10.113`. [revyos/Build/手动替换内核(Manually replace the kernel).md at [...]

This article lacks necessary steps (required by the new version of the kernel).
The firmware needs to be replaced.

closed 2025-01-15 by RevySR