revyos/revyos #78: Milk-V meles 无法打开摄像头

discodyer, 2024-10-29:
## 运行环境

- MilkV meles v1.3
- RevyOS 0721 https://mirror.iscas.ac.cn/revyos/extra/images/meles/20240720/
- USB摄像头 https://item.taobao.com/item.htm?_u=t2nf4lkkf03f&id=557946541067 和另一款 https://item.taobao.com/item.htm?_u=t2nf4lkka99b&id=568221988838

## 问题描述

最简单的OpenCV 打开摄像头程序无法正常捕捉摄像头画面

```python
import cv2
import numpy as np

cap = cv2.VideoCapture(0)

if not cap.isOpened():
    print("Can not open Camera!")
    exit()

while True:
    ret, frame = cap.read()
    if not ret:
        print("Cannot read frame!")
        break

    cv2.imshow('Camera', frame)

    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

cap.release()
cv2.destroyAllWindows()
```

运行后会卡住，并不会显示画面

[image]

此时按下ctrl+C终止程序，会出现下面的信息

> (python3:2474): GStreamer-CRITICAL **: 17:11:44.952:
> Trying to dispose element pipeline0, but it is in PAUSED instead of the NULL state.
> You need to explicitly set elements to the NULL state before
> dropping the final reference, to allow them to clean up.
> This problem may also be caused by a refcounting bug in the
> application or some element.
> 
> [ WARN:0@38.452] global ./modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
> 
> (python3:2474): GStreamer-CRITICAL **: 17:11:44.953:
> Trying to dispose element videoconvert0, but it is in PLAYING instead of the NULL state.
> You need to explicitly set elements to the NULL state before
> dropping the final reference, to allow them to clean up.
> This problem may also be caused by a refcounting bug in the
> application or some element.
> 
> [ WARN:0@38.453] global ./modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
> 
> (python3:2474): GStreamer-CRITICAL **: 17:11:44.954:
> Trying to dispose element appsink0, but it is in READY instead of the NULL state.
> You need to explicitly set elements to the NULL state before
> dropping the final reference, to allow them to clean up.
> This problem may also be caused by a refcounting bug in the
> application or some element.
> 
> [ WARN:0@38.455] global ./modules/videoio/src/cap_v4l.cpp (902) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
> Traceback (most recent call last):
>   File "/home/debian/cam_test/test.py", line 4, in <module>
>     cap = cv2.VideoCapture(0)
[... 83 lines cut ...]
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> Cannot connect to server socket err = No such file or directory
> Cannot connect to server request channel
> jack server is not running or cannot be started
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
> ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
> ALSA lib pcm_a52.c:1001:(_snd_pcm_a52_open) a52 is only for playback
> ALSA lib confmisc.c:1369:(snd_func_refer) Unable to find definition 'cards.0.pcm.iec958.0:CARD=0,AES0=6,AES1=130,AES2=0,AES3=2'
> ALSA lib conf.c:5180:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
> ALSA lib conf.c:5703:(snd_config_expand) Evaluate error: No such file or directory
> ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2 CARD 0}
> ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
> ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
> ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
> ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
> Cannot connect to server socket err = No such file or directory
> Cannot connect to server request channel
> jack server is not running or cannot be started
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> 
> (guvcview:2497): dbind-WARNING **: 17:13:18.206: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files

discodyer, 2025-01-22:
lsusb可以读取到设备信息

[image]

[image]

当打开cheese的时候，dmesg提示这个

[image]

lazyparser, 2025-01-22:
感谢反馈！

@RevySR 看看assign？

discodyer, 2025-03-19:
update: 最新版本20250123测试也有同样的问题

## 补充复现步骤：

首先下载并烧录全新镜像 https://mirror.iscas.ac.cn/revyos/extra/images/meles/20250123/

执行`sudo apt install python3-opencv`安装opencv

接入摄像头并运行上述python脚本

```python
import cv2
import numpy as np

cap = cv2.VideoCapture(0)

if not cap.isOpened():
    print("Can not open Camera!")
    exit()

while True:
    ret, frame = cap.read()
    if not ret:
        print("Cannot read frame!")
        break

    cv2.imshow('Camera', frame)

    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

cap.release()
cv2.destroyAllWindows()
```

问题和上面描述的相同

discodyer, 2025-03-24:
## 摄像头兼容列表

| 品牌 | 型号 | 分辨率 | lsusb | 兼容性 | 商品页面 |
| - | - | - | - | - | - |
| RMONCAM/林柏视 | HD900 | 1280x720 | ID 0601:0405 Jazz Hipster Corp. Rmoncam A2 720P | ❌ | [TB](https://item.taobao.com/item.htm?_u=t2nf4lkkf03f&id=557946541067) |
| S-YUE/晟悦 | HD1313 | 1280x720 |  | ❌ | [TB](https://item.taobao.com/item.htm?_u=t2nf4lkka99b&id=568221988838) |
| Lenovo ThinkPlus | WL22A | 1280x960 | ID 2f9d:0024 Generic WL24A | ✅ |  |
| aoni奥尼 | C90 | 1920x1080 | ID 1bcf:2284 Sunplus Innovation Technology Inc. Full HD webcam | ✅ |  |
| HIKVISION | DS-E11 | 1280x720 |  | ❓ | 有其他问题，不推荐使用 |