revyos/revyos #82: licheepi 4a 无法打开默认浏览器

FIFCC, 2024-12-15:
设备：Licheepi 4a 16+128

问题描述：点击web browser后无法打开默认浏览器，弹窗显示

`Failed to execute default web browser`

kagura114, 2024-12-28:
根本原因应该是命令行下打不开 chromium （包括直接运行和 `xdg-open` 等），firefox 没有这个问题，设为默认浏览器就好了，出问题的位置 log:`[30412:30412:1228/204136.963863:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=11`

kagura114, 2024-12-29:
应该是缺少 --no-sandbox 的问题

FIFCC, 2025-01-13:
经过测试，在最新镜像[20250110](https://mirror.iscas.ac.cn/revyos/extra/images/lpi4a/20250110/)中，点击桌面浏览器图标可正常打开默认浏览器chromium。

通过命令行启动chromium时请使用下面的命令`chromium -no-sandbox -use-gl=egl`

[image]

closed 2025-03-12 by KevinMX