revyos/revyos #87: lp4a amd电脑烧录镜像无法启动系统

nihui, 2024-12-20:
按boot插线识别识别

根据文档
fastboot usb 0 方式烧录镜像后无法启动系统

[image]

[image]

nihui, 2024-12-20:
在intel 12400电脑上，按boot插线可以识别到，并且烧录后可以启动系统
已排除线材原因

FIFCC, 2024-12-20:
在R5 5600G+b550m配置的pc上成功复现
[image]

nihui, 2024-12-20:
amd机器用windows也能识别设备，仅linux下有问题

kagura114, 2024-12-21:
> amd机器用windows也能识别设备，仅linux下有问题

请问在进入 fastboot 后，重启电脑主机，是否可以刷写。我这边测试是可以的（5600G,6.12.4），如果是的话可能是 Linux 下某个驱动的问题。

nihui, 2024-12-22:
> > amd机器用windows也能识别设备，仅linux下有问题
> 
> 请问在进入 fastboot 后，重启电脑主机，是否可以刷写。我这边测试是可以的（5600G,6.12.4），如果是的话可能是 Linux 下某个驱动的问题。

按boot插线，然后再重启系统，无法识别

[image]

kagura114, 2024-12-23:
> > > amd机器用windows也能识别设备，仅linux下有问题
> > 
> > 
> > 请问在进入 fastboot 后，重启电脑主机，是否可以刷写。我这边测试是可以的（5600G,6.12.4），如果是的话可能是 Linux 下某个驱动的问题。
> 
> 按boot插线，然后再重启系统，无法识别
> 
> ![图片](https://private-user-images.githubusercontent.com/171016/397962430-5952040b-aa5a-43c6-8169-8c36b8611ba7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ5MzUyMDAsIm5iZiI6MTczNDkzNDkwMCwicGF0aCI6Ii8xNzEwMTYvMzk3OTYyNDMwLTU5NTIwNDBiLWFhNWEtNDNjNi04MTY5LThjMzZiODYxMWJhNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyM1QwNjIxNDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YzI5MzlkYjUwNDQwMDA1OTY5MTMzZmRjYmJhNTQ4ODQ4OGY0ZWM0MWJmZjY0MGRkYjVlMDU3YjIzMWQ5ZWI4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.CLMzV8ObQVuJs91HKhtR7kzw_cGypZ9_cSOmS5wsea4)

我找了一个 B550M 的主板，由 CPU 接出的 USB 可以使用，您可以根据主板的说明尝试一下