revyos/revyos #95: 请求打包 zerotier-one

HougeLangley, 2025-01-03:
zerotier-one 是一个内网穿透工具，目前官网提供了默认的 debian 源可以提供下载 Debian 12 Bookworm, Debian 11 Bullseye 都默认提供了 zerotier-one ，不过目前 Debian 13/Sid trixie 还没有提供。

目前通过 zerotier-one github 官网下载源码可以成功编译、打包 trixie 的 deb 。因为我个人打包 deb 不够好，修改了 `debian/source/format` 从原先的 3.0 (quilt) 改成了 1.0 以避免需要源码包（不干净的手段）。

```
sipeed@licheepi4a-console ~/zerotierone-1.14.2> cat ../zerotier-one_1.14.2_riscv64.buildinfo 
Format: 1.0
Source: zerotier-one
Binary: zerotier-one zerotier-one-dbgsym
Architecture: riscv64 source
Version: 1.14.2
Checksums-Md5:
 d4c997f167842fb9c3111f7126fdd3fa 676 zerotier-one_1.14.2.dsc
 afa9c642628902e7bcdd8c8fa3fd424c 55836 zerotier-one-dbgsym_1.14.2_riscv64.deb
 d4944f1562e048a7ce93544ddba95d2f 914368 zerotier-one_1.14.2_riscv64.deb
Checksums-Sha1:
 014b7d47bc60685a789dcd13fa21f3e134e63d16 676 zerotier-one_1.14.2.dsc
 e9b59ba04027a70975f56e9bec48d24b520d4f1b 55836 zerotier-one-dbgsym_1.14.2_riscv64.deb
 97093e88aa751f01a1c7bed5e71e33b371942b3b 914368 zerotier-one_1.14.2_riscv64.deb
Checksums-Sha256:
 98ac320beaddf54af08d9383b1095e690e15eff2d46072cebbb0968dca772f72 676 zerotier-one_1.14.2.dsc
 6a03484978a5bb762aaa0013a805db39489e14eeafb220036edfbb72dd8f0f13 55836 zerotier-one-dbgsym_1.14.2_riscv64.deb
 bb860706d1e4ed1120cb24f2e281a49996d6e5faf1fdf1b36e67e600645dd597 914368 zerotier-one_1.14.2_riscv64.deb
Build-Origin: Debian
Build-Architecture: riscv64
Build-Date: Fri, 03 Jan 2025 23:03:41 +0800
Build-Tainted-By:
 merged-usr-via-aliased-dirs
 usr-local-has-programs
Installed-Build-Depends:
 autoconf (= 2.71-3),
 automake (= 1:1.16.5-1.3),
 autopoint (= 0.21-12),
 autotools-dev (= 20220109.1),
 base-files (= 13.5),
 base-passwd (= 3.6.1),
 bash (= 5.2.32-1),
 binutils (= 2.43.1-3revyos1),
 binutils-common (= 2.43.1-3revyos1),
 binutils-riscv64-linux-gnu (= 2.43.1-3revyos1),
 bsdextrautils (= 2.38.1-5+b1),
 bsdutils (= 1:2.38.1-5+b1),
 build-essential (= 12.9+b1),
 bzip2 (= 1.0.8-5+b1),
 coreutils (= 9.1-1),
 cpp (= 4:14.1.0-2),
 cpp-14 (= 14.2.0-11revyos1),
 cpp-14-riscv64-linux-gnu (= 14.2.0-11revyos1),
 cpp-riscv64-linux-gnu (= 4:14.1.0-2),
 dash (= 0.5.12-2),
 debconf (= 1.5.82),
 debhelper (= 13.20),
 debianutils (= 5.20),
 dh-autoreconf (= 20),
 dh-strip-nondeterminism (= 1.13.1-1),
 diffutils (= 1:3.8-4),
 dpkg (= 1.22.11),
 dpkg-dev (= 1.22.11),
 dwz (= 0.15-1),
 file (= 1:5.44-3),
 findutils (= 4.9.0-4),
 g++ (= 4:14.1.0-2),
 g++-14 (= 14.2.0-11revyos1),
 g++-14-riscv64-linux-gnu (= 14.2.0-11revyos1),
 g++-riscv64-linux-gnu (= 4:14.1.0-2),
 gcc (= 4:14.1.0-2),
 gcc-14 (= 14.2.0-11revyos1),
 gcc-14-base (= 14.2.0-11revyos1),
 gcc-14-riscv64-linux-gnu (= 14.2.0-11revyos1),
 gcc-riscv64-linux-gnu (= 4:14.1.0-2),
 gettext (= 0.21-12),
 gettext-base (= 0.21-12),
 grep (= 3.8-5),
 groff-base (= 1.22.4-10),
 gzip (= 1.12-1+b1),
 hostname (= 3.23+nmu1),
 init-system-helpers (= 1.67),
 intltool-debian (= 0.35.0+20060710.6),
 libacl1 (= 2.3.1-3),
 libarchive-zip-perl (= 1.68-1),
 libasan8 (= 14.2.0-11revyos1),
 libatomic1 (= 14.2.0-11revyos1),
 libattr1 (= 1:2.5.1-4),
 libaudit-common (= 1:3.1.2-2),
 libaudit1 (= 1:3.1.2-2),
 libbinutils (= 2.43.1-3revyos1),
 libblkid1 (= 2.38.1-5+b1),
 libbz2-1.0 (= 1.0.8-5+b1),
 libc-bin (= 2.40-3revyos1),
 libc-dev-bin (= 2.40-3revyos1),
 libc6 (= 2.40-3revyos1),
 libc6-dev (= 2.40-3revyos1),
 libcap-ng0 (= 0.8.3-1+b3),
 libcap2 (= 1:2.66-4),
 libcc1-0 (= 14.2.0-11revyos1),
 libcrypt-dev (= 1:4.4.33-2),
 libcrypt1 (= 1:4.4.33-2),
 libctf-nobfd0 (= 2.43.1-3revyos1),
 libctf0 (= 2.43.1-3revyos1),
 libdb5.3 (= 5.3.28+dfsg2-1),
 libdebconfclient0 (= 0.272),
 libdebhelper-perl (= 13.20),
 libdpkg-perl (= 1.22.11),
 libelf1 (= 0.188-2.1),
 libfile-find-rule-perl (= 0.34-3),
 libfile-stripnondeterminism-perl (= 1.13.1-1),
 libgcc-14-dev (= 14.2.0-11revyos1),
 libgcc-s1 (= 14.2.0-11revyos1),
 libgdbm-compat4 (= 1.23-3),
 libgdbm6 (= 1.23-3),
 libgmp10 (= 2:6.2.1+dfsg1-1.1),
 libgomp1 (= 14.2.0-11revyos1),
 libgprofng0 (= 2.43.1-3revyos1),
 libicu72 (= 72.1-5),
 libisl23 (= 0.25-1),
 libitm1 (= 14.2.0-11revyos1),
 libjansson4 (= 2.14-2),
 liblsan0 (= 14.2.0-11revyos1),
 liblzma5 (= 5.6.2-2),
 libmagic-mgc (= 1:5.44-3),
 libmagic1 (= 1:5.44-3),
 libmd0 (= 1.0.4-2),
 libmount1 (= 2.38.1-5+b1),
 libmpc3 (= 1.3.1-1),
 libmpfr6 (= 4.2.0-1),
 libnumber-compare-perl (= 0.03-3),
 libpam-modules (= 1.5.2-6),
 libpam-modules-bin (= 1.5.2-6),
 libpam-runtime (= 1.5.2-6),
 libpam0g (= 1.5.2-6),
 libpcre2-8-0 (= 10.42-4revyos2),
 libperl5.36 (= 5.36.0-7),
 libpipeline1 (= 1.5.7-1),
 libselinux1 (= 3.4-1+b6),
 libsframe1 (= 2.43.1-3revyos1),
 libsmartcols1 (= 2.38.1-5+b1),
 libstdc++-14-dev (= 14.2.0-11revyos1),
 libstdc++6 (= 14.2.0-11revyos1),
 libsub-override-perl (= 0.09-4),
 libsystemd0 (= 256.5-2),
 libtext-glob-perl (= 0.11-3),
 libtinfo6 (= 6.5-2),
 libtool (= 2.4.7-5),
 libtsan2 (= 14.2.0-11revyos1),
 libubsan1 (= 14.2.0-11revyos1),
 libuchardet0 (= 0.0.7-1),
 libudev1 (= 256.5-2),
 libunistring2 (= 1.0-2),
 libuuid1 (= 2.38.1-5+b1),
 libxml2 (= 2.12.7+dfsg+really2.9.14-0.1),
 libzstd1 (= 1.5.6+dfsg-1),
 linux-libc-dev (= 6.10.3-1),
 login (= 1:4.13+dfsg1-1+b1),
 m4 (= 1.4.19-3),
 make (= 4.3-4.1+b1),
 man-db (= 2.11.2-2),
 mawk (= 1.3.4.20200120-3.1+b1),
 ncurses-base (= 6.5-2),
 ncurses-bin (= 6.5-2),
 patch (= 2.7.6-7+b1),
 perl (= 5.36.0-7),
 perl-base (= 5.36.0-7),
 perl-modules-5.36 (= 5.36.0-7),
 po-debconf (= 1.0.21+nmu1),
 rpcsvc-proto (= 1.4.3-1),
 sed (= 4.9-1),
 sensible-utils (= 0.0.17+nmu1),
 sysvinit-utils (= 3.10-3),
 tar (= 1.34+dfsg-1.2),
 usrmerge (= 35),
 util-linux (= 2.38.1-5+b1),
 util-linux-extra (= 2.38.1-5+b1),
 xz-utils (= 5.6.2-2),
 zlib1g (= 1:1.2.13.dfsg-1)
Environment:
 DEB_BUILD_OPTIONS="parallel=4"
 LANG="en_US.UTF-8"
 SOURCE_DATE_EPOCH="1729670400"
sipeed@licheepi4a-console ~/zerotierone-1.14.2>
```

当然，在后续估计官方会出正式版本。就看朋友们的意愿了。

KevinMX, 2025-01-03:
https://github.com/zerotier/ZeroTierOne/blob/dev/LICENSE.txt

ZeroTier One 所使用的许可证不是开源软件许可证。
打包进软件源可能会有潜在的法律问题。

HougeLangley, 2025-01-18:
我先关喽，各位朋友新年快乐

closed 2025-01-18 by HougeLangley