revyos/revyos #97: blank screen on revyos release 20241229_032148

viktar-siarheichyk, 2025-01-05:
Iiyama ProLite XUB2792UHSU
After initial installation is works fine, but when it is left unattended and power save screen off occurs, it never wakes up, and after reboot lighted prompt is shown on the screen, but after login it goes black. 
[Xorg.0.log](https://github.com/user-attachments/files/18310926/Xorg.0.log)
[journal.today.log](https://github.com/user-attachments/files/18310927/journal.today.log)

viktar-siarheichyk, 2025-01-14:
> > Iiyama ProLite XUB2792UHSU 初始安装后运行良好，但当无人看管且省电屏幕关闭时，它永远不会唤醒，重启后屏幕上会显示亮起的提示，但登录后它会变黑[。Xorg.0.log ](https://github.com/user-attachments/files/18310927/journal.today.log) [journal.today.log](https://github.com/user-attachments/files/18310926/Xorg.0.log) [](https://github.com/user-attachments/files/18310927/journal.today.log)
> 
> 
> 
> After testing, in the latest image [20250110](https://mirror.iscas.ac.cn/revyos/extra/images/lpi4a/20250110/), the screen will wake up and display the desktop image normally when clicking the mouse after the screen turns off to save power.
> 
> 
> 
> The 20241229 version currently has several issues. We recommend downloading and using the new image version. Thank you for using RevyOS!

Frankly, nothing changed on the 4K monitor, sorry…

FIFCC, 2025-01-14:
> Frankly, nothing changed on the 4K monitor, sorry…

Oh, 4K monitors may encounter various issues, such as those listed in the [Known compatibility issue] (https://github.com/orgs/revyos/discussions/83#discussioncomment-11573456). We are actively working on resolving them. Thank you for your feedback!

KevinMX, 2025-01-14:
[image]

OK so I tried on a LG 27UL550 and it appears to be working...

For `parse-edid` you'll need `sudo apt install read-edid`

viktar-siarheichyk, 2025-01-18:
Jan 18 18:32:55 marrow.eq.by kernel: dwhdmi-th1520 ffef540000.dw-hdmi-tx: PHY configuration failed (clock 49847000)
Jan 18 18:32:56 marrow.eq.by kernel: dwhdmi-th1520 ffef540000.dw-hdmi-tx: TMDS link of scrambling_status is not ready
Jan 18 18:33:30 marrow.eq.by kernel: dwhdmi-th1520 ffef540000.dw-hdmi-tx: PHY configuration failed (clock 49847000)
Jan 18 18:33:41 marrow.eq.by kernel: dwhdmi-th1520 ffef540000.dw-hdmi-tx: PHY configuration failed (clock 49847000)

viktar-siarheichyk, 2025-01-18:
Jan 18 16:31:20 marrow.eq.by kernel: BUG: Bad page state: 16699 messages suppressed
Jan 18 16:31:20 marrow.eq.by kernel: BUG: Bad page state in process gdbus  pfn:4b5e4
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80032a82>] get_signal+0x992/0x9b6
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff800058be>] arch_do_signal_or_restart+0x326/0x5b4
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff800bf0e6>] exit_to_user_mode_prepare+0xea/0x164
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80a43e26>] syscall_exit_to_user_mode+0x22/0x46
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80a4393a>] do_trap_ecall_u+0x52/0x14a
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80a4ee20>] ret_from_exception+0x0/0x64
Jan 18 16:31:36 marrow.eq.by kernel: BUG: Bad page state in process gdbus  pfn:49ed7
Jan 18 16:31:36 marrow.eq.by kernel: page:000000003dc2223b refcount:0 mapcount:0 mapping:0000000000000000 index:0x0 pfn:0x49ed7
Jan 18 16:31:36 marrow.eq.by kernel: flags: 0xfffc00000004000(reserved|node=0|zone=0|lastcpupid=0x3fff)
Jan 18 16:31:36 marrow.eq.by kernel: page_type: 0xffffffff()
Jan 18 16:31:36 marrow.eq.by kernel: raw: 0fffc00000004000 0000000000000000 dead000000000122 0000000000000000
Jan 18 16:31:36 marrow.eq.by kernel: raw: 0000000000000000 0000000000000000 00000000ffffffff 0000000000000000
Jan 18 16:31:36 marrow.eq.by kernel: page dumped because: PAGE_FLAGS_CHECK_AT_FREE flag(s) set
Jan 18 16:31:36 marrow.eq.by kernel: Modules linked in: sd_mod t10_pi crc64_rocksoft_generic crc64_rocksoft crc_t10dif crct10dif_generic crc64 crct10dif_common sg uas usb_storage scsi_mod sc>
Jan 18 16:31:36 marrow.eq.by kernel: CPU: 0 PID: 2561 Comm: gdbus Tainted: G    B   W          6.6.66-th1520 #2025.01.10.02.53+1c6721ec2
Jan 18 16:31:36 marrow.eq.by kernel: Hardware name: Sipeed Lichee Pi 4A 16G (DT)
Jan 18 16:31:36 marrow.eq.by kernel: Call Trace:
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff800069c8>] dump_backtrace+0x28/0x30
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80a35022>] show_stack+0x38/0x44
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80a42ee8>] dump_stack_lvl+0x44/0x5c
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80a42f18>] dump_stack+0x18/0x20
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80257fc6>] bad_page+0x11a/0x162
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80258050>] free_page_is_bad_report+0x42/0xa2
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80258dfa>] free_unref_page_prepare+0x148/0x316
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff8025bad6>] free_unref_page+0x5a/0x1c0
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff8025be5a>] __free_pages+0x106/0x10c
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff025fb50a>] free_memblk_pages+0x76/0x26c [vidmem]
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff025fb9be>] GFP_Free+0xb6/0x158 [vidmem]
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff025fbbbc>] vidalloc_release+0x15c/0x1c8 [vidmem]
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff802c7de4>] __fput+0xba/0x21a
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff802c7f96>] ____fput+0x18/0x20
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff800422b6>] task_work_run+0xa8/0xe6
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80023dfc>] do_exit+0x316/0x8f0
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80024566>] do_group_exit+0x30/0x80
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80032a82>] get_signal+0x992/0x9b6
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff800058be>] arch_do_signal_or_restart+0x326/0x5b4
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff800bf0e6>] exit_to_user_mode_prepare+0xea/0x164
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80a43e26>] syscall_exit_to_user_mode+0x22/0x46
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80a4393a>] do_trap_ecall_u+0x52/0x14a
Jan 18 16:31:36 marrow.eq.by kernel: [<ffffffff80a4ee20>] ret_from_exception+0x0/0x64