revyos/revyos #101: 20241229 LPi4A 串口/控制台运行 sudo reboot 后 mmc2 报错，无法进入系统；拔电源重启恢复

lazyparser, 2025-01-08:
weird

[image]

KevinMX, 2025-01-09:
See https://github.com/revyos/revyos/issues/91#issuecomment-2579809364, please try downgrading U-Boot firmware to 20240720

kagura114, 2025-01-10:
Lpi4A 16+128 20241229 重启后可以看到完全一样的报错，但是没有影响正常开机使用

KevinMX, 2025-01-10:
> Lpi4A 16+128 20241229 重启后可以看到完全一样的报错，但是没有影响正常开机使用

概率问题 会导致 eMMC 读写错误 目前建议降级 U-Boot

FIFCC, 2025-01-13:
> weird
> 
> <img alt="image" width="1020" src="https://private-user-images.githubusercontent.com/1802104/401238449-ea302a57-de66-4d5e-a414-649bfac2035b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3Mzc2MjgsIm5iZiI6MTczNjczNzMyOCwicGF0aCI6Ii8xODAyMTA0LzQwMTIzODQ0OS1lYTMwMmE1Ny1kZTY2LTRkNWUtYTQxNC02NDliZmFjMjAzNWIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDExMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMTNUMDMwMjA4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTlhMTRjODM5MjgzY2JjYjA2ZTI1YmJhZGNhY2ZiNTBkZWU0NGQ4YTliODhlNzE0MThiZTdkN2Q3ZWRlMTBhZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.xfISydiTQn02POlZPWQIKYdBYLvDq8mbtIVSq7Sifcs">

经过测试，在最新镜像[20250110](https://mirror.iscas.ac.cn/revyos/extra/images/lpi4a/20250110/)中此问题已经解决，在执行`sudo reboot`后系统可以正常进行重启，未出现mmc2报错。

KamijoToma, 2025-02-19:
经过测试，在20250123版本中问题复现。使用20250123构建的uboot，Lichee Pi4A 16G+128G emmc启动

KamijoToma, 2025-03-29:
same issue on RevyOS 20250323

KevinMX, 2025-03-30:
mmc2 corresponds to the on board SDIO Wi-Fi.
Currently we suspect only LPi4A boards with AIC8800 are affected, while RTL8723 ones are working fine.
(I can't reproduce this issue on my RTL8723 board either.)
My suggest is avoid warm reboots, only use cold boot for now.