revyos/revyos #102: img20241229 LPi4A USB无法识别键盘和鼠标等设备

lazyparser, 2025-01-08:
串口调试登陆查看 lsusb，插拔无线键盘和鼠标的接收器，lsusb 的输出结果无变化。

[image]

KevinMX, 2025-01-09:
Workaround: See https://github.com/revyos/revyos/issues/91#issuecomment-2579809364

FIFCC, 2025-01-13:
> 串口调试登陆查看 lsusb，插拔无线键盘和鼠标的接收器，lsusb 的输出结果无变化。
> 
> <img alt="image" width="500" src="https://private-user-images.githubusercontent.com/1802104/401238841-b9ee824a-ca6b-4e31-ad85-b89a930cf7e2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3MzczMDQsIm5iZiI6MTczNjczNzAwNCwicGF0aCI6Ii8xODAyMTA0LzQwMTIzODg0MS1iOWVlODI0YS1jYTZiLTRlMzEtYWQ4NS1iODlhOTMwY2Y3ZTIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDExMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMTNUMDI1NjQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTZjMzg2ODBmNjBkNzM5ZGQxM2I1NDlmMTVhNGE5OGNjMzljZDk3MmU4MmI3NDI2M2M3ZGFjNThlYWE3N2Q3NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.uIFi4GVEH_3E-qGzooeG-CwrssP5Mmaj4k5ZCcca0n8">

在最新版本的镜像[20250110](https://mirror.iscas.ac.cn/revyos/extra/images/lpi4a/20250110/)中经过测试，此问题已解决。

此为未插入无线键鼠接收器时lsusb后的回显
[image]

此为插入无线键鼠接收器后lsusb的回显
[image]

根据对比可见新增了`Bus 001 Device 007: ID 046d:c534 Logitech, Inc. Nano Receiver`

lIlIlxy, 2025-01-22:
在20250110版本中USB无法识别键盘和鼠标等设备的问题再次出现
暂时没有串口调试设备无法登录查看lsusb

KevinMX, 2025-01-22:
> 在20250110版本中USB无法识别键盘和鼠标等设备的问题再次出现 暂时没有串口调试设备无法登录查看lsusb

请尝试直接按 Reset 按钮，检查重启后 USB 供电是否恢复。
> 可能需要多次重启。

lIlIlxy, 2025-01-22:
> > 在20250110版本中USB无法识别键盘和鼠标等设备的问题再次出现 暂时没有串口调试设备无法登录查看lsusb
> 
> 请尝试直接按 Reset 按钮，检查重启后 USB 供电是否恢复。
> 
> > 可能需要多次重启。

已解决，在数次重启之后，谢谢