revyos/revyos #106: 无法正确识别显示器分辨率

stydxm, 2025-01-08:
该显示器实际分辨率为3440*1440，这里虽然识别出了尺寸但未能正确识别分辨率，也无法手动设置为21:9的比例

[image]

20241229版的系统，硬件为lp4a 16+128

FIFCC, 2025-01-13:
> 该显示器实际分辨率为3440*1440，这里虽然识别出了尺寸但未能正确识别分辨率，也无法手动设置为21:9的比例 [...]

在最新版本[20250110](https://mirror.iscas.ac.cn/revyos/extra/images/lpi4a/20250110/)的镜像中，经过测试，在1920x1080的屏幕中可以正常显示屏幕分辨率，并且可以正确修改为其它比例

stydxm, 2025-01-13:
> 在最新版本[20250110](https://mirror.iscas.ac.cn/revyos/extra/images/lpi4a/20250110/)的镜像中，经过测试，在1920x1080的屏幕中可以正常显示屏幕分辨率，并且可以正确修改为其它比例

手动修改分辨率，仍然没有找到21:9的选项

[image]

KamijoToma, 2025-02-02:
试试用 `xrandr` 命令自定义显示分辨率试试？[参考这里](https://unix.stackexchange.com/questions/227876/how-to-set-custom-resolution-using-xrandr-when-the-resolution-is-not-available-i)
我尝试了一下，新建了一个 `2560x1080_30` 的配置，能成功切换，但是由于我接入的是采集卡无法正常输出。

stydxm, 2025-02-02:
> 试试用 `xrandr` 命令自定义显示分辨率试试？[参考这里](https://unix.stackexchange.com/questions/227876/how-to-set-custom-resolution-using-xrandr-when-the-resolution-is-not-available-i) 我尝试了一下，新建了一个 `2560x1080_30` 的配置，能成功切换，但是由于我接入的是采集卡无法正常输出。

现在板子不在身边，等年后我试下，感谢

KamijoToma, 2025-02-13:
> > 试试用 `xrandr` 命令自定义显示分辨率试试？[参考这里](https://unix.stackexchange.com/questions/227876/how-to-set-custom-resolution-using-xrandr-when-the-resolution-is-not-available-i) 我尝试了一下，新建了一个 `2560x1080_30` 的配置，能成功切换，但是由于我接入的是采集卡无法正常输出。
> 
> 现在板子不在身边，等年后我试下，感谢

你好，请问有新进展吗？

stydxm, 2025-02-17:
> 试试用 `xrandr` 命令自定义显示分辨率试试？[参考这里](https://unix.stackexchange.com/questions/227876/how-to-set-custom-resolution-using-xrandr-when-the-resolution-is-not-available-i) 我尝试了一下，新建了一个 `2560x1080_30` 的配置，能成功切换，但是由于我接入的是采集卡无法正常输出。

我这里测试并不能成功切换，即使 `xrandr --verbose` 显示 `CRTC: 0`，但仍然报错 `xrandr: Configure crtc 0 failed`

[image]

KamijoToma, 2025-02-17:
> 我这里测试并不能成功切换，即使 `xrandr --verbose` 显示 `CRTC: 0`，但仍然报错 `xrandr: Configure crtc 0 failed`
>
> [image]

了解了，我没有更进一步的解决办法了😂您可以尝试openEuler或archlinux等其他适配了th1520的发行版，看看能不能正常工作。issue先保持开放，后续有新变化欢迎向我们反馈😊