revyos/revyos #108: hwprobe没有xtheadvector的id

ziyu04, 2025-01-12:
环境信息 revyos 20241229 内核 6.6.66-th1520

KamijoToma, 2025-02-02:
你好，请问期望在何处出现`xtheadvector`的ID？就系统本身来说，我测试了`lscpu`和`/proc/cpuinfo`都没有看到这个字样。

Hello, could you please clarify where you expect to see the ID for `xtheadvector`? To RevyOS itself, I have checked both `lscpu` and `/proc/cpuinfo`, but I couldn't find this.

ziyu04, 2025-02-11:
抱歉，这个目前似乎没有引入。据我所知，hwprobe似乎下个开发版本会引入吗？

KamijoToma, 2025-02-17:
> 抱歉，这个目前似乎没有引入。据我所知，hwprobe似乎下个开发版本会引入吗？

你好，我是参与 RevyOS 项目的实习生，翻看[提交记录](https://github.com/revyos/th1520-linux-kernel/commit/4f7f71aa5ad742cf423cac00e1e67711009f7cb4)了解到，三天前刚刚加入了hwprobe相关支持，请关注后续新发布的版本。😊

KamijoToma, 2025-02-19:
你好，在20250123镜像版本中查看`/proc/cpuinfo`已经可以查看到xtheadvector相关信息

[image]

ziyu04, 2025-03-19:
好的，谢谢您。

closed 2026-01-07 by RevySR