revyos/revyos #110: revyos docs 刷写文档修改

FIFCC, 2025-01-15:
- [ ]  1、左侧镜像刷写栏需要列出所有支持的设备

- [x] 2、需要做一个索引，所有镜像支持都需要写出来
添加了一个总页面，现在用户点击镜像刷写会跳转到一个汇总页面，可以看到所有支持设备所对应的镜像刷写教程情况
https://docs.revyos.dev/Installation/install/
并且在刷写教程开头增加了一个连接可以跳转到其它刷写文档

- [x] 3、需要添加一个licheepi4a的图片，让用户在刷写前进行对比
已添加

- [x] 4、演示环境描述更加具体
已完成，演示环境已更加细化，增加了架构，系统具体版本，内核版本

- [x] 5、除开演示环境外的说明
其它环境下的刷写过程参考需要去搜集并且给实验，可以当成实习生任务进行安排

- [x] 6、目录结构优化
完成一半，现在在开头已经修改为刷写方式二选一，可以点击跳转

- [x] 7、拨码开关图片
已添加图片并进行了说明

- [x] 8、早期版本没有拨码开关说明
已添加，但是说明不够具体，后续需要改进

- [x] 9、关于拨码开关缺失问题引导到论坛或ruyisdk qq群
需要根据具体情况考虑

- [x] 10、从sd卡启动的部分结构需要修改
已完成，现有目录结构如下：镜像获取-硬件准备-烧录方式介绍（图形界面或者命令行界面，可进行点击跳转）-板卡接线说明

- [x] 11、现有安装信息进行扩展
已完成，在每个操作中添加了其它可用的操作，例如在zstd中添加了也可以使用tar、7z等解压zst文件
对于现有使用的zstd安装：如何安装，如何验证安装成功，如何查看版本、如何使用
新增内容
对于wget的安装：安装命令，wget说明，wget使用

- [x] 12、apt前需要加sudo
已添加，sudo说明也已经添加

- [x] 13、镜像版本说明
已添加，在文档中已添加0720和0110的版本说明，并说明可以根据内核版本选择镜像

- [ ] 14、需要添加相关演示图片
图片已存在，需要整理后添加

- [x] 15、烧写部分需要调整目录结构
已修改，现在在硬件和镜像部分结束后添加了一个烧录方式介绍的说明，可以点击跳转

- [x] 16、需要对dd命令进行说明
已修改，对dd进行了介绍，并且细化了dd的步骤，对于每步添加了说明

- [ ] 17、系统启动需要添加说明
说明了启动插线的顺序，并且说明了电源如何接，配图未添加，并且电源部分可以再拓展一下说明

- [x] 18、需要对串口线进行说明
暂未添加，需要说明产品本身不附带串口线，什么是串口线，什么样的串口线可以接，如何接串口线，可不可以不接串口线（是否必须）

- [x] 19、需要修改emmc部分准备工作的内容
在准备工作部分需要进行添加说明，列举接下来的行动

- [ ] 20、需要把代码块转为可复制
注意代码块不能带美元符号

- [ ] 21、串口连接需要说明
串口连接不能仅限于minicom，需要进行说明，增加其它工具选择

- [ ] 22、如何安装串口工具
对于推荐的串口工具如何进行安装

- [ ] 23、串口工具使用说明
如何使用串口工具，需要做哪些设置，需要以什么顺序连接开发板和线材

- [ ] 24、获取镜像部分重复
目录结构问题，需要进行修改

- [ ] 25、板卡规格介绍
需要对板卡规格进行说明和介绍

- [x] 26、解压给出预期大小
在sd卡部分已添加，给出了预先需要预留出的大小，并说明了未解压文件大小和解压后文件的大小

- [ ] 27、串口刷写的目录结构修改
把串口刷写放到最后，在前面做一个跳转栏可以直接点击跳转

- [ ] 28、修改emmc部分的目录结构
应该修改为准备工作-不接入串口-接入串口

- [ ] 29、修改串口线连接的图片
需要添加标注，修改颜色

- [ ] 30、串口线连接视频
可以添加一个视频、作为串口线连接的演示，避免图片说明不能让用户理解

- [ ] 31、需要添加命令执行后的回显
在sd卡部分添加了相应的图片并进行说明

- [ ] 32、对fastboot usb 0进行说明
命令说明，进入后会有回显吗，怎么判断进入了

- [ ] 33、需要对板卡型号进行说明
同25

- [ ] 34、刷写出现问题后需要有入口解决
在sd卡部分添加了出现问题导向docs的issue页面

- [ ] 35、需要添加每个步骤刷写成功后的图片

- [x] 36、添加电源线接线说明
在sd卡部分已添加

- [ ] 37、需要说明登录后修改密码
需要看看是否做一个功能出来

- [ ] 38、pioneer需要说明版本

- [ ] 39、英文版本跟随修改

- [x] 40、不可用的空链接需要删除
已删除

- [ ] 41、需要对镜像文件进行分开说明

- [ ] 42、首页导向changlog的连接会404
需要确认连接并修改

- [ ] 43、需要添加其它支持设备的changlog

- [ ] 44、需要添加changlog的默认页面
做一个导向跳转到想查看的设备

- [ ] 45、系统使用页面需要添加一个页面
同44

- [x] 46、首页对RuyiSDK介绍需要有外链
已添加

FIFCC, 2025-01-16:
- [ ] 46.所有的外链需要加符号

- [ ] 47.简介第一段要求修改
芯片生态>生态芯片

- [ ] 48.简介第一段需要修改
长句修改为短句

- [ ] 49.revyos小队需要介绍
做一个页面介绍revyos

- [ ] 50.简介第三段需要拆开讲
避免长修饰句型，用短句型隔开

- [ ] 51.满足基本需求应该有演示
做一个页面用来展示

- [ ] 52.主要宣传4a和meles
meles在4a前面

- [ ] 53.需要确认产品型号与官方一致

- [ ] 54.需要一个页面做镜像站列表列举

- [ ] 55.支持设备排序需要修改
4a-meles-pioneer，按照支持的优先程度排序

- [ ] 56.对于旧镜像或其他设备添加备注
体现使用有风险

- [ ] 57.优化gcc的标题需要进行修改
表达准确意思

- [ ] 58.thead相关内容需要修改
可修改的内容改为xuantie

- [ ] 59.gcc文章内容需要扩展

- [ ] 60.主页目录结构有问题
需要进行整合和修改，要么整合到一块，要么每个模块点开后都是一个index

- [ ] 61.benchmark中需要加一个index

- [ ] 62.死链需要进行检查
修改或删除

- [ ] 63.刷写教程需要重新进行排序

- [ ] 64.外链需要进行备注说明

- [ ] 65.需要对licheepi4a的图片进行修改
需要进行拆解后的详细说明，添加一张底板图片

- [ ] 66.内核版本说明删除
没必要添加

- [ ] 67.镜像需要添加说明

- [ ] 68.截图需要英文界面
需要重新截图，并避免个人信息，英文图片可以中英文档同时使用

- [ ] 69. changelog补充
添加主页并且添加其他支持设备的changelog

- [ ] 70.issues页面优先翻译

- [ ] 71.发送邮件部分需要修改

- [ ] 72.issue页面语言重新组织
软化语气

- [ ] 73.issue页面开头介绍需要更全面
可以添加特性需求提交和许愿

- [ ] 74.需要着重体现已知问题
可以添加一个faq或者链接

- [ ] 75.关注中英文不一致的问题

- [ ] 76.ruyisdk首页滚动栏需要添加revyos的新版本信息

KamijoToma, 2025-02-15:
- [ ] 77. 为[此处](https://github.com/revyos/docs/pull/23#discussion_r1957096156)添加一个案例截图