revyos/revyos #121: licheepi4a开机时连接hdmi 4k显示器会让hdmi输出错误

DingWH03, 2025-04-02:
硬件：licheepi4a 16g+128g，emmc启动，显示器拓硕27寸
软件版本：boot-lpi4a-20250323_154524.ext4
开机时若插入显示器则画面黑屏，插入采集卡后显示花屏画面。若开机时不插显示器或插入采集卡(1080p)则可正常输出画面。
如果需要帮忙调试，也可以根据指导提供一些信息。
一些可能会用到的信息：
sipeed@revyos-lpi4a:~$ sudo dmesg | grep -i hdmi
[sudo] password for sipeed: 
[    1.006731] /soc/dw-hdmi-tx@ffef540000: Fixed dependency cycle(s) with /soc/dc8200@ffef600000
[    1.015504] /soc/dc8200@ffef600000: Fixed dependency cycle(s) with /soc/dw-hdmi-tx@ffef540000
[    1.044759] /soc/dw-hdmi-tx@ffef540000: Fixed dependency cycle(s) with /soc/dc8200@ffef600000
[    1.053863] /soc/dw-hdmi-tx@ffef540000: Fixed dependency cycle(s) with /soc/dc8200@ffef600000
[    1.062730] /soc/dc8200@ffef600000: Fixed dependency cycle(s) with /soc/dw-hdmi-tx@ffef540000
[    5.468676] dwhdmi-th1520 ffef540000.dw-hdmi-tx: Detected HDMI TX controller v2.14a with HDCP (DWC HDMI 2.0 TX PHY)
[    5.480086] dwhdmi-th1520 ffef540000.dw-hdmi-tx: registered DesignWare HDMI I2C bus driver
[    5.489054] vs-drm display-subsystem: bound ffef540000.dw-hdmi-tx (ops dw_hdmi_th1520_ops)
[    6.562146] dwhdmi-th1520 ffef540000.dw-hdmi-tx: TMDS link of scrambling_status is not ready
sipeed@revyos-lpi4a:~$ sudo dmesg | grep -i vs-drm
[    5.460576] vs-drm display-subsystem: bound ffef600000.dc8200 (ops dc_component_ops)
[    5.489054] vs-drm display-subsystem: bound ffef540000.dw-hdmi-tx (ops dw_hdmi_th1520_ops)
[    5.498864] [drm] Initialized vs-drm 1.0.0 20191101 for display-subsystem on minor 0
[    5.506839] vs-drm display-subsystem: [drm] Cannot find any crtc or sizes
[    5.513911] vs-drm display-subsystem: [drm] Cannot find any crtc or sizes
[    7.072883] vs-drm display-subsystem: [drm] fb0: vs-drmdrmfb frame buffer device
sipeed@revyos-lpi4a:~$ cat /sys/class/drm/card0-HDMI-A-1/status
connected
sipeed@revyos-lpi4a:~$ ls /dev/fb*
/dev/fb0

DingWH03, 2025-04-02:
[image]
这是采集卡采集到的画面

KevinMX, 2025-04-02:
能否确认下显示器和采集卡的具体型号/芯片信息？可以直接定位到购买链接也可以。
这边用 MS2130 芯片的采集卡调试的时候都没有遇到过花屏问题。

DingWH03, 2025-04-02:
> 能否确认下显示器和采集卡的具体型号/芯片信息？可以直接定位到购买链接也可以。 这边用 MS2130 芯片的采集卡调试的时候都没有遇到过花屏问题。

```
$ hwinfo --monitor
20: None 00.0: 10002 LCD Monitor                                
  [Created at monitor.125]
  Unique ID: rdCR.rLw16Jkfqh5
  Parent ID: _Znp.iH5RHOo8m+D
  Hardware Class: monitor
  Model: "G27N3"
  Vendor: BKU 
  Device: eisa 0x2700 "G27N3"
  Serial ID: "0000000000001"
  Resolution: 720x400@70Hz
  Resolution: 640x480@60Hz
  Resolution: 800x600@56Hz
  Resolution: 800x600@60Hz
  Resolution: 1024x768@60Hz
  Resolution: 1920x1080@60Hz
  Resolution: 1600x1200@60Hz
  Resolution: 3840x2160@60Hz
  Size: 600x330 mm
  Year of Manufacture: 2022
  Week of Manufacture: 9
  Detailed Timings #0:
     Resolution: 3840x2160
     Horizontal: 3840 3888 3920 4000 (+48 +80 +160) -hsync
       Vertical: 2160 2163 2168 2222 (+3 +8 +62) +vsync
    Frequencies: 533.25 MHz, 133.31 kHz, 60.00 Hz
  Driver Info #0:
    Max. Resolution: 3840x2160
    Vert. Sync Range: 48-60 Hz
    Hor. Sync Range: 160-160 kHz
    Bandwidth: 533 MHz
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #17 (VGA compatible controller)
```
以上是我的显示器详细信息，采集卡使用的也是ms2130主控。
另外采集卡出现花屏是开机时插入显示器导致，若插入采集卡开机则一切正常，所以问题应该不出在采集卡这边。

KevinMX, 2025-04-02:
看一下 EDID 信息？参考 https://github.com/revyos/revyos/issues/117

Icenowy, 2025-04-02:
看上去可能和之前的问题相关？

Icenowy, 2025-04-02:
这个 timing 里面的 dotclock 是 533.25MHz ，但是驱动里预置的 4K 配置似乎是 594MHz ，有所不同，可能有匹配问题。

KevinMX, 2025-04-02:
> 看上去可能和之前的问题相关？

嗯，怀疑是同一类问题，已经关联了

Icenowy, 2025-04-02:
我现阶段怀疑这个驱动没能成功拒绝它不认识的显示模式被设置（怀疑 mode_valid 没被执行，检查了下代码，似乎 verisilicon 驱动里没调用过 `drm_atomic_helper_check_modeset` ），但又没做好设置的时候的 fallback 导致 HDMI PHY 遇到不认识的显示模式摆大烂乱输出（
#117 的 proposed fix 则是修复了 HDMI PHY 配置的时候的 fallback 逻辑（

DingWH03, 2025-04-02:
> 看一下 EDID 信息？参考 [#117](https://github.com/revyos/revyos/issues/117)

您好，edid信息如下：
```
sipeed@revyos-lpi4a:~$ hexdump /sys/class/drm/card0-HDMI-A-1/edid
0000000 ff00 ffff ffff 00ff 7509 2700 0000 0000
0000010 2009 0301 3c80 7821 592e af95 424f 26af
0000020 500f bf54 80ef c0e1 00d1 c0d1 00b3 c0a9
0000030 0095 0081 c081 e808 3000 70f2 805a 58b0
0000040 004a 4a58 0021 1a00 0000 ff00 3000 3030
0000050 3030 3030 3030 3030 3030 0000 fd00 2800
0000060 0f3c 3ccc 0a00 2020 2020 2020 0000 fc00
0000070 4700 3732 334e 200a 2020 2020 2020 0301
0000080 0302 f147 614d 5a60 0510 0304 1f02 1213
0000090 5f01 0923 077f 0183 0000 036d 000c 0010
00000a0 3c38 0020 0160 0302 d867 c45d 7801 0388
00000b0 1a68 0000 0101 4030 e3ed e005 e201 7f0f
00000c0 06e6 0107 536b 9e4f 00e0 a078 32a0 1050
00000d0 3540 ba00 2189 0000 7b1a 807f 7078 1f38
00000e0 3040 4520 5500 2150 0000 561a 005e a0a0
00000f0 29a0 3050 3520 5500 2150 0000 001a c300
0000100
```
由于本人还是学生，回寝室时才能操作，回复不是很及时

DingWH03, 2025-04-02:
> 能不能收集一下花屏时的 `/sys/kernel/debug/dri/*/state` 和 `/sys/class/drm/*/modes` 信息？

你要的信息如下：
```
root@revyos-lpi4a:~# cat /sys/kernel/debug/dri/0/state 
plane[39]: Primary
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[51]: Overlay
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=1
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[63]: Overlay_1
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=2
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[74]: Primary_1
	crtc=DC8200
	fb=121
		allocated by = Xorg
		refcount=2
		format=XR24 little-endian (0x34325258)
		modifier=0x0
		size=3840x2160
		layers:
			size[0]=3840x2160
			pitch[0]=15360
			offset[0]=0
			obj[0]:
				name=1
				refcount=4
				start=00101fac
				size=33177600
				imported=no
	crtc-pos=3840x2160+0+0
	src-pos=3840.000000x2160.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=3
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[86]: Overlay_2
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=4
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[98]: Overlay_3
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=5
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[109]: Cursor
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=ff
	color-encoding=ITU-R BT.601 YCbCr
	color-range=YCbCr limited range
plane[112]: Cursor_1
	crtc=DC8200
	fb=120
		allocated by = Xorg
		refcount=1
		format=AR24 little-endian (0x34325241)
		modifier=0x0
		size=64x64
		layers:
			size[0]=64x64
			pitch[0]=256
			offset[0]=0
			obj[0]:
				name=0
				refcount=3
				start=00101fa8
				size=16384
				imported=no
	crtc-pos=64x64+1899+1057
	src-pos=64.000000x64.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=ff
	color-encoding=ITU-R BT.601 YCbCr
	color-range=YCbCr limited range
crtc[31]: DC8200
	enable=0
	active=0
	self_refresh_active=0
	planes_changed=0
	mode_changed=0
	active_changed=0
	connectors_changed=0
	color_mgmt_changed=0
	plane_mask=0
	connector_mask=0
	encoder_mask=0
	mode: "": 0 0 0 0 0 0 0 0 0 0 0x0 0x0
crtc[35]: DC8200
	enable=1
	active=0
	self_refresh_active=0
	planes_changed=0
	mode_changed=0
	active_changed=0
	connectors_changed=0
	color_mgmt_changed=0
	plane_mask=88
	connector_mask=1
	encoder_mask=1
	mode: "": 60 594000 3840 4016 4104 4400 2160 2164 2174 2250 0x0 0x9
connector[116]: HDMI-A-1
	crtc=DC8200
	self_refresh_aware=0
	max_requested_bpc=0
	colorspace=Default
```
```
sipeed@revyos-lpi4a:~$ cat /sys/class/drm/card0-HDMI-A-1/modes 
3840x2160
3840x2160
3840x2160
3840x2160
3840x2160
3840x2160
2560x1080
2048x1152
1920x1200
1920x1080
1920x1080
1920x1080i
1920x1080i
1920x1080
1680x1050
1600x900
1440x900
1280x800
1152x864
1280x720
1280x720
1280x720
1280x720
1024x768
1024x768
1024x768
800x600
800x600
800x600
800x600
720x576
720x480
720x480
720x480
720x480
640x480
640x480
640x480
640x480
640x480
sipeed@revyos-lpi4a:
```

DingWH03, 2025-04-02:
> 你要的信息如下：
>
> ```
> root@revyos-lpi4a:~# cat /sys/kernel/debug/dri/0/state
> plane[39]: Primary
> 	crtc=(null)
> 	fb=0
> 	crtc-pos=0x0+0+0
> 	src-pos=0.000000x0.000000+0.000000+0.000000
> 	rotation=1 [...]

上面的是插着4k显示器时的输出，下面是拔了显示器插入采集卡时的输出：
```
sipeed@revyos-lpi4a:~$ cat /sys/class/drm/card0-HDMI-A-1/modes 
1920x1080
4096x2160
3840x2160
1920x1080
1920x1080
1920x1080
1920x1080
1920x1080
1920x1080
1600x900
1280x1024
1440x900
1280x960
1280x800
1280x720
1280x720
1280x720
1280x720
1280x720
1280x720
1280x720
1280x720
1280x720
1280x720
1280x720
1440x576
1024x768
800x600
720x576
720x576
720x576
720x480
720x480
720x480
720x480
720x480
640x480
640x480
640x480
sipeed@revyos-lpi4a:~$ sudo -i
[sudo] password for sipeed: 
root@revyos-lpi4a:~# cat /sys/kernel/debug/dri/0/state 
plane[39]: Primary
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[51]: Overlay
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=1
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[63]: Overlay_1
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=2
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[74]: Primary_1
	crtc=DC8200
	fb=121
		allocated by = Xorg
		refcount=2
		format=XR24 little-endian (0x34325258)
		modifier=0x0
		size=3840x2160
		layers:
			size[0]=3840x2160
			pitch[0]=15360
			offset[0]=0
			obj[0]:
				name=1
				refcount=4
				start=00101fac
				size=33177600
				imported=no
	crtc-pos=3840x2160+0+0
	src-pos=3840.000000x2160.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=3
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[86]: Overlay_2
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=4
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[98]: Overlay_3
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=5
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
plane[109]: Cursor
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=ff
	color-encoding=ITU-R BT.601 YCbCr
	color-range=YCbCr limited range
plane[112]: Cursor_1
	crtc=DC8200
	fb=119
		allocated by = Xorg
		refcount=1
		format=AR24 little-endian (0x34325241)
		modifier=0x0
		size=64x64
		layers:
			size[0]=64x64
			pitch[0]=256
			offset[0]=0
			obj[0]:
				name=0
				refcount=3
				start=00101fa8
				size=16384
				imported=no
	crtc-pos=64x64+1899+1057
	src-pos=64.000000x64.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=ff
	color-encoding=ITU-R BT.601 YCbCr
	color-range=YCbCr limited range
crtc[31]: DC8200
	enable=0
	active=0
	self_refresh_active=0
	planes_changed=0
	mode_changed=0
	active_changed=0
	connectors_changed=0
	color_mgmt_changed=0
	plane_mask=0
	connector_mask=0
	encoder_mask=0
	mode: "": 0 0 0 0 0 0 0 0 0 0 0x0 0x0
crtc[35]: DC8200
	enable=1
	active=1
	self_refresh_active=0
	planes_changed=1
	mode_changed=0
	active_changed=0
	connectors_changed=0
	color_mgmt_changed=0
	plane_mask=88
	connector_mask=1
	encoder_mask=1
	mode: "": 60 594000 3840 4016 4104 4400 2160 2164 2174 2250 0x0 0x9
connector[116]: HDMI-A-1
	crtc=DC8200
	self_refresh_aware=0
	max_requested_bpc=0
	colorspace=Default
```
另外，在连接到显示器非采集卡时风扇会明显加速转动几秒钟

Icenowy, 2025-04-02:
之前所给出的 EDID 是否为显示器 EDID ？

Icenowy, 2025-04-02:
> ```
> $ hwinfo --monitor
> 20: None 00.0: 10002 LCD Monitor
>   [Created at monitor.125]
>   Unique ID: rdCR.rLw16Jkfqh5
>   Parent ID: _Znp.iH5RHOo8m+D
>   Hardware Class: [...]

非常奇怪此处显示的时序似乎与显示器 EDID 中读出的时序不完全匹配， EDID 读出的时序频率为 594MHz
以及该显示器的原生分辨率是？

Icenowy, 2025-04-02:
对 state 文件的内容进行了分析，看上去试图设置的是正常的 4K@60 594MHz 显示模式；这种模式下感觉不应该有问题才对（小声

DingWH03, 2025-04-02:
> 之前所给出的 EDID 是否为显示器 EDID ？

是的，是显示器的EDID

DingWH03, 2025-04-02:
> 以及是否开机时插入显示器但之后换成采集卡才会触发花屏？

是这样的，插入显示器不会花屏，但是输出黑屏，只有此时拔出显示器插入采集卡出现花屏

DingWH03, 2025-04-02:
> 对 state 文件的内容进行了分析，看上去试图设置的是正常的 4K@60 594MHz 显示模式；这种模式下感觉不应该有问题才对（小声

也许是尝试设置正常的594MHz但没有成功？我可能需要检查一下是否可能是其他问题，上面我提供的信息似乎有些混乱了，明天我将整理下面3种情况放在文件中

1. 插着显示器开机无法正常输出画面时信息
2. 拔出显示器接入到采集卡时信息
3. 另一台正常连接到显示器工作输出正常画面的电脑的信息

DingWH03, 2025-04-03:
今天我重新捋了一遍，发现前面似乎误导你们了，前面的显示器信息hwinfo是我使用连接在同一显示器的另一接入dp的电脑截出来的，不是licheepi4a给出的hwinfo，事实上licheepi4a的hwinfo无论显示是否正常`hwinfo --monitor`均无输出(已在root权限下)
当前的信息如下：
1. 开机时插入显示器，则显示器无法接收到输出，接入采集卡之后可能花屏也可能黑屏（尝试很多次现象不一致）
2. 插入采集卡开机，采集卡正常显示，拔出，后转为插入显示屏，可以显示但似乎不正常，附上[全程视频](http://ssh.cxhap.top:25611/VID_20250403_080425.mp4)
3. 正常显示和异常显示的两种状态下读取出来的edid和state均相同
总结就是只要不插着显示器开机，最起码显示屏能点亮，采集卡画面是正常的

[黑屏之后插入采集卡花屏.md](https://github.com/user-attachments/files/19577802/default.md)
[开机插入显示器黑屏.md](https://github.com/user-attachments/files/19577799/default.md)
[其它电脑连接该显示器.md](https://github.com/user-attachments/files/19577801/default.md)
[正常使用该显示器.md](https://github.com/user-attachments/files/19577800/default.md)