revyos/revyos #123: [LPi4A] Missing WLAN0 Sometimes

Pairman, 2025-04-06:
Hello. I found that sometimes after powering and pooting, ```wlan0``` is missing in both ```ifconfig``` or ```nmcli d s```. The issues persists after simply rebooting , but can be fixed with re-powering.

I cannot right as the time of writing, yet it has occurred twice.

RevyOS build 20250122:
```bash
$ uname -a
Linux revyos-lpi4a 6.6.73-th1520 #2025.01.22.09.36+39bfa82ff SMP Wed Jan 22 09:53:08 UTC 2025 riscv64 G
NU/Linux
```

KevinMX, 2025-04-06:
Hi, please check the model of your Wi-Fi chip, near the USB Type-C port & GPIO headers.
See the picture below, marked in the red square (I tooked the picture from Sipeed's wiki)
If it's not RTL8723DS (like this one in the picture) then it's probably AIC8800, and this is a known issue.

[image]

Pairman, 2025-04-06:
Can confirm it's AIC8800.

Pairman, 2025-04-06:
It happened again. This time I captured all the logs: [1.log](https://github.com/user-attachments/files/19621471/1.log)

stydxm, 2025-04-12:
I found this happening not only with`reboot`. I mean if `poweroff` in system then press the physical power button, it still throws the same error. It seems only replugging in the power wire can make wlan work again.