revyos/revyos #126: LicheePi 4A 使用 NPU（TH1520） 运行 yolov5n 时遇到问题

lalafua, 2025-04-17:
## 环境
- LicheePi 4A： 8g ram + 32g emmc
- 镜像版本： 6.6.82-th1520
- Python： 3.11.1

## 问题表现
用 NPU 运行时无法识别到图片中的物体(使用的图片为 yolov5n 自带的测试图片)
```bash
(.venv) debian@revyos-lpi4a:~/Workspace/sim_llm/riscv/yolov5n_npu$ sudo python3 inference.py 
LIBGL: Initialising gl4es
LIBGL: v1.1.5 built on Oct  8 2023 14:20:52
LIBGL: Using GLES 2.0 backend
LIBGL: loaded: libGLESv2.so
LIBGL: loaded: libEGL.so
LIBGL: Using GLES 2.0 backend
X11 connection rejected because of wrong authentication.
X11 connection rejected because of wrong authentication.
X11 connection rejected because of wrong authentication.
LIBGL: Error while gathering supported extension (eglInitialize: EGL_NOT_INITIALIZED), default to none
LIBGL: Targeting OpenGL 2.1
LIBGL: Not trying to batch small subsequent glDrawXXXX
LIBGL: try to use VBO
LIBGL: Force texture for Attachment color0 on FBO
LIBGL: Hack to trigger a SwapBuffers when a Full Framebuffer Blit on default FBO is done
LIBGL: glX Will try to recycle EGL Surface
LIBGL: Current folder is:/home/debian/Workspace/sim_llm/riscv/yolov5n_npu
LIBGL: Not using PSA (prgbin_n=0, notexarray=0)
 ********** preprocess image **********
 ******* run yolov5 and postprocess *******
INFO: NNA clock:406720 [kHz]
INFO: Heap :anonymous (0x2)
INFO: Heap :dmabuf (0x2)
INFO: Heap :unified (0x5)
WARNING: Mapping to the on chip ram failed (128 > 0), continuing...
Run graph execution time: 5.67983ms, FPS=176.06
detect num: 0
id:	label	score		x1		y1		x2		y2
free(): invalid pointer
Aborted
 ********** draw bbox **********
LIBGL: Shuting down
```

相关日志
```
(.venv) debian@revyos-lpi4a:~/Workspace/sim_llm/riscv/yolov5n_npu$ sudo dmesg | grep -i vha
[    0.669181] /soc/vha@fffc800000: Fixed dependency cycle(s) with /soc/vha@fffc800000
[    0.732648] /soc/vha@fffc800000: Fixed dependency cycle(s) with /soc/vha@fffc800000
[ 5569.280035] vha_plat_probe: Version: VHA DT driver version : REL_3.8-cl6140200
[ 5569.288895] vha_plat_probe: registers 0xfffc800000-0xfffc8fffff
[ 5569.294974] ax3xxx-nna fffc800000.vha: trimming system conf for core region!
[ 5569.302100] ax3xxx-nna fffc800000.vha: vha_plat_dt_hw_init dev->dma_mask : 00000000c08bdb97 : 0xffffffff
[ 5569.311715] ax3xxx-nna fffc800000.vha: vha_plat_dt_hw_init forcing custom mask from DT : 0xffffffffff
[ 5569.321034] ax3xxx-nna fffc800000.vha: vha_dev_get_props: Product id: 0x8470000
[ 5569.328411] ax3xxx-nna fffc800000.vha: vha_dev_get_props: Core id: 0x1c0004190005de
[ 5569.336105] ax3xxx-nna fffc800000.vha: vha_dev_get_props: MMU version:3 (40bit)
[ 5569.343466] ax3xxx-nna fffc800000.vha: vha_dev_get_props: Total onchip memory: 0 [kB]
[ 5569.351336] ax3xxx-nna fffc800000.vha: vha_dev_get_props: Devices: DUMMY:0 CNN:1
[ 5569.361731] ax3xxx-nna fffc800000.vha: vha_devfreq_init: Success to register the NPU to DevFreq.
[ 5569.370901] ax3xxx-nna fffc800000.vha: vha_add_dev: Core freq[kHz]: to be calibrated
[ 5569.379147] ax3xxx-nna fffc800000.vha: vha_dev_calibrate: Starting core frequency measurement (268435455)...
[ 5570.049741] ax3xxx-nna fffc800000.vha: vha_measure_core_freq: Measured core clock frequency[kHz]: 406720
[ 5628.686910] ax3xxx-nna fffc800000.vha: vha_map_to_onchip: moving a page to on chip ram failed!
[ 6942.747456] ax3xxx-nna fffc800000.vha: vha_map_to_onchip: moving a page to on chip ram failed!
```

## 参考文档
NPU 环境配置来自： [外设使用-NPU](https://wiki.sipeed.com/hardware/zh/lichee/th1520/lpi4a/6_peripheral.html#NPU)
yolov5n 配置来自： [典型应用-yolov5n](https://wiki.sipeed.com/hardware/zh/lichee/th1520/lpi4a/8_application.html#Yolov5n) 
示例代码来自： [hbb-examples](https://github.com/zhangwm-pt/hhb-examples/blob/main/th1520_npu/yolov5n/inference.py)

> 若在输出中有 img_mem，vha 和 vha_info 这三个模块，NPU驱动即加载成功。

但是这个版本的镜像貌似并没有自动加载 `vha` , `vha_info`, `img_mem` 这三个驱动，我是手动加载的。

## 一些排查
为防止是模型的原因造成的不识别问题，还是参考上面的文档，我用 cpu 跑了一下 yolov5n，可以正常运行。
- yolov5n
	>将上述 NPU 步骤中的 HHB 编译命令替换为：
	```bash
	hhb -D --model-file mobilenetv2-12.onnx --data-scale 0.017 --data-mean "124 117 104"  --board c920  --postprocess save_and_top5 --input-name "input" --output-name "output" --input-shape "1 3 224 224"
	```
	>g++ 编译后处理命令替换为：
	```bash
	riscv64-unknown-linux-gnu-g++ main.cpp -I../prebuilt_opencv/include/opencv4 -L../prebuilt_opencv/lib   -lopencv_imgproc   -lopencv_imgcodecs -L../prebuilt_opencv/lib/opencv4/3rdparty/ -llibjpeg-turbo -llibwebp -llibpng -llibtiff -llibopenjp2    -lopencv_core -ldl  -lpthread -lrt -lzlib -lcsi_cv -latomic -static -o mobilenetv2_example
	```

	yolov5n 对 cpu 的配置几乎与 npu 相同，只是在编译时更改了几个参数，运行结果如下(仍然是同一张图片)：
	```bash
	(.venv) debian@revyos-lpi4a:~/Workspace/sim_llm/riscv/yolov5n_cpu$ sudo python3 inference.py 
	LIBGL: Initialising gl4es
	LIBGL: v1.1.5 built on Oct  8 2023 14:20:52
	LIBGL: Using GLES 2.0 backend
	LIBGL: loaded: libGLESv2.so
	LIBGL: loaded: libEGL.so
	LIBGL: Using GLES 2.0 backend
	X11 connection rejected because of wrong authentication.
	X11 connection rejected because of wrong authentication.
	X11 connection rejected because of wrong authentication.
	LIBGL: Error while gathering supported extension (eglInitialize: EGL_NOT_INITIALIZED), default to none
	LIBGL: Targeting OpenGL 2.1
	LIBGL: Not trying to batch small subsequent glDrawXXXX
	LIBGL: try to use VBO
	LIBGL: Force texture for Attachment color0 on FBO
	LIBGL: Hack to trigger a SwapBuffers when a Full Framebuffer Blit on default FBO is done
	LIBGL: glX Will try to recycle EGL Surface
	LIBGL: Current folder is:/home/debian/Workspace/sim_llm/riscv/yolov5n_cpu
	LIBGL: Not using PSA (prgbin_n=0, notexarray=0)
	 ********** preprocess image **********
	 ******* run yolov5 and postprocess *******
	Run graph execution time: 371.21136ms, FPS=2.69
	detect num: 4
	id:	label	score		x1		y1		x2		y2
	[0]:	0	0.899609	274.486389	158.510849	359.157715	332.118591
	[1]:	0	0.880201	80.017410	184.470093	190.141861	349.840637
	[2]:	0	0.844358	219.474869	221.711838	283.615723	333.643250
	[3]:	33	0.667759	67.194008	174.118088	203.020660	220.667618
	 ********** draw bbox **********
	[274.486389, 158.510849, 359.157715, 332.118591, 0.899609, 0]
	[80.01741, 184.470093, 190.141861, 349.840637, 0.880201, 0]
	[219.474869, 221.711838, 283.615723, 333.64325, 0.844358, 0]
	[67.194008, 174.118088, 203.02066, 220.667618, 0.667759, 33]
	LIBGL: Shuting down
	```

yuri-yuki, 2025-04-17:
复现成功，大概定位了下原因，应该是yolov5n.c预处理或后处理的问题，NPU本身没有问题，部分已研究的情况如下：

**测试1**：配置Mobilenetv2进行x86模拟和NPU实际运行，输出的tensor基本一致，见下图1（x86模拟输出）图2（npu计算输出）：

[image]

[image]

手动查找top1对应的class和syn（ImageNet数据集），结果确实为波斯猫，说明推理正确。

hhb生成命令
```
hhb -D --model-file resnet50-v1-7.onnx --data-scale 0.017 --data-mean "124 117 104"  --board th1520  --postprocess save_and_top5 --input-name "data" --output-name "resnetv17_dense0_fwd" --input-shape "1 3 224 224" --calibrate-dataset persian_cat.jpg  --quantization-scheme "int8_asym" --simulate-data persian_cat.jpg
```

hhb模拟命令
```
hhb -S --model-file resnet50-v1-7.onnx --data-scale 0.017 --data-mean "124 117 104"  --board c920  --postprocess save_and_top5 --input-name "data" --output-name "resnetv17_dense0_fwd" --input-shape "1 3 224 224" --calibrate-dataset persian_cat.jpg  --quantization-scheme "int8_asym" --simulate-data persian_cat.jpg
```

**测试2**：配置yolov5n进行x86模拟和NPU实际运行，输出的tensor基本一致。见下面的输出。

```
(.venv) root@8305ff3668b0:~/hhb-examples/th1520_npu/yolov5n# hhb -S --model-file yolov5n.onnx --data-scale-div 255 --board th1520 --input-name "images" --output-name "/model.24/m.0/Conv_output_0;/model.24/m.1/Conv_output_0;/model.24/m.2/Conv_output_0" --input-shape "1 3 384 640" --calibrate-dataset kite.jpg  --quantization-scheme "int8_asym" --simulate-data kite.jpg 
............
[2025-04-17 18:58:37] (HHB LOG): cd hhb_out; qemu-riscv64 -cpu c920 hhb_runtime ./hhb.bm kite.jpg.0.bin 
Run graph execution time: 62600.49609ms, FPS=0.01597

=== tensor info ===
shape: 1 3 384 640 
data pointer: 0x4000ca2010

=== tensor info ===
shape: 1 255 48 80 
data pointer: 0x40e2f0
The max_value of output: 3.359876
The min_value of output: -16.967373
The mean_value of output: -5.476504
The std_value of output: 4.439581
 ============ top5: ===========
329636: 3.359876
656036: 3.359876
667150: 3.275879
348430: 3.191882
336654: 3.107885

=== tensor info ===
shape: 1 255 24 40 
data pointer: 0x2ffe50
The max_value of output: 6.492655
The min_value of output: -14.110703
The mean_value of output: -5.466867
The std_value of output: 3.632962
 ============ top5: ===========
168619: 6.492655
168579: 6.146380
168620: 5.886674
168647: 5.713537
5419: 5.453830

=== tensor info ===
shape: 1 255 12 20 
data pointer: 0x2e9620
The max_value of output: 5.638582
The min_value of output: -13.081510
The mean_value of output: -6.005158
The std_value of output: 3.741653
 ============ top5: ===========
21749: 5.638582
21787: 5.413038
21769: 5.262676
42149: 5.112314
1349: 5.037133
```

```
(.venv) debian@revyos-lpi4a:~/demo2/hhb_out$ ./hhb_runtime hhb.bm images.0.bin
INFO: NNA clock:406105 [kHz]
INFO: Heap :anonymous (0x2)
INFO: Heap :dmabuf (0x2)
INFO: Heap :unified (0x5)
WARNING: Mapping to the on chip ram failed (128 > 0), continuing...
FATAL: Importing 737280 bytes of CPU memory has failed (wrong memory alignment)
Run graph execution time: 19.09660ms, FPS=52.36535

=== tensor info ===
shape: 1 3 384 640
data pointer: 0x15f2d90

=== tensor info ===
shape: 1 255 48 80
data pointer: 0x3fb6956000
The max_value of output: 3.443872
The min_value of output: -17.639347
The mean_value of output: -5.491457
The std_value of output: 4.565122
 ============ top5: ===========
329636: 3.443872
656036: 3.443872
800654: 3.275879
800732: 3.275879
336654: 3.191882

=== tensor info ===
shape: 1 255 24 40
data pointer: 0x3fb691a000
The max_value of output: 6.492655
The min_value of output: -14.110703
The mean_value of output: -5.498645
The std_value of output: 3.708484
 ============ top5: ===========
168619: 6.492655
168579: 6.059812
168620: 5.886674
5419: 5.540399
168647: 5.540399

=== tensor info ===
shape: 1 255 12 20
data pointer: 0x3fb8980000
The max_value of output: 5.638582
The min_value of output: -13.382234
The mean_value of output: -6.034372
The std_value of output: 3.783273
 ============ top5: ===========
21749: 5.638582
21787: 5.413038
21769: 5.337857
1349: 5.262676
21750: 5.187495
```

可见hhb推理输出基本一致。

从上述两个测试应该可以看出，可能是yolov5n.c后处理或前处理的问题。

hhb本身的输出应该没有太大问题。yolov5n.c相当于实现了`hhb_out/hhb_runtime`，并且实现了yolo网络需要的后处理过程。

yuri-yuki, 2025-04-18:
编写了一个后处理脚本，证实是原项目的yolov5n.c后处理的问题。

后处理脚本参见https://gist.github.com/yuri-yuki/b4945b1541526dbf15a9f8a0ba91ac4b

验证方法：

1. 将`hhb_out`目录复制到板子上，运行命令`./hhb_runtime hhb.bm kite.jpg.0.bin`，这个命令将在npu上运行yolo模型，但不进行后处理
2. 程序运行完成后会出现三个文件`kite.jpg.0.bin_output0_1_255_48_80.txt` `kite.jpg.0.bin_output1_1_255_24_40.txt` `kite.jpg.0.bin_output2_1_255_12_20.txt`（如果没有这三个，检查一下hhb编译命令是否带`--postprocess save_and_top5`），这是yolo运行的结果tensor，还没有后处理
3. 重命名为output1.txt...output3.txt，运行上面提供的python脚本，能够成功检测到三个人和一个风筝。

[image]

原项目提供的`yolov5n.c`结合了hhb生成的几个c文件和yolo本身的后处理，只需要更改后处理的函数即可。

KamijoToma, 2025-04-19:
Hi, I wrote a python binding of libshl-th1520.so and implemented NPU model inference with numpy and opencv. You can try it [here](https://github.com/KamijoToma/th1520_npu_python)

The example shows a inference with yolov5n and TH1520 NPU:

[image]

lalafua, 2025-04-20:
>编写了一个后处理脚本，证实是原项目的yolov5n.c后处理的问题。

非常感谢您的研究以及提供的脚本！的确是示例项目代码的问题。
[image]

>Hi, I wrote a python binding of libshl-th1520.so and implemented NPU model inference with numpy and opencv. You can try it [here](https://github.com/KamijoToma/th1520_npu_python)

Thanks for sharing your project! This looks very helpful, I will try it later.

RevySR, 2025-04-20:
已澄清问题

closed 2025-04-20 by RevySR