revyos/revyos #128: No docker in apt install

LTXWorld, 2025-05-13:
LicheePi4A 16G + revyos_20250420

```bash
sudo apt install docker.io
Error: Unable to locate package docker.io
Error: Couldn't find any package by glob 'docker.io'
debian@revyos-lpi4a:~$ sudo apt install docker
Error: Unable to locate package docker
```

我在官方文档中的系统使用中看到有 docker的使用,但是包管理器中没有docker?

如果我要使用dokcer是否需要自己构建?

LTXWorld, 2025-05-13:
apt update 遇到问题
```bash
uname -a
Linux revyos-lpi4a 6.6.82-th1520 #2025.03.11.14.28+9292e379e SMP Tue Mar 11 14:48:30 UTC 2025 riscv64 GNU/Linux
sudo apt update
Err:1 https://fast-mirror.isrc.ac.cn/revyos/revyos-gles-21 revyos-gles-21 InRelease
  The following signatures were invalid: EXPKEYSIG 2FB3A9E77911527E Han Gao <gaohan@iscas.ac.cn>
Get:2 https://fast-mirror.isrc.ac.cn/revyos/revyos-addons revyos-addons InRelease [9,442 B]
Err:2 https://fast-mirror.isrc.ac.cn/revyos/revyos-addons revyos-addons InRelease
  The following signatures were invalid: EXPKEYSIG 2FB3A9E77911527E Han Gao <gaohan@iscas.ac.cn>
Get:3 https://fast-mirror.isrc.ac.cn/revyos/revyos-kernels revyos-kernels InRelease [5,806 B]
Err:3 https://fast-mirror.isrc.ac.cn/revyos/revyos-kernels revyos-kernels InRelease
  The following signatures were invalid: EXPKEYSIG 2FB3A9E77911527E Han Gao <gaohan@iscas.ac.cn>
Get:4 https://fast-mirror.isrc.ac.cn/revyos/revyos-base sid InRelease [12.3 kB]
Err:4 https://fast-mirror.isrc.ac.cn/revyos/revyos-base sid InRelease
  The following signatures were invalid: EXPKEYSIG 2FB3A9E77911527E Han Gao <gaohan@iscas.ac.cn>
Warning: GPG error: https://fast-mirror.isrc.ac.cn/revyos/revyos-gles-21 revyos-gles-21 InRelease: The following signatures were invalid: EXPKEYSIG 2FB3A9E77911527E Han Gao <gaohan@iscas.ac.cn>
Error: The repository 'https://mirror.iscas.ac.cn/revyos/revyos-gles-21 revyos-gles-21 InRelease' is not signed.
Notice: Updating from such a repository can't be done securely, and is therefore disabled by default.
Notice: See apt-secure(8) manpage for repository creation and user configuration details.
Warning: GPG error: https://fast-mirror.isrc.ac.cn/revyos/revyos-addons revyos-addons InRelease: The following signatures were invalid: EXPKEYSIG 2FB3A9E77911527E Han Gao <gaohan@iscas.ac.cn>
Error: The repository 'https://mirror.iscas.ac.cn/revyos/revyos-addons revyos-addons InRelease' is not signed.
Notice: Updating from such a repository can't be done securely, and is therefore disabled by default.
Notice: See apt-secure(8) manpage for repository creation and user configuration details.
Warning: GPG error: https://fast-mirror.isrc.ac.cn/revyos/revyos-kernels revyos-kernels InRelease: The following signatures were invalid: EXPKEYSIG 2FB3A9E77911527E Han Gao <gaohan@iscas.ac.cn>
Error: The repository 'https://mirror.iscas.ac.cn/revyos/revyos-kernels revyos-kernels InRelease' is not signed.
Notice: Updating from such a repository can't be done securely, and is therefore disabled by default.
Notice: See apt-secure(8) manpage for repository creation and user configuration details.
Warning: GPG error: https://fast-mirror.isrc.ac.cn/revyos/revyos-base sid InRelease: The following signatures were invalid: EXPKEYSIG 2FB3A9E77911527E Han Gao <gaohan@iscas.ac.cn>
Error: The repository 'https://mirror.iscas.ac.cn/revyos/revyos-base sid InRelease' is not signed.
Notice: Updating from such a repository can't be done securely, and is therefore disabled by default.
Notice: See apt-secure(8) manpage for repository creation and user configuration details
```

请问这个该如何解决呢?

LTXWorld, 2025-05-13:
> LicheePi4A 16G + revyos_20250420
> 
> sudo apt install docker.io
> Error: Unable to locate package docker.io
> Error: Couldn't find any package by glob 'docker.io'
> debian@revyos-lpi4a:~$ sudo apt install docker
> Error: Unable to locate package docker
> 我在官方文档中的系统使用中看到有 docker的使用,但是包管理器中没有docker?
> 
> 如果我要使用dokcer是否需要自己构建?

我修改了apt的配置,添加了 trusted使其可以install git,docker等,但是该如何执行docker呢?

```bash
debian@revyos-lpi4a:~$ sudo systemctl status docker
Unit docker.service could not be found.
debian@revyos-lpi4a:~$ sudo docker pull riscv64/debian:unstable
sudo: docker: command not found
```

KevinMX, 2025-05-13:
> apt update 遇到问题
> 
> 请问这个该如何解决呢?

https://github.com/revyos/revyos/issues/125

LTXWorld, 2025-05-13:
> > apt update 遇到问题
> > 请问这个该如何解决呢?
> 
> [#125](https://github.com/revyos/revyos/issues/125)

请问docker有支持吗?

KevinMX, 2025-05-13:
正确的包名应为 `docker.io`。

KevinMX, 2025-05-19:
我猜楼主应该已经解决了 Docker 问题；实际上我们也未能复现 Docker 安装好后无法使用的情况。

先 Close 了，有问题可以重开/开新的。