revyos/revyos #130: SD 卡在大 IO 读时报错的现象，在最新版本的 revyos 中，又出现了

atzlinux, 2025-05-23:
https://github.com/revyos/revyos/issues/43

这个问题在 20250420 版 revyos 中，又出现了：

root@revyos-lpi4a:~# uname -a
Linux revyos-lpi4a 6.6.82-th1520 #2025.03.11.14.28+9292e379e SMP Tue Mar 11 14:48:30 UTC 2025 riscv64 GNU/Linux

root@revyos-lpi4a:~# hdparm -t /dev/mmcblk1

/dev/mmcblk1:
 Timing buffered disk reads:  72 MB in  4.39 seconds =  16.42 MB/sec

IO 不到 20MB

dmesg 有报错：

[Fri May 23 14:59:15 2025] blk_print_req_error: 5 callbacks suppressed
[Fri May 23 14:59:15 2025] I/O error, dev mmcblk1, sector 139808 op 0x0:(READ) flags 0x80700 phys_seg 4 prio class 2
[Fri May 23 14:59:16 2025] I/O error, dev mmcblk1, sector 140296 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 2
[Fri May 23 14:59:16 2025] I/O error, dev mmcblk1, sector 141400 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 2
[Fri May 23 14:59:16 2025] I/O error, dev mmcblk1, sector 141401 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 2
[Fri May 23 14:59:16 2025] I/O error, dev mmcblk1, sector 142376 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 2
[Fri May 23 14:59:16 2025] I/O error, dev mmcblk1, sector 142379 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 2
[Fri May 23 14:59:17 2025] I/O error, dev mmcblk1, sector 143361 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 2
[Fri May 23 14:59:17 2025] I/O error, dev mmcblk1, sector 143362 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 2
[Fri May 23 14:59:17 2025] I/O error, dev mmcblk1, sector 144384 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 2
[Fri May 23 14:59:17 2025] I/O error, dev mmcblk1, sector 144386 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 2

KamijoToma, 2025-05-23:
同样的命令，使用SanDisk Ultra A1 64G存储卡运行无问题。

```
debian@revyos-lpi4a:~$ uname -a
Linux revyos-lpi4a 6.6.82-th1520 #2025.03.11.14.28+9292e379e SMP Tue Mar 11 14:48:30 UTC 2025 riscv64 GNU/Linux
debian@revyos-lpi4a:~$ sudo hdparm -t /dev/mmcblk1
[  176.063204] audit: type=1101 audit(1747985275.570:128): pid=880 uid=1000 auid=1000 ses=2 subj=unconfined msg='op=PAM:accounting grantors=pam_permit acct="debian" exe="/usr/bin/sudo" hostname=revyos-lpi4a addr=? terminal=/dev/ttyS0 res=success'
[  176.084855] audit: type=1123 audit(1747985275.570:129): pid=880 uid=1000 auid=1000 ses=2 subj=unconfined msg='cwd="/home/debian" cmd=68647061726D202D74202F6465762F6D6D63626C6B31 exe="/usr/bin/sudo" terminal=ttyS0 res=success'
[  176.105540] audit: type=1110 audit(1747985275.610:130): pid=880 uid=1000 auid=1000 ses=2 subj=unconfined msg='op=PAM:setcred grantors=pam_permit acct="root" exe="/usr/bin/sudo" hostname=revyos-lpi4a addr=? terminal=/dev/ttyS0 res=success'
[  176.126738] audit: type=1105 audit(1747985275.614:131): pid=880 uid=1000 auid=1000 ses=2 subj=unconfined msg='op=PAM:session_open grantors=pam_limits,pam_permit,pam_unix acct="root" exe="/usr/bin/sudo" hostname=revyos-lpi4a addr=? terminal=/dev/ttyS0 res=success'

/dev/mmcblk1:
 Timing buffered disk reads: 262 MB in  3.00 seconds =  87.29 MB/sec
[  182.328371] audit: type=1106 audit(1747985281.830:132): pid=880 uid=1000 auid=1000 ses=2 subj=unconfined msg='op=PAM:session_close grantors=pam_limits,pam_permit,pam_unix acct="root" exe="/usr/bin/sudo" hostname=revyos-lpi4a addr=? terminal=/dev/ttyS0 res=success'
[  182.351834] audit: type=1104 audit(1747985281.830:133): pid=880 uid=1000 auid=1000 ses=2 subj=unconfined msg='op=PAM:setcred grantors=pam_permit acct="root" exe="/usr/bin/sudo" hostname=revyos-lpi4a addr=? terminal=/dev/ttyS0 res=success'
debian@revyos-lpi4a:~$
```

RevySR, 2025-05-23:
麻烦提供一下flash卡具体型号 我们对flash卡支持是有限的 无法获取所有的flash卡做tuning动作
一般来说会保证主流flash卡可以工作

atzlinux, 2025-05-23:
我刚才修改了 设备树 文件， 将 SD 卡对应的 mmc1 的最大频率设置为 100MHz 后，

IO 读速率能够上 40M 了。

***@***.***:~# hdparm -t /dev/mmcblk1

/dev/mmcblk1:
 Timing buffered disk reads: 136 MB in  3.02 seconds =  44.97 MB/sec

此时 dmesg 也没有 IO 报错。

                ***@***.*** {
                        compatible = "xuantie,th1520-dwcmshc";
                        reg = <0xff 0xe7090000 0x00 0x10000>;
                        interrupts = <0x40 0x04>;
                        clocks = <0x02 0x7a 0x2d 0x05>;
                        clock-names = "core", "bus";
                        status = "okay";
                        pinctrl-names = "default";
                        pinctrl-0 = <0x43>;
                        bus-width = <0x04>;
                        max-frequency = <0x5f5e100>;
                        no-sdio;
                };

在这里的 max-frequency 设置为 100MHz， 16 进制为 0x5f5e100

在 2025/5/23 16:26, Han Gao 写道:
> RevySR left a comment (revyos/revyos#130)
>
> 麻烦提供一下flash卡具体型号 我们对flash卡支持是有限的 无法获取所有的flash卡做tuning动作
> 一般来说会保证主流flash卡可以工作
在华强北买的卡，我在系统和 卡 外表能够看到的信息为：

系统 dmesg 信息：

[    5.317984] mmc1: new ultra high speed SDR104 SDXC card at address 1388
[    5.333448] mmcblk1: mmc1:1388 USD00 230 GiB

卡照片见附件。

-- 
肖盛文 xiao sheng wen
微信(wechat)：atzlinux
《铜豌豆 Linux》https://www.atzlinux.com
基于 Debian 的 Linux 中文 桌面 操作系统
GnuPG Public Key: 0x2F338C7DC7909957

RevySR, 2025-05-23:
> 我刚才修改了 设备树 文件， 将 SD 卡对应的 mmc1 的最大频率设置为 100MHz 后，
>
> IO 读速率能够上 40M 了。
>
> ***@***.***:~# hdparm -t /dev/mmcblk1
>
> /dev/mmcblk1:
>  Timing buffered disk reads: 136 MB in  3.02 seconds =  44.97 MB/sec
>
> 此时 [...]

似乎不是常见的卡 部分品牌标识似乎都标错误了 这种卡似乎都违反SD-3C[1]的一些许可

[2]是授权的生产的SD卡的厂商 我看这个似乎也不在其中 是不是应该和您的经销商联系一下 确认一下合格的卡呢

https://www.sd-3c.com/FrequentlyAskedQuestions.aspx?Language=zh-CN[1]
https://www.sd-3c.com/Licensees.aspx?Language=zh-CN

RevySR, 2025-05-26:
https://github.com/revyos/th1520-linux-kernel/commit/c9a17b235e0153ae8b1b6dd3812a8d00eb1ad826

lts branch merged.

closed 2025-05-26 by RevySR