revyos/revyos #131: milk-v meles th1520 npu加载与使用报错

IfReturn, 2025-06-06:
系统版本：20250526  
开发板：milk-v meles  
复现步骤：
```shell
sudo apt install python3-full
python3 -m venv ./shl
./shl/bin/pip install shl-python
LIB=$(./shl/bin/python3 -m shl --whereis th1520)/lib
export LD_LIBRARY_PATH=$LIB:$LD_LIBRARY_PATH
cd hhb_out
./hhb_runtime ./hhb.bm ../bus.jpg
```
其中hhb编译命令：
```shell
hhb -C --model-file ./yolov8n.onnx --data-scale-div 255 --board th1520 --input-name "images" --output-name "output0" --input-shape "1 3 640 640" --calibrate-dataset bus.jpg  --quantization-scheme "int8_asym"
```
yolov8n.onnx从ultralytics包直接导出
报错：
```shell
Strided_slice software implementation in nna_ddk has memory leak, please use heterogeneous execution.Strided_slice software implementation in nna_ddk has memory leak, please use heterogeneous execution.ERROR: (NNA_INTERNAL_FUNC) (/usr/src/debug/npu-ax3386/2.0.2-r0/lib/source/offline/mapping_tool/CnnNode.hpp:1359) All inputs must have the same #axes.
ERROR: (NNA_INTERNAL_FUNC) Failure initialising layer 'img_concat19'
ERROR: (NNA_INTERNAL_FUNC) Failure to initialise dimensions of network
ERROR: (NNA_INTERNAL_FUNC) CNN Model generation failed: std::exception
ERROR: (csiImgdnnGenerateMBS) Error when creating imgdnn network binary
ERROR: (csiNna_imgdnnCreateNetworkBinary) Error while creating network object
Could not create network object
FATAL: could not open driver '/dev/vha0': No such file or directory
ERROR: (nna_imgdnnCreateContext) Could not create NNA session, error
ERROR: (nna_imgdnnLoadNetworkObject) Invalid null context
Please create or load network object first.
ERROR: (nna_imgdnnNetworkObjectGetInputs) Cannot get inputs from null network object
Could not retrieve inputs of network object
ERROR: (nna_imgdnnNetworkObjectGetInputs) Cannot get inputs from null network object
Could not retrieve inputs of network object
Run graph execution time: 0.75571ms, FPS=1323.26
```
```shell
sudo dmesg
[ 1155.577640] hhb_runtime[4214]: unhandled signal 11 code 0x1 at 0x0000000000000000 in libshl_th1520.so.2[3f94cc0000+137000]
[ 1155.588833] CPU: 1 PID: 4214 Comm: hhb_runtime Not tainted 6.6.92-th1520 #2025.05.26.14.02+c9a17b235
[ 1155.598057] Hardware name: Milk-V Meles (DT)
[ 1155.602375] epc : 0000003f94d4bb78 ra : 0000003f94d12212 sp : 0000003fd6f31e40
[ 1155.609644]  gp : 000000000007b800 tp : 0000003f94c18780 t0 : 00000000025ce6e2
[ 1155.616907]  t1 : 0000003f94d4bb70 t2 : 0000003f94dfbbb0 s0 : 00000000108f2470
[ 1155.624171]  s1 : 0000000000000000 a0 : 0000000000000000 a1 : 0000000010a649d0
[ 1155.631435]  a2 : 00000000108f25e0 a3 : 00000000108f2470 a4 : 0000003f94de7ed0
[ 1155.638696]  a5 : 0000000000000000 a6 : 0000000010a819d8 a7 : 0000000000000010
[ 1155.645956]  s2 : 0000000010a649d0 s3 : 000000000006e000 s4 : 00000000108f2470
[ 1155.653221]  s5 : 0000000000000000 s6 : 000000000006e000 s7 : 000000000006e000
[ 1155.660487]  s8 : 000000000006e000 s9 : 0000000000000000 s10: 0000000000000001
[ 1155.667750]  s11: 0000000000000004 t3 : 0000003f94e1f110 t4 : 0000003f94cd8d40
[ 1155.675016]  t5 : 0000003f94cc7120 t6 : 0000000000000023
[ 1155.680372] status: 8000000200006020 badaddr: 0000000000000000 cause: 000000000000000d
```
同时，观察到加载th1520时有报错
```shell
sudo dmesg|grep th1520
[    0.000000] Linux version 6.6.92-th1520 (builder@revyos-riscv-builder) (riscv64-linux-gnu-gcc (Ubuntu 13.2.0-4ubuntu3) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.41) #2025.05.26.14.02+c9a17b235 SMP Mon May 26 14:22:33 UTC 2025
[    3.311535] th1520-fm-clk ffef010000.clock-controller: succeed to register th1520 fullmask clock driver
[    3.406396] th1520_audiosys_clk_probe audiosys_regmap=0xffffffd9005bcc00
[    3.478825] th1520-system-monitor system-monitor: system_monitor init done!
[    3.544894] dwhdmi-th1520 ffef540000.dw-hdmi-tx: Detected HDMI TX controller v2.14a with HDCP (DWC HDMI 2.0 TX PHY)
[    3.556643] dwhdmi-th1520 ffef540000.dw-hdmi-tx: registered DesignWare HDMI I2C bus driver
[    3.565793] vs-drm display-subsystem: bound ffef540000.dw-hdmi-tx (ops dw_hdmi_th1520_ops)
[    3.631221] Watchdog module: th1520-wdt loaded
[    3.668012] xuantie,th1520-mbox-client mbox_910t_client2: Successfully registered
[    3.699186] th1520_efuse ffff210000.efuse: succeed to register th1520 efuse driver
[    8.358299] th1520-aon aon_subsys: RPC send msg timeout
[    8.363583] th1520-aon-reg aon_subsys:appcpu_dvdd: failed to get Voltages!
[    8.376872] th1520-aon-reg aon_subsys:appcpu_dvdd: Failed to initialize regulatorappcpu_dvdd
[... 126 lines cut ...]
[   40.230401] th1520-aon aon_subsys: tx_done timeout
[   40.235571] th1520-aon aon_subsys: RPC send msg failed: -110
[   40.774373] th1520-aon aon_subsys: tx_done timeout
[   40.779517] th1520-aon aon_subsys: RPC send msg failed: -110
[   41.990306] th1520-aon aon_subsys: tx_done timeout
[   41.995215] th1520-aon aon_subsys: RPC send msg failed: -110
[   46.268839] th1520-dwmac ffe7070000.ethernet end0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   46.374418] th1520-dwmac ffe7070000.ethernet end0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   46.400156] th1520-dwmac ffe7070000.ethernet end0: No Safety Features support found
[   46.407955] th1520-dwmac ffe7070000.ethernet end0: IEEE 1588-2008 Advanced Timestamp supported
[   46.417197] th1520-dwmac ffe7070000.ethernet end0: registered PTP clock
[   46.453565] th1520-dwmac ffe7070000.ethernet end0: configuring for phy/rgmii-id link mode
[   52.646348] th1520-aon aon_subsys: tx_done timeout
[   52.651244] th1520-aon aon_subsys: RPC send msg failed: -110
[   53.190346] th1520-aon aon_subsys: tx_done timeout
[   53.195259] th1520-aon aon_subsys: RPC send msg failed: -110
[   53.734656] th1520-aon aon_subsys: tx_done timeout
[   53.739568] th1520-aon aon_subsys: RPC send msg failed: -110
[   68.902292] th1520-aon aon_subsys: tx_done timeout
[   68.907250] th1520-aon aon_subsys: RPC send msg failed: -110
[   69.450300] th1520-aon aon_subsys: tx_done timeout
[   69.455185] th1520-aon aon_subsys: RPC send msg failed: -110
[   69.990322] th1520-aon aon_subsys: tx_done timeout
[   69.995247] th1520-aon aon_subsys: RPC send msg failed: -110
```

IfReturn, 2025-06-10:
烧录了SPL固件之后，`dmesg`里面没有th1520相关的报错了，但是在运行hhb_runtime的时候还是一样的报错

yuri-yuki, 2025-06-12:
> 烧录了SPL固件之后，`dmesg`里面没有th1520相关的报错了，但是在运行hhb_runtime的时候还是一样的报错

仅重新烧录了uboot固件么？从uboot到系统镜像全部重新刷一下试下？注意检查下镜像校验和

RevySR, 2025-09-30:
长时间未反馈 暂时关闭

closed 2025-09-30 by RevySR