revyos/revyos #133: LicheePi 4A(AIC8800) RevyOS 20250526 SDHCI and WiFi Module Fail to Load During Warm Boot

Ruixi-Cheng, 2025-06-17:
# Logs
Here are the logfiles for cold boot and warm boot.
(System on emmc)

[WarmBootLog.txt](https://github.com/user-attachments/files/20768162/WarmBootLog.txt)
[ColdBootLog.txt](https://github.com/user-attachments/files/20768161/ColdBootLog.txt)

Here are same log files, recorded with asciinema.
[lip4aColdBootLog.txt](https://github.com/user-attachments/files/20768103/lip4aColdBootLog.txt)
[lip4aWarmBootLog.txt](https://github.com/user-attachments/files/20768104/lip4aWarmBootLog.txt)
(GitHub cannot upload .cast extension files)

System logs show repeated errors while warm boot
```
[    4.677573] mmc2: Got command interrupt 0x00000001 even though no command operation was in progress.
[    4.687122] mmc2: sdhci: ============ SDHCI REGISTER DUMP ===========
[    4.693839] mmc2: sdhci: Sys addr:  0x00000000 | Version:  0x00000005
[    4.700555] mmc2: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
[    4.707271] mmc2: sdhci: Argument:  0x00000000 | Trn mode: 0x00000000
[    4.713989] mmc2: sdhci: Present:   0x03fa0000 | Host ctl: 0x00000001
[    4.720714] mmc2: sdhci: Power:     0x0000000f | Blk gap:  0x00000000
[    4.727438] mmc2: sdhci: Wake-up:   0x00000000 | Clock:    0x0000decf
[    4.734163] mmc2: sdhci: Timeout:   0x00000000 | Int stat: 0x00000000
[    4.740888] mmc2: sdhci: Int enab:  0x00ff1003 | Sig enab: 0x00ff1003
[    4.747613] mmc2: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[    4.754339] mmc2: sdhci: Caps:      0x3f69c881 | Caps_1:   0x08008177
[    4.761047] mmc2: sdhci: Cmd:       0x00000502 | Max curr: 0x00191919
[    4.767765] mmc2: sdhci: Resp[0]:   0x00001009 | Resp[1]:  0x00000000
[    4.774490] mmc2: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[    4.781206] mmc2: sdhci: Host ctl2: 0x00001008
[    4.785842] mmc2: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x0000000000000000
[    4.793278] mmc2: sdhci: ============================================
```

# Appearance
Cold boot works fine, but after warm boot, SD card r/w issues occur and Wi-Fi module fails to function entirely.

[image]
[image]

(When I use the system on an SD card, it even sometimes restarts repeatedly during the OpenSBI stage)

# Speculation
This may be a problem related to the SDIO interface
Same Problem with #101 #123

KevinMX, 2025-06-30:
> i encounter the similar Wifi/BT issue with 20250110 on licheepi 4a 16+128G, i didn't catch serial output, but no wifi and bluetooth connection available after boot.

Should be visible in `dmesg` if yours is the AIC8800 variant.

We've found the root cause and a workaround/fix is on it's way.

lucmann, 2025-06-30:
@KevinMX i'm not sure if it is AIC8800 variant, anyway hope this issue could be fixed ASAP.

[image]

KevinMX, 2025-07-01:
> [@KevinMX](https://github.com/KevinMX) i'm not sure if it is AIC8800 variant, anyway it'd be appreciated if this issue could be fixed ASAP.
> 
> [image]

It is.

lucmann, 2025-07-01:
FWIW, [here](https://github.com/user-attachments/files/21000408/minicom.log) is the serial log at boot